zephyrproject-rtos/zephyr #75535: Add 4-bit SD Card support to ESP32-S3

EricNRS, 2024-07-06:
The ESP32 SD/MMC Host Controller Driver (drivers/sdhc/sdhc_esp32.c) supports 1-bit to 4-bit SDIO, but currently only supports the ESP32.  Is it possible easily extend this for the ESP32-S3 or is SD/MMC controller substantially different requiring more extensive changes?

The following configuration was tested with v3.7.0-rc2 added to the test the support and it exposed the lack of the low-level HAL layer for the ESP32-S3.  The HAL layer looks to be relatively simple, but needs someone knowledgeable in this area to take a look:
```
/home/work/zephyrproject-3.7/zephyr/drivers/sdhc/sdhc_esp32.c:18:10: fatal error: hal/sdmmc_ll.h: No such file or directory
18 | #include <hal/sdmmc_ll.h>
```

```
/ {
	soc {
		sdhc: sdhc@60028000 {
			compatible = "espressif,esp32-sdhc";
			reg = <0x60028000 0x1000>;
			interrupts = <SDIO_HOST_INTR_SOURCE>;
			interrupt-parent = <&intc>;
			clocks = <&rtc ESP32_SDMMC_MODULE>;
			#address-cells = <1>;
			#size-cells = <0>;

			sdhc0: sdhc@0 {
				compatible = "espressif,esp32-sdhc-slot";
				reg = <0>;
				status = "okay";

				pinctrl-0 = <&sdhc0_default>;
				pinctrl-names = "default";
				power-delay-ms = <100>;
				max-bus-freq = <52000000>;
				bus-width = <4>;
		
				clk-pin = <8>;
				cmd-pin = <9>;
				d0-pin = <10>;
				d1-pin = <11>;
				d2-pin = <12>;
		
				mmc {
					compatible = "zephyr,sdmmc-disk";
					status = "okay";
				};				
			};

			sdhc1: sdhc@1 {
				compatible = "espressif,esp32-sdhc-slot";
				reg = <1>;
				status = "disabled";
			};
		};
	};
};

&pinctrl {

	sdhc0_default: sdhc0_default {
		group1 {
			pinmux = < ESP32_PINMUX(6, ESP_SDHOST_CARD_DETECT_N_1, ESP_NOSIG) >;
			bias-pull-up;
			output-high;
		};
	};
};
```

**Describe the solution you'd like**
Add support for ESP32-S3 SD/MMC Host Controller

**Describe alternatives you've considered**
As a backup, the 1-bit (SPI) driver built into Zephyr does work, but is substantially slower.

EricNRS, 2024-07-06:
@sylvioalves FYI.  Not urgent since SPI mode does work with the native Zephyr driver, but would be extremely useful to speed up file access.

raffarost, 2024-07-10:
Hi @EricNRS 
we have the implementation below, which still couldn't be tested for lack of a proper board.
Would you be willing to see if it works?

https://github.com/zephyrproject-rtos/zephyr/pull/75725
https://github.com/zephyrproject-rtos/hal_espressif/pull/302

raffarost, 2024-07-11:
@EricNRS thanks for your feedback. I checked with the debugger and the piece of code changed to enable clock should be okay. Other variable to check would be pinctrl, but I still didn't find any fault in the configuration. I got a working setup here, so I'll work on it when I can and let you know.

raffarost, 2024-07-11:
> Hi @EricNRS we have the implementation below, which still couldn't be tested for lack of a proper board. Would you be willing to see if it works?
> 
> [ZEP](https://github.com/zephyrproject-rtos/zephyr/pull/75725) [HAL](https://github.com/zephyrproject-rtos/hal_espressif/pull/302)

@EricNRS implementation was updated with the fix needed. I could test it only in 1 bit mode however, as I'm lacking a proper 4 bit SD socket. Could you try again and let me know? If it works in 4 bit mode I'll open the PR for review. Thanks

EricNRS, 2024-07-11:
@raffarost with the new changes, the SD card is correctly initializing and working in 4-bit mode!

Just as a side note, I did have some issues with it not working correctly and then working, but with extremely slow performance, but that is likely because the SD Card is not being reset since I do not have power-control support the version of hardware I am testing with.  Removing the SD Card and reinserting it resolved the issue and I haven't seen any issues since.

Correctly initialized in 4-bit mode, 50 MHz.  Signals look good on my hardware.
```
[00:00:00.744,000] <dbg> sd: sd_init_io: Host controller support 3.3V max
[00:00:00.746,000] <dbg> sd: sd_init_io: Resetting power to card
[00:00:00.748,000] <inf> sdhc: SDHC I/O: slot: 0, bus width 1, clock 0Hz, card power OFF, voltage 3.3V
[00:00:01.250,000] <inf> sdhc: SDHC I/O: slot: 0, bus width 1, clock 0Hz, card power ON, voltage 3.3V
[00:00:01.753,000] <inf> sdhc: SDHC I/O: slot: 0, bus width 1, clock 400000Hz, card power ON, voltage 3.3V
[00:00:01.757,000] <dbg> sd: sd_send_interface_condition: Found SDHC with CMD8 support
[00:00:01.793,000] <dbg> sd: sdmmc_send_ocr: SDMMC responded to ACMD41 after 3 attempts
[00:00:01.795,000] <dbg> sd: card_read_cid: Card MID: 0x3, OID: DS
[00:00:01.797,000] <dbg> sd: sdmmc_request_rca: Card relative addr: 43690
[00:00:01.800,000] <dbg> sd: sdmmc_read_csd: Card block count 62333952, block size 512
[00:00:01.802,000] <inf> sdhc: SDHC I/O: slot: 0, bus width 1, clock 25000000Hz, card power ON, voltage 3.3V
[00:00:01.805,000] <inf> sdhc: Bus clock successfully set to 25000 kHz
[00:00:01.809,000] <dbg> sd: sdmmc_read_scr: SD reports specification version 8
[00:00:01.814,000] <dbg> sd: sdmmc_set_bus_speed: Setting bus clock to: 50000000
[00:00:01.816,000] <inf> sdhc: SDHC I/O: slot: 0, bus width 1, clock 50000000Hz, card power ON, voltage 3.3V
[00:00:01.818,000] <inf> sdhc: Bus clock successfully set to 50000 kHz
[00:00:01.820,000] <inf> sdhc: Bus timing successfully changed to HS
[00:00:01.821,000] <inf> sdhc: SDHC I/O: slot: 0, bus width 4, clock 50000000Hz, card power ON, voltage 3.3V
[00:00:01.824,000] <inf> sdhc: Bus width set successfully to 4 bit
[00:00:01.826,000] <dbg> sd: card_read: READ: Sector = 0, Count = 1
[00:00:01.831,000] <dbg> sd: card_read: READ: Sector = 8192, Count = 1
[00:00:01.834,000] <dbg> sd: card_read: READ: Sector = 8193, Count = 1
[00:00:01.836,000] <dbg> fs: fs_mount: fs mounted at /SD:
[00:00:01.838,000] <inf> sdcard: SD card successfully mounted
```

Performance is approximately 30% better in 4-bit mode compared with single SPI mode:
```
uart> fs erase_write_test SD:/test.bin 131072 10
...
Loop #10 done in 298ms.
Total: 3174ms, Per loop: ~317ms, Speed: ~403.3KiBps
```

For comparison, using the `zephyr,sdhc-spi-slot` driver with SPI directly which ends up selecting 24 MHz.
```
[00:00:00.741,000] <dbg> sd: sd_init_io: Host controller support 3.3V max
[00:00:00.743,000] <dbg> sd: sd_init_io: Resetting power to card
[00:00:00.750,000] <dbg> sd: sd_send_interface_condition: Found SDHC with CMD8 support
[00:00:00.911,000] <dbg> sd: sdmmc_send_ocr: SDMMC responded to ACMD41 after 14 attempts
[00:00:00.915,000] <dbg> sd: card_read_cid: Card MID: 0x3, OID: DS
[00:00:00.918,000] <dbg> sd: sdmmc_read_csd: Card block count 62333952, block size 512
[00:00:00.920,000] <inf> sd: Maximum SD clock is under 25MHz, using clock of 24000000Hz
[00:00:00.925,000] <dbg> sd: sdmmc_read_scr: SD reports specification version 8
. . .

uart> fs erase_write_test SD:/test.bin 131072 10
...
Loop #10 done in 99ms.
Total: 1045ms, Per loop: ~104ms, Speed: ~306.2KiBps
```

EricNRS, 2024-07-11:
Thank you for your work on this, I really appreciate it!

raffarost, 2024-07-12:
hi @EricNRS, happy it worked!
I didn't encounter poor performance states during testing, but it may happen for the card to get stuck in very specific situations like after a failed or interrupted command sequence. It shouldn't happen in normal use though, but you can add to your hardware design a power line (`pwr-gpios`) and let Zephyr API control it. It will power cycle the card during initialization.

I found a bit odd 4 bit being 30% faster than 1 bit SPI, so I might take some time to do some benchmark testing.

Thanks for your feedback. I'll soon open the PR for review so we can have it merged when possible.

EricNRS, 2024-07-12:
@raffarost I was a bit surprised the performance was only 30% faster as well, but it could be an issue with the test or elsewhere in the call sequence.  I am also using FatFS on this SD card, so there is a chance that this is the performance bottleneck.  There also appears to be a busy loop somewhere in the call sequence since when I had slow performance issues, it triggered the watchdog as well, so somewhere the code is not yielding while waiting, but this may be a deliberate feature.

domints, 2024-07-29:
> @sylvioalves FYI. Not urgent since SPI mode does work with the native Zephyr driver, but would be extremely useful to speed up file access.

Honestly I am not convinced about "does work with the native Zephyr driver" - I spent almost whole afternoon trying to figure out CRC errors on 3 different 8GB SDHC cards. In the end I decided "what the heck, I'll try", and put in 64GB SDXC card. This SDXC card worked correctly. Why? I don't have a clue, I am sure though that builtin ESP32 SD controller works with those 8GB cards correctly, because I was able to use them within Platformio with Arduino framework.

EricNRS, 2024-07-30:
@domints Have you looked at the signals with an oscilloscope and verified voltage levels?  You may be seeing signal reflections that will only cause issues at specific clock rates.

domints, 2024-07-30:
@EricNRS I didn't (since it's all homogenous block, M5Stack Core2), however as I said, it works correctly at the same clock speed (4MHz) with the same SD card (all 3 of them) with Platformio, so I am not entirely sure if it might be electrical issue - it's the same power source, with the same PMIC configuration. I am trying to see if this esp32-sdhc-slot device will work, but configuration of device trees gives me problems when it comes to gpio definition. Whatever I try from what I see in this thread it complains that it wants number or parenthesis expression in the pinmux line.

EricNRS, 2024-07-30:
@domints You can share your DTS settings in the espressif channel in Discord for assistance (https://discord.com/channels/720317445772017664/883444902971727882).

As for the M5Stack Core2, [the v1.0 schematics](https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/schematic/Core/CORE2_V1.0_SCH.pdf) show that the SDCard slot only works in SPI mode and there are no series termination resistors (typically 22 ohms, but depends upon board impedance).  The[ v1.1 schematics](https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/products/core/Core2%20v1.1/Sch_Core2_v1.1_2024-02-22.pdf) show 33 ohm resistors in series, but it is still just SPI mode.

V1.0 schematics - expect errors IMHO due to lack of series resistors:
[image]

v1.1 schematics - has series resistors, but no pull-ups, so you need to make sure the card is inserted during boot.  It is still just 1-bit SPI mode:
[image]

For what it is worth, I did all of my testing for the standalone SPI test at 24 MHz: `spi-max-frequency = <24000000>;`

sylvioalves, 2024-09-16:
Fixed in #[75725](https://github.com/zephyrproject-rtos/zephyr/pull/75725)

closed 2024-09-16 by sylvioalves