zephyrproject-rtos/zephyr #75584: drivers: stm32_qdec: timers 2 and 5 does not count pulses

jeppenodgaard, 2024-07-08:
**Describe the bug**
Using timers 2 or 5 as decoders does not seem to work.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? nucleo_h563zi.
 - What have you tried to diagnose or workaround this issue? No.
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit? I do not know.

Only channel 1 and 2 works as decoders which has been chosen.

**To Reproduce**
Steps to reproduce the behavior:
1. Create `samples/sensor/qdec/boards/nucleo_h563zi.overlay`:
```c
#include <zephyr/dt-bindings/sensor/qdec_stm32.h>

/ {
        aliases {
                qenca = &phase_a;
                qencb = &phase_b;
                qdec0 = &qdec5;
        };

        encoder-emulate {
                compatible = "gpio-leds";
                phase_a: phase_a {
                        gpios = <&gpiog 4 GPIO_ACTIVE_HIGH>;
                };
                phase_b: phase_b {
                        gpios = <&gpiob 0 GPIO_ACTIVE_HIGH>; /* CN10.31 */
                };
        };
};

/* Disable to use pa0 */
&pwm3 {
        status = "disabled";
};

&timers2 {
        status = "okay";
        st,prescaler = <1>;
        qdec2: qdec2 {
                compatible = "st,stm32-qdec";
                status = "okay";
                // pinctrl-0 = <&tim2_ch1_pa0>; /* CN10.29: does not work */
                pinctrl-0 = <&tim2_ch1_pa15>; /* CN7.9: does not work */
                pinctrl-names = "default";
                st,encoder-mode = <0x10003>;
                st,input-polarity-inverted;
                st,counts-per-revolution = <360>;
        };
};

&timers3 {
        status = "okay";
        st,prescaler = <1>;
        qdec3: qdec3 {
                compatible = "st,stm32-qdec";
                status = "okay";
                pinctrl-0 = <&tim3_ch2_pc7>; /* CN7.11: works */
                pinctrl-names = "default";
                st,encoder-mode = <0x10003>;
                st,input-polarity-inverted;
                st,counts-per-revolution = <360>;
        };
};

&timers4 {
        status = "okay";
        st,prescaler = <1>;
        qdec4: qdec4 {
                compatible = "st,stm32-qdec";
                status = "okay";
		pinctrl-0 = <&tim4_ch2_pd13>; /* CN10.19: works */
                pinctrl-names = "default";
                st,encoder-mode = <0x10003>;
                st,input-polarity-inverted;
                st,counts-per-revolution = <360>;
        };
};

&timers5 {
        status = "okay";
        st,prescaler = <1>;
        qdec5: qdec5 {
                compatible = "st,stm32-qdec";
                status = "okay";
                pinctrl-0 = <&tim5_ch1_pa0>; /* CN10.29: does not work */
                pinctrl-names = "default";
                st,encoder-mode = <0x10003>;
                st,input-polarity-inverted;
                st,counts-per-revolution = <360>;
        };
};
```
2. `west build -p auto -b nucleo_h563zi samples/sensor/qdec`
3. `west flash`
4. Connect CN10.31 with the qdec input CN10.29.
5. Verify that the value printed does not change.
6. Change alias `qdec0` to another qdec node and redo steps 2-5.

**Expected behavior**
Timers 2 and 5 decoders count pulses.

**Impact**
Decoder does not work for timer 2 and 5 and thus some inputs cannot be used as decoders.

**Logs and console output**
Timer 4 (working):
```
*** Booting Zephyr OS build v3.7.0-rc2-71-ga6e672f9b236 ***
Quadrature decoder sensor test
Quadrature encoder emulator enabled with 100 ms period
Position = 5 degrees
Position = 10 degrees
Position = 15 degrees
```
Timer 5 (not working):
```
*** Booting Zephyr OS build v3.7.0-rc2-71-ga6e672f9b236 ***
Quadrature decoder sensor test
Quadrature encoder emulator enabled with 100 ms period
Position = 0 degrees
Position = 0 degrees
Position = 0 degrees
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.16.5
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

FRASTM, 2024-08-29:
Timer 2 & 5 of the stm32h573 are 32-bit GP timers  while timers 3 & 4 are 16bit. Could this be a part of the problem ?

jeppenodgaard, 2024-09-06:
`st,encoder-mode = <0x10003>;` only works with timer channel 2 (tim_ti2):

> **Clock plus direction encoder mode**
In addition to the quadrature encoder mode, the timer offers support other types of
encoders.
In the “clock plus direction” mode shown on Figure 477, the clock is provided on a single
line, on tim_ti2, while the direction is forced using the tim_ti1 input.

Due to my pinctrl values in the overlay only channel 2 was tested for timers 3 and 4 and channel 1 for timers 2 and 5.