zephyrproject-rtos/zephyr #75603: PPP/LCP: LCP statemachine stuck after link termination

ljdevdev, 2024-07-08:
I have a curious observation, and I am not quite sure, if I am using the driver wrong, or I encountered a bug. The LCP-Statemachine gets stuck after calling net_if_down on the PPP interface. Do I need to reset the driver somehow? 

**Obeserved behaviour**
I am running a custom AT-Command modem driver that hands over to the Zephyr-PPP driver after successfully establishing a mobile link. 

When the modem terminates the PPP link (see log below) the PPP-driver switches to phase `PPP_DEAD` and the LCP statemachine switches to  `PPP_STOPPED`

The AT-Command driver then takes over, calls `net_if_down` , reestablishes the mobile connection and calls `net_if_up` on the ppp-interface.

The LCP-Statemachine then gets stuck in state `PPP_STOPPED` and never establishes a new link. 

**Solution** 
The problem goes away if I reset  `lcp.fsm.state` to `PPP_INITIAL` in `ppp_link_terminated` (https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/l2/ppp/link.c#L115):

```c
void ppp_link_terminated(struct ppp_context *ctx)
{
        k_sem_give(&ctx->wait_ppp_link_terminated);

        if (ctx->phase == PPP_DEAD) {
                return;
        }

        /* TODO: cleanup things etc here if needed */

        ppp_change_phase(ctx, PPP_DEAD);
/*+*/   ppp_change_state(&ctx->lcp.fsm, PPP_INITIAL);

        NET_DBG("[%p] Link terminated", ctx);
}
```

I am unsure though, if this is the proper place or way to reset the LCP statemachine.

**To Reproduce**
Unfortuantely I have no easy way to reproduce this without a modem that terminates the connection. If someone points me to a testsuite where I could add this as a test, I would be happy to do so.

**Expected behavior**
The statemachine should start from the same initial state each time ppp_start is called.

**Impact**
This bug makes the PPP driver unusable in a commercial product

**Logs and console output**
```

[00:05:13.538,000] <dbg> net_sock: zsock_received_cb: (rx_q[0]): ctx=0x20205e88, pkt=0x20274270, st=0, user_data=(nil)
[00:05:14.261,000] <dbg> net_ppp: ppp_input_byte: Address byte (0xff) start
[00:05:14.265,000] <dbg> net_l2_ppp: ppp_fsm_input: (rx_q[0]): [LCP/0x20207350] LCP Terminate-Req (5) id 3 payload len 28
[00:05:14.265,000] <dbg> net_l2_ppp: fsm_recv_terminate_req: (rx_q[0]): [LCP/0x20207350] Current state OPENED (9)
[00:05:14.265,000] <dbg> net_l2_ppp: fsm_recv_terminate_req: (rx_q[0]): [LCP/0x20207350] Terminated by peer (No network protocols running)
[00:05:14.265,000] <dbg> net_l2_ppp: ppp_change_state_debug: (rx_q[0]): [LCP/0x20207350] state OPENED (9) => STOPPING (5) (fsm_recv_terminate_req():877)
[00:05:14.265,000] <dbg> net_l2_ppp: ppp_fsm_lower_down: (rx_q[0]): [IPCP/0x20207420] Current state OPENED (9)
[00:05:14.265,000] <dbg> net_l2_ppp: ppp_change_state_debug: (rx_q[0]): [IPCP/0x20207420] state OPENED (9) => STARTING (1) (ppp_fsm_lower_down():282)
[00:05:14.265,000] <dbg> net_if: net_if_ipv4_addr_rm: (rx_q[0]): [0] interface 0x202030e8 address 100.84.56.8 removed
[00:05:14.265,000] <dbg> net_l2_ppp: ppp_change_phase_debug: (rx_q[0]): [0x20207318] phase RUNNING (4) => TERMINATE (5) (ppp_network_down():37)
[00:05:14.265,000] <dbg> net_l2_ppp: ppp_network_down: (rx_q[0]): [0x20207318] Proto IPv4 (0x0021) down (0)
[00:05:14.265,000] <dbg> net_l2_ppp: ppp_fsm_close: (rx_q[0]): [IPCP/0x20207420] Current state STARTING (1)
[00:05:14.265,000] <dbg> net_l2_ppp: ppp_change_state_debug: (rx_q[0]): [IPCP/0x20207420] state STARTING (1) => INITIAL (0) (ppp_fsm_close():240)
[00:05:14.265,000] <dbg> net_l2_ppp: ppp_fsm_lower_down: (rx_q[0]): [IPV6CP/0x202074f8] Current state ACK_RECEIVED (7)
[00:05:14.265,000] <dbg> net_l2_ppp: ppp_change_state_debug: (rx_q[0]): [IPV6CP/0x202074f8] state ACK_RECEIVED (7) => STARTING (1) (ppp_fsm_lower_down():268)
[00:05:14.265,000] <dbg> net_l2_ppp: ppp_fsm_close: (rx_q[0]): [IPV6CP/0x202074f8] Current state STARTING (1)
[00:05:14.265,000] <dbg> net_l2_ppp: ppp_change_state_debug: (rx_q[0]): [IPV6CP/0x202074f8] state STARTING (1) => INITIAL (0) (ppp_fsm_close():240)
[00:05:14.265,000] <dbg> net_l2_ppp: ppp_fsm_lower_down: (rx_q[0]): [PAP/0x202075c0] Current state CLOSED (2)
[00:05:14.265,000] <dbg> net_l2_ppp: ppp_change_state_debug: (rx_q[0]): [PAP/0x202075c0] state CLOSED (2) => INITIAL (0) (ppp_fsm_lower_down():273)
[00:05:14.265,000] <dbg> net_l2_ppp: ppp_change_phase_debug: (rx_q[0]): [0x20207318] phase TERMINATE (5) => DEAD (0) (ppp_link_down():133)
[00:05:14.265,000] <dbg> net_l2_ppp: ppp_change_phase_debug: (rx_q[0]): [0x20207318] phase DEAD (0) => ESTABLISH (1) (lcp_down():163)
[00:05:14.266,000] <dbg> net_l2_ppp: ppp_send_pkt: (rx_q[0]): [LCP/0x20207350] Sending 6 bytes pkt 0x20273390 (options len 0)
[00:05:14.266,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20273390, prio 1) network packet iface 0x202030e8/2
[00:05:14.266,000] <dbg> net_l2_ppp: net_pkt_hexdump: send L2
[... 134 lines cut ...]
LCP ACK received    : yes
IPCP state          : INITIAL (0)
IPCP retransmits    : 10
IPCP NACK loops     : 0
IPCP NACKs recv     : 0
IPCP current id     : 2
IPCP ACK received   : yes
IPv6CP state        : INITIAL (0)
IPv6CP retransmits  : 10
IPv6CP NACK loops   : 0
IPv6CP NACKs recv   : 0
IPv6CP current id   : 1
IPv6CP ACK received : yes
PAP state           : INITIAL (0)
PAP retransmits     : 0
PAP NACK loops      : 0
PAP NACKs recv      : 0
PAP current id      : 0
PAP ACK received    : no
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 16.1
 - Zephyr 3.6

rlubos, 2024-07-09:
One thing that strucks me is that the driver shouldn't really take control over the network interface admin state (I. .e it should not use `net_if_up()`/`net_if_down()`), but rather control CARRIER/DORMANT flags (`net_if_carrier/dormant_on/off()`), see https://docs.zephyrproject.org/latest/connectivity/networking/api/net_if.html#network-interface-state-management.

Nonetheless, I'm not sure if it's correct that on `net_if_down()` we skip LCP state transitions if PPP is in DEAD phase:
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/l2/ppp/ppp_l2.c#L258
That's why LCP ends up stuck in STOPPED state.
@bjarki-andreasen @tomi-font You're more experienced with PPP interfaces, what's your view on this?

bjarki-andreasen, 2024-07-09:
It seems you are cutting the carrier right after calling `net_if_down()` which can cause the network interface to get stuck, as it tries to terminate the link properly. Your driver should be calling `net_if_carrier_off()` when the data connection is closed, which immediately resets the internal state machine as attempting to close down properly is not possible :) `net_if_up()` should be used by the application to signal that the interface should be up once it is possible, which is after your modem has registered to a network and the PPP link is up, at which point your driver should call `net_if_carrier_on()`

ljdevdev, 2024-07-09:
Thank you very much for your swift reply :) I'll check that that's working and then close the issue.

ljdevdev, 2024-07-15:
@bjarki-andreasen , I tried what you suggested (actually multiple combinations, since I wasn't exactly sure about the necessary order and timing between the calls). 

My current setup is as follows:

0. Boot with CONFIG_PPP_NET_IF_NO_AUTO_START=y
    - Because it seems that I need to call `net_if_up()` to properly set up the PPP uart
1. Set up mobile link
2. Call `net_if_up()` (otherwise the uart callback is not set and no PPP responses are received by the driver)
3. Call `net_if_carrier_on()`
4. IP Communication + Link failure
5. Call `net_if_carrier_off()`
6. Reconnect
7. go to (2.)

This produces the exact same behavior as before (see log below)

Now I dug a bit into the code and found that `net_if_carrier_on()`, as well as the functions for setting up the PPP uart are called from inside `net_if_up`:

From  `ppp_start` (https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/net/ppp.c#L1098) 
```c
		ppp_async_uart_rx_enable(context);
#else
		uart_irq_rx_disable(context->dev);
		uart_irq_tx_disable(context->dev);
		ppp_uart_flush(context->dev);
		uart_irq_callback_user_data_set(context->dev, ppp_uart_isr,
						context);
		uart_irq_rx_enable(context->dev);
#endif
	}
#endif /* !CONFIG_NET_TEST */

	net_if_carrier_on(context->iface);
	return 0;
```
This is called in `ppp_up` (https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/l2/ppp/ppp_l2.c#L211) which is called in `ppp_enable`(https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/l2/ppp/ppp_l2.c#L293) which is called in `net_if_up`. 

So it is clear, that `net_if_up` must be called after the mobile link was set up and a `net_if_carrier_on` call seems redundant (in fact it works fine without an additional call to that function). This may not be the canonic way, but it seems to be the way the driver is built.

As for the call to `net_if_carrier_off()`, it does not change the state of the LCP fsm and therefore I run into the same problem again.

## Log:
[00:05:20.994,000] <dbg> net_l2_ppp: fsm_recv_terminate_req: (rx_q[0]): [LCP/0x20207350] Current state STOPPING (5)
[00:05:20.994,000] <dbg> net_l2_ppp: fsm_recv_terminate_req: (rx_q[0]): [LCP/0x20207350] Invalid state STOPPING (5)
[00:05:21.001,000] <dbg> net_ppp: ppp_input_byte: Invalid (0x0d) byte, expecting Address
[00:05:21.001,000] <dbg> net_ppp: ppp_change_state: (ppp_workq): [0x202076f8] state ADDRESS (1) => START (0)
[00:05:24.990,000] <dbg> net_l2_ppp: ppp_fsm_timeout: (sysworkq): [LCP/0x20207350] Current state STOPPING (5)
[00:05:24.990,000] <dbg> net_l2_ppp: ppp_change_state_debug: (sysworkq): [LCP/0x20207350] state STOPPING (5) => STOPPED (3) (ppp_fsm_timeout():136)
[00:05:24.990,000] <dbg> net_l2_ppp: ppp_change_phase_debug: (sysworkq): [0x20207318] phase ESTABLISH (1) => DEAD (0) (ppp_link_terminated():117)
[00:05:24.990,000] <dbg> net_l2_ppp: ppp_link_terminated: (sysworkq): [0x20207318] Link terminated
**[00:05:25.036,000] <dbg> net_if: update_operational_state: (seModemDriverAppRun): iface 0x202030e8, oper state DOWN admin UP carrier OFF dormant OFF**
[00:05:29.286,000] <dbg> net_if: net_if_ipv6_prefix_set_timer: (rx_q[0]): Prefix lifetime 9250560 sec
[00:05:29.287,000] <dbg> net_if: net_if_ipv6_addr_update_lifetime: (rx_q[0]): Updating expire time of fd26:52ff:a04d:1:4:9fff:fe91:15b1 by 9250560 secs
[00:05:29.287,000] <dbg> net_if: net_if_ipv6_prefix_set_timer: (rx_q[0]): Prefix lifetime 9250560 sec
[00:05:29.287,000] <dbg> net_if: net_if_ipv6_router_update_lifetime: (rx_q[0]): Updating expire time of fe80::21d:aaff:fef7:640 by 1800 secs
[00:05:29.287,000] <dbg> net_if: net_if_stop_rs: (rx_q[0]): Stopping ND/RS for iface 0x20203068
[00:05:29.287,000] <dbg> net_if: address_lifetime_timeout: (sysworkq): Waiting for 2147483647 ms
[00:05:32.756,000] modem                   DEBUG: --> AT+CEREG?
[00:05:34.757,000] modem                   DEBUG: <--  
[00:05:34.808,000] modem                   DEBUG: CEREG Command failed.
[00:05:35.819,000] modem                   DEBUG: --> AT+CEREG?
[00:05:35.838,000] modem                   DEBUG: <-- AT+CEREG?  /+CEREG: 0,0 / /OK /
[00:05:36.878,000] <dbg> net_if: net_if_tx: (rx_q[0]): Processing (pkt 0x202733d4, prio 1) network packet iface 0x20203068/1
[00:05:36.879,000] <dbg> net_if: net_if_tx: (rx_q[0]): Processing (pkt 0x202733d4, prio 1) network packet iface 0x20203068/1
[00:05:36.879,000] <dbg> net_if: net_if_tx: (rx_q[0]): Processing (pkt 0x202733d4, prio 1) network packet iface 0x20203068/1
[00:05:36.899,000] modem                   DEBUG: --> AT+CEREG?
[00:05:36.905,000] modem                   DEBUG: <-- AT+CEREG?  /+CEREG: 0,0 / /OK /
[00:05:37.406,000] <dbg> net_if: net_if_ipv6_prefix_set_timer: (rx_q[0]): Prefix lifetime 9250560 sec
[00:05:37.406,000] <dbg> net_if: net_if_ipv6_addr_update_lifetime: (rx_q[0]): Updating expire time of fd26:52ff:a04d:1:4:9fff:fe91:15b1 by 9250560 secs
[00:05:37.406,000] <dbg> net_if: net_if_ipv6_prefix_set_timer: (rx_q[0]): Prefix lifetime 9250560 sec
[... 23 lines cut ...]
[00:05:40.924,000] <dbg> net_if: net_if_stop_rs: (rx_q[0]): Stopping ND/RS for iface 0x20203068
[00:05:40.924,000] <dbg> net_if: address_lifetime_timeout: (sysworkq): Waiting for 2147483647 ms
[00:05:42.906,000] <dbg> net_if: net_if_tx: (rx_q[0]): Processing (pkt 0x202733d4, prio 1) network packet iface 0x20203068/1
[00:05:42.906,000] <dbg> net_if: net_if_tx: (rx_q[0]): Processing (pkt 0x202733d4, prio 1) network packet iface 0x20203068/1
[... cut ...]
[00:05:43.961,000] <dbg> net_l2_ppp: ppp_fsm_lower_up: (net_mgmt): [LCP/0x20207350] Current state STOPPED (3)
**[00:05:43.961,000] <dbg> net_l2_ppp: ppp_fsm_lower_up: (net_mgmt): [LCP/0x20207350] Invalid state STOPPED (3)**
[00:05:43.961,000] <dbg> net_l2_ppp: ppp_fsm_open: (net_mgmt): [LCP/0x20207350] Current state STOPPED (3)
[00:05:43.961,000] modem                   DEBUG: PPP interface up. Waiting for IP-address.
[00:05:43.962,000] modem                    INFO: PPP device IP address: 100.84.56.8
[00:05:43.963,000] modem-performance-test  WARNING: Socket bind failed

bjarki-andreasen, 2024-07-15:
@ljdevdev This is a larger issue, where I am very much in opposition to the way `ppp.c` works, IMO misusing the carrier... See
* #67627 
* #67731

The modem subsystem, with `modem_cellular.c` uses the interface and carrier appropriately, you may want to look into using it, or at least adapting your driver to behave similarly :)

ljdevdev, 2024-07-16:
@bjarki-andreasen I am fairly certain, that all of this has nothing to do with the way I set up the PPP conection (carrier vs net_if vs modem_cellular.c etc).

If I look at the actual LCP communication happening here I get the following:

The modem sends a terminate request (05)

```
[00:05:18.371,000] <dbg> net_l2_ppp: net_pkt_hexdump: 0x2027400c
                                     c0 21 05 03 00 20 4e 6f  20 6e 65 74 77 6f 72 6b |.!... No  network
                                     20 70 72 6f 74 6f 63 6f  6c 73 20 72 75 6e 6e 69 | protoco ls runni
                                     6e 67                                            |ng       
[00:05:18.371,000] <dbg> net_l2_ppp: ppp_fsm_input: (rx_q[0]): [LCP/0x20207350] LCP Terminate-Req (5) id 3 payload len 28
[00:05:18.372,000] <dbg> net_l2_ppp: fsm_recv_terminate_req: (rx_q[0]): [LCP/0x20207350] Current state OPENED (9)
[00:05:18.372,000] <dbg> net_l2_ppp: fsm_recv_terminate_req: (rx_q[0]): [LCP/0x20207350] Terminated by peer (No network protocols running)
[00:05:18.372,000] <dbg> net_l2_ppp: ppp_change_state_debug: (rx_q[0]): [LCP/0x20207350] state OPENED (9) => STOPPING (5) (fsm_recv_terminate_req():877)

```

The ppp driver sends a terminate ack (06):
```
[00:05:18.373,000] <dbg> net_l2_ppp: net_pkt_hexdump: send L2
[00:05:18.373,000] <dbg> net_l2_ppp: net_pkt_hexdump: 0x2027345c
                                     c0 21 06 03 00 04                                |.!....    
```

The lcp fsm then runs into a timeout, which puts it into mode STOPPED
```
[00:05:23.372,000] <dbg> net_l2_ppp: ppp_fsm_timeout: (sysworkq): [LCP/0x20207350] Current state STOPPING (5)
[00:05:23.373,000] <dbg> net_l2_ppp: ppp_change_state_debug: (sysworkq): [LCP/0x20207350] state STOPPING (5) => STOPPED (3) (ppp_fsm_timeout():136)
```

Now I guess if I  call `net_if_carrier_on()` in this state, that may or may not be fine. This is what modem_cellular.c does, and that works because it uses cmux (I think). I don't use cmux (yet) and so I need to pass controll over the uart back and forth.

What did not work was calling `net_if_down()` after link termination and then `net_if_up()` again. And that seems to be something, that an application might want to do.

So if an application asynchronously calls `net_if_down()` between a  terminate request and reestablishment of the link, I think the driver would still freeze.

ljdevdev, 2024-07-16:
For the time being I found an ugly hack, to get this working for my driver without dragging a manual patch of a zephyr along:

In the long run I think it would be great to have a PPP driver in zephyr that is usable with and without cmux and with and without any given cellular driver. 

If there is any interest in pursuing this further I am happy to help, otherwise I think we can close this.

carlescufi, 2024-09-04:
Thanks @ljdevdev closing this as you suggest.

closed 2024-09-04 by carlescufi