zephyrproject-rtos/zephyr #75614: BLE Mesh Sample using HCI SPI controller

matt-wood-ct, 2024-07-08:
**Notes (delete this)**
Github Discussions (https://github.com/zephyrproject-rtos/zephyr/discussions)
are available to first verify that the issue is a genuine Zephyr bug and not a
consequence of Zephyr services misuse.

This issue list is only for bugs in the main Zephyr code base
(https://github.com/zephyrproject-rtos/zephyr/). If the bug is for a project
fork (such as NCS) specific feature, please open an issue in the fork project
instead.

**Describe the bug**
So I am attempting to validate using a NRF52832 SOC as an external BLE controller for use in a BLE Mesh application and I am struggling to get BLE Mesh to work using a BLE controller connected via SPI.

**Test setup**
I have two NRF52DK kits with the following jumper connections:
| Host | Controller | Function |
| -------- | ------- | ------- |
| P0.03 | P0.21*   | Reset |
| P0.04 | P0.04   | IRQ |
| P0.28    |  P0.28 | SPI_SCK |
| P0.29    |  P0.29 | SPI_MOSI |
| P0.30    |  P0.30 | SPI_MISO |
| P0.31    |  P0.31 | SPI_CS |

\* P0.21 functions as the HW RST on these dev kits

**To Reproduce**
[Using_SPI_HCI_for_mesh_and_beacon_samples.patch](https://github.com/user-attachments/files/16132505/Using_SPI_HCI_for_mesh_and_beacon_samples.patch)

1. Add custom board overlay files to the `hci_spi` and `mesh` samples (see attached patch file)
2. Adjust prj.conf files as per the [bluetooth architecture pages instructions for HCI](https://docs.zephyrproject.org/latest/connectivity/bluetooth/bluetooth-arch.html)  (see attached patch file)
3. Build and flash as normal

**Expected behavior**
Mesh sample to run as it does with a combined host and controller build on a single dev kit

**Impact**

Showstopper for using mesh on an external BLE controller

**Logs and console output**
Host:
```
*** Booting Zephyr OS build v3.7.0-rc2-188-gdcfc3e787254 ***
Initializing...
Bluetooth initialized
Mesh initialized
```
Controller (log level increased as it ordinarily would say nothing):
```
*** Booting Zephyr OS build v3.7.0-rc2-188-gdcfc3e787254 ***
[00:00:00.269,744] <dbg> bt_ctlr_hci_driver: hci_driver_send: enter
--- 46 messages dropped ---
[00:00:00.281,707] <dbg> bt_ctlr_hci_driver: cmd_handle: Replying with event of 14 bytes
--- 57 messages dropped ---
[00:00:00.289,672] <dbg> bt_ctlr_hci_driver: cmd_handle: Replying with event of 6 bytes
[00:00:00.289,703] <dbg> bt_hci_raw: bt_hci_recv: buf 0x200093f8 len 6
[00:00:00.289,733] <dbg> bt_ctlr_hci_driver: hci_driver_send: exit: 0
[00:00:00.289,794] <dbg> hci_spi: spi_send: buf 0x200093f8 type 1 len 6
[00:00:00.290,679] <dbg> hci_spi: bt_tx_thread: buf 0x20009018 type 0 len 3
[00:00:00.290,710] <dbg> bt_hci_raw: bt_send: buf 0x20009018 len 3
[00:00:00.290,710] <dbg> bt_ctlr_hci_driver: hci_driver_send: enter
[00:00:00.290,740] <dbg> bt_ctlr_hci: le_read_supp_states: states: 0x000003ffff3fffff
[00:00:00.290,771] <dbg> bt_ctlr_hci_driver: cmd_handle: Replying with event of 14 bytes
[00:00:00.290,802] <dbg> bt_hci_raw: bt_hci_recv: buf 0x200093f8 len 14
[00:00:00.290,832] <dbg> bt_ctlr_hci_driver: hci_driver_send: exit: 0
[00:00:00.290,863] <dbg> hci_spi: spi_send: buf 0x200093f8 type 1 len 14
[00:00:00.291,839] <dbg> hci_spi: bt_tx_thread: buf 0x20009018 type 0 len 11
[00:00:00.291,870] <dbg> bt_hci_raw: bt_send: buf 0x20009018 len 11
[00:00:00.291,870] <dbg> bt_ctlr_hci_driver: hci_driver_send: enter
[00:00:00.291,900] <dbg> bt_ctlr_hci_driver: cmd_handle: Replying with event of 6 bytes
[00:00:00.291,931] <dbg> bt_hci_raw: bt_hci_recv: buf 0x200093f8 len 6
[00:00:00.291,961] <dbg> bt_ctlr_hci_driver: hci_driver_send: exit: 0
[00:00:00.292,022] <dbg> hci_spi: spi_send: buf 0x200093f8 type 1 len 6
[00:00:00.292,938] <dbg> hci_spi: bt_tx_thread: buf 0x20009018 type 0 len 11
[00:00:00.292,968] <dbg> bt_hci_raw: bt_send: buf 0x20009018 len 11
[00:00:00.292,968] <dbg> bt_ctlr_hci_driver: hci_driver_send: enter
[00:00:00.292,999] <dbg> bt_ctlr_hci_driver: cmd_handle: Replying with event of 6 bytes
[00:00:00.293,029] <dbg> bt_hci_raw: bt_hci_recv: buf 0x200093f8 len 6
[00:00:00.293,060] <dbg> bt_ctlr_hci_driver: hci_driver_send: exit: 0
[00:00:00.293,121] <dbg> hci_spi: spi_send: buf 0x200093f8 type 1 len 6
[00:00:00.566,986] <dbg> hci_spi: bt_tx_thread: buf 0x20009018 type 0 len 3
[00:00:00.567,016] <dbg> bt_hci_raw: bt_send: buf 0x20009018 len 3
[00:00:00.567,016] <dbg> bt_ctlr_hci_driver: hci_driver_send: enter
[00:00:00.567,047] <dbg> bt_ctlr_hci_driver: cmd_handle: Replying with event of 12 bytes
[00:00:00.567,047] <dbg> bt_hci_raw: bt_hci_recv: buf 0x200093f8 len 12
[00:00:00.567,077] <dbg> bt_ctlr_hci_driver: hci_driver_send: exit: 0
[00:00:00.567,138] <dbg> hci_spi: spi_send: buf 0x200093f8 type 1 len 12
[00:00:00.572,967] <dbg> hci_spi: bt_tx_thread: buf 0x20009018 type 0 len 28
[00:00:00.572,967] <dbg> bt_hci_raw: bt_send: buf 0x20009018 len 28
[00:00:00.572,998] <dbg> bt_ctlr_hci_driver: hci_driver_send: enter
[00:00:00.573,028] <dbg> bt_ctlr_hci_driver: cmd_handle: Replying with event of 6 bytes
[00:00:00.573,028] <dbg> bt_hci_raw: bt_hci_recv: buf 0x200093f8 len 6
[00:00:00.573,059] <dbg> bt_ctlr_hci_driver: hci_driver_send: exit: 0
[00:00:00.573,120] <dbg> hci_spi: spi_send: buf 0x200093f8 type 1 len 6
```

**Environment (please complete the following information):**
 - Windows
 - Zephyr SDK 0.16.3
 - Zephyr dcfc3e7872542038abc51da7d6a468ac76e71613 (main at time of posting)

**Additional context**
I did validate my device tree and configurations using a simple BLE beacon sample, this operated without fault and proved the connections are correct.

aescolar, 2024-07-09:
CC @jhedberg

cvinayak, 2024-07-10:
@matt-wood-ct  Please add the below to `prj.conf` of your hci_spi sample, so that it periodically logs the thread program stack usage (also, acts as a heart beat assuring the Controller is alive).

```
CONFIG_THREAD_ANALYZER=y
CONFIG_THREAD_ANALYZER_AUTO=y
CONFIG_THREAD_ANALYZER_AUTO_INTERVAL=5
CONFIG_THREAD_NAME=y
```

Do provide the log here...

matt-wood-ct, 2024-07-10:
> @matt-wood-ct Please add the below to `prj.conf` of your hci_spi sample, so that it periodically logs the thread program stack usage (also, acts as a heart beat assuring the Controller is alive). [...]

Added that config to both side, just in case, can confirm both sides are running
See logs below:

<details>
  <summary>Host:</summary>

Log:
  

```
*** Booting Zephyr OS build v3.7.0-rc2-188-gdcfc3e787254 ***
Initializing...
Thread analyze:
 BT RX WQ            : STACK: unused 2448 usage 176 / 2624 (6 %); CPU: 0 %
      : Total CPU cycles used: 0
 thread_analyzer     : STACK: unused 544 usage 480 / 1024 (46 %); CPU: 44 %
      : Total CPU cycles used: 369
 BT LW WQ            : STACK: unused 1168 usage 176 / 1344 (13 %); CPU: 0 %
      : Total CPU cycles used: 0
 sysworkq            : STACK: unused 1872 usage 176 / 2048 (8 %); CPU: 0 %
      : Total CPU cycles used: 1
 idle                : STACK: unused 272 usage 48 / 320 (15 %); CPU: 0 %
      : Total CPU cycles used: 0
 main                : STACK: unused 696 usage 328 / 1024 (32 %); CPU: 21 %
      : Total CPU cycles used: 454
 ISR0                : STACK: unused 1824 usage 224 / 2048 (10 %)
Bluetooth initialized
Mesh initialized
Thread analyze:
 BT Mesh settings workq: STACK: unused 720 usage 176 / 896 (19 %); CPU: 0 %
      : Total CPU cycles used: 0
 0x200019a0          : STACK: unused 296 usage 344 / 640 (53 %); CPU: 0 %
      : Total CPU cycles used: 153
 BT RX WQ            : STACK: unused 2448 usage 176 / 2624 (6 %); CPU: 0 %
      : Total CPU cycles used: 0
 thread_analyzer     : STACK: unused 544 usage 480 / 1024 (46 %); CPU: 1 %
      : Total CPU cycles used: 3156
 BT LW WQ            : STACK: unused 640 usage 704 / 1344 (52 %); CPU: 0 %
      : Total CPU cycles used: 71
 sysworkq            : STACK: unused 1040 usage 1008 / 2048 (49 %); CPU: 5 %
      : Total CPU cycles used: 9507
 idle                : STACK: unused 256 usage 64 / 320 (20 %); CPU: 91 %
      : Total CPU cycles used: 153933
 ISR0                : STACK: unused 1824 usage 224 / 2048 (10 %)
Thread analyze:
 BT Mesh settings workq: STACK: unused 720 usage 176 / 896 (19 %); CPU: 0 %
      : Total CPU cycles used: 0
 0x200019a0          : STACK: unused 296 usage 344 / 640 (53 %); CPU: 0 %
      : Total CPU cycles used: 153
 BT RX WQ            : STACK: unused 2448 usage 176 / 2624 (6 %); CPU: 0 %
      : Total CPU cycles used: 0
 thread_analyzer     : STACK: unused 544 usage 480 / 1024 (46 %); CPU: 1 %
      : Total CPU cycles used: 5643
 BT LW WQ            : STACK: unused 640 usage 704 / 1344 (52 %); CPU: 0 %
      : Total CPU cycles used: 71
 sysworkq            : STACK: unused 1040 usage 1008 / 2048 (49 %); CPU: 2 %
      : Total CPU cycles used: 9507
 idle                : STACK: unused 256 usage 64 / 320 (20 %); CPU: 94 %
      : Total CPU cycles used: 317774
 ISR0                : STACK: unused 1824 usage 224 / 2048 (10 %)
Thread analyze:
 BT Mesh settings workq: STACK: unused 720 usage 176 / 896 (19 %); CPU: 0 %
      : Total CPU cycles used: 0
 0x200019a0          : STACK: unused 296 usage 344 / 640 (53 %); CPU: 0 %
      : Total CPU cycles used: 153
 BT RX WQ            : STACK: unused 2448 usage 176 / 2624 (6 %); CPU: 0 %
      : Total CPU cycles used: 0
 thread_analyzer     : STACK: unused 544 usage 480 / 1024 (46 %); CPU: 1 %
[... 185 lines cut ...]
 main                : STACK: unused 32 usage 480 / 512 (93 %); CPU: 0 %
      : Total CPU cycles used: 106
 ISR0                : STACK: unused 1820 usage 228 / 2048 (11 %)
Thread analyze:
 bt_tx_thread        : STACK: unused 512 usage 512 / 1024 (50 %); CPU: 0 %
      : Total CPU cycles used: 181
 BT RX               : STACK: unused 408 usage 360 / 768 (46 %); CPU: 0 %
      : Total CPU cycles used: 2
 BT RX pri           : STACK: unused 96 usage 352 / 448 (78 %); CPU: 0 %
      : Total CPU cycles used: 1
 thread_analyzer     : STACK: unused 552 usage 472 / 1024 (46 %); CPU: 0 %
      : Total CPU cycles used: 298
 sysworkq            : STACK: unused 848 usage 176 / 1024 (17 %); CPU: 0 %
      : Total CPU cycles used: 1
 logging             : STACK: unused 128 usage 640 / 768 (83 %); CPU: 2 %
      : Total CPU cycles used: 20497
 idle                : STACK: unused 256 usage 64 / 320 (20 %); CPU: 97 %
      : Total CPU cycles used: 801443
 main                : STACK: unused 32 usage 480 / 512 (93 %); CPU: 0 %
      : Total CPU cycles used: 106
 ISR0                : STACK: unused 1820 usage 228 / 2048 (11 %)

```
  
</details>

matt-wood-ct, 2024-07-10:
To make mesh on HCI work do I need `CONFIG_BT_HCI_MESH_EXT` enabled?
I just tried to turn it on but it has a dependency I can't resolve, it wants `CONFIG_BT_LL_SW_SPLIT=n` but that can never be the case because if you are a Bluetooth controller you are required to have a `BT_LL_CHOICE` and `BT_LL_SW_SPLIT` is the only choice...

matt-wood-ct, 2024-07-10:
FYI I did also try using nordics hci_lpuart sample in conjunction with the same beacon and mesh samples configured for H4 HCI rather then SPI on the NRF52832 ICs and found the same result, beacon works, mesh does not appear to work.  
Note, I don't have logs for this since the NRF52832 only has one UART so I had to use that for the interchip comms. In principle you could get RTT up and running but I was just trying to do a quick and dirty test so I used minimal overlays and confs.

PavelVPV, 2024-07-10:
Hi @matt-wood-ct, I'm on a leave atm and can't try the sample at my desk, but the first thing comes up to my mind when trying to build mesh separately of controller is misconfiguration of host and controller. Though IIRC you should have had an error when host and controller are misconfigured, it is worth comparing bluetooth configuration (CONFIG_BT_* Kconfigs from `.config` file in build folder) of single image build (when host and controller are built together) and multi image build (the one you have).

matt-wood-ct, 2024-07-11:
Hi @PavelVPV, thank you for your time,
I have extracted those configs 
and I am going through the differences now, will probably take me a few hours to go through (I'm having a busy day) so in the mean time here are the 3 files.  

[Controller.txt](https://github.com/user-attachments/files/16175833/Controller.txt)
[Host.txt](https://github.com/user-attachments/files/16175834/Host.txt)
[Combined.txt](https://github.com/user-attachments/files/16175836/Combined.txt)

jori-nordic, 2024-07-22:
Controller is missing `CONFIG_BT_EXT_ADV=y`. 
Reproduced the issue on my (and a colleague's) desk. Added config, and it works as you can see from the video. We're able to provision and have two DKs control each other's LED. 
[it-werks.webm](https://github.com/user-attachments/assets/18fc7f84-a097-4fc5-a1b3-2cad67c7710d)

matt-wood-ct, 2024-07-22:
> Controller is missing `CONFIG_BT_EXT_ADV=y`. Reproduced the issue on my (and a colleague's) desk. Added config, and it works as you can see from the video. We're able to provision and have two DKs [...]

@jori-nordic Thank you so much for this I will retest soon to make sure but it looks like you've cracked it 😄 

This is a really important proving exercise for the project I am working on

jori-nordic, 2024-07-22:
No problem :)

Feel free to review and +1 the doc change if the instructions work for you: https://github.com/zephyrproject-rtos/zephyr/pull/76161