zephyrproject-rtos/zephyr #75617: samples/subsys/logging/logger/sample.logger.usermode fails to build for bytesensi_l & bl654_dvk/nrf52840/pa

aescolar, 2024-07-08:
**Describe the bug**
samples/subsys/logging/logger/sample.logger.usermode fails to build for both
* bytesensi_l 
* bl654_dvk/nrf52840/pa

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p bytesensi_l -T samples/subsys/logging/logger -s sample.logger.usermode

**Expected behavior**
No build errors (or platforms filtered out from test)

**Impact**
2 failures in weekly CI

**Logs and console output**
```
warning: LOG_PRINTK (defined at subsys/logging/Kconfig.processing:8) has direct dependencies !USERSPACE && !LOG_MODE_MINIMAL && LOG with value n, but is currently being y-selected by the following symbols:
 - LOG_BACKEND_RTT_FORCE_PRINTK (defined at subsys/logging/backends/Kconfig.rtt:104), with value y, direct dependencies LOG_BACKEND_RTT && !LOG_FRONTEND_ONLY && !LOG_MODE_MINIMAL && LOG (value: y), and select condition LOG_BACKEND_RTT && !LOG_FRONTEND_ONLY && !LOG_MODE_MINIMAL && LOG (value: y)
```

**Environment (please complete the following information):**
 - OS: Linux, Ubuntu 22.04 CI
 - Toolchain (e.g Zephyr SDK, ...): 0.16.8
 - Commit SHA or Version used: main as of now 980d3f4c4f4ef4ec15c5bcd538155a184df6e1b8

aescolar, 2024-07-08:
CC @dariis @rerickson1 @greg-leach

morandg, 2024-07-09:
Same for ``bytesensi_l`` that produces the same error as the other excluded platform.

commit 0c985887 (2024-07-09): samples: logging: disable usermode for some platforms

Those platfroms use rtt as debug port that conflicts with usermode, as
in 27d519b260b25826689a4e5283af8a7f188d121c.

Fixes #75617
closed 2024-07-09 by nashif