zephyrproject-rtos/zephyr #75650: samples/cpp/hello_world: Linkage fails for esp32_devkitc_wroom/esp32/procpu

elazar-vayyar, 2024-07-09:
**Describe the bug**
Building the sample **cpp**/hello_world fails for the Espressif board `esp32_devkitc_wroom/esp32/procpu`. 
The failure is in the linkage stage - see build output below.
Most of the linkage errors are for missing symbols ("undefined reference to...") of file API functions - `open`, `close`, `read`, `write`, etc. 

**The build command** used (run in working directory `zephyr/samples/cpp/hello_world/`):
```
west build -b "esp32_devkitc_wroom/esp32/procpu"
```

**Zephyr code version**: latest commit at hash `dcfc3e7872542038abc51da7d6a468ac76e71613` (Jul 2 2024, 17:02:48).

Note: Building the "regular" (C) `samples/hello_world` works fine for this board.

**To Reproduce**
Steps to reproduce the behavior:
1. `cd zephyr/samples/cpp/hello_world`
2. `west build -b "esp32_devkitc_wroom/esp32/procpu"`

**Expected behavior**
Build should succeed.

**Impact**
One cannot validate that C++ support actually works for the mentioned board (and for ESP32 in general). 
This blocks any C++ application development for ESP32.

**Logs and console output**
```
~/zephyrproject/zephyr/samples/cpp/hello_world$ west build -b "esp32_devkitc_wroom/esp32/procpu"
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/user/zephyrproject/zephyr/samples/cpp/hello_world
-- CMake version: 3.22.1
-- Found Python3: /home/user/zephyrproject/.venv/bin/python3 (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/user/.cache/zephyr
-- Zephyr version: 3.7.0-rc2 (/home/user/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: esp32_devkitc_wroom, qualifiers: esp32/procpu
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.8 (/home/user/zephyr-sdk-0.16.8)
-- Found toolchain: zephyr 0.16.8 (/home/user/zephyr-sdk-0.16.8)
-- Found Dtc: /home/user/zephyr-sdk-0.16.8/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/user/zephyrproject/zephyr/boards/espressif/esp32_devkitc_wroom/esp32_devkitc_wroom_procpu.dts
-- Generated zephyr.dts: /home/user/zephyrproject/zephyr/samples/cpp/hello_world/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/user/zephyrproject/zephyr/samples/cpp/hello_world/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /home/user/zephyrproject/zephyr/samples/cpp/hello_world/build/zephyr/dts.cmake
Parsing /home/user/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/user/zephyrproject/zephyr/boards/espressif/esp32_devkitc_wroom/esp32_devkitc_wroom_procpu_defconfig'
Merged configuration '/home/user/zephyrproject/zephyr/samples/cpp/hello_world/prj.conf'
Configuration saved to '/home/user/zephyrproject/zephyr/samples/cpp/hello_world/build/zephyr/.config'
Kconfig header saved to '/home/user/zephyrproject/zephyr/samples/cpp/hello_world/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: /home/user/zephyr-sdk-0.16.8/xtensa-espressif_esp32_zephyr-elf/xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/user/zephyr-sdk-0.16.8/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-gcc
esptool path: /home/user/zephyrproject/modules/hal/espressif/tools/esptool_py/esptool.py
-- Using ccache: /usr/bin/ccache
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/zephyrproject/zephyr/samples/cpp/hello_world/build
-- west build: building application
[1/198] Preparing syscall dependency handling

[4/198] Generating include/generated/zephyr/version.h
-- Zephyr version: 3.7.0-rc2 (/home/user/zephyrproject/zephyr), build: v3.7.0-rc2-188-gdcfc3e787254
[193/198] Linking CXX executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/user/zephyrproject/zephyr/samples/cpp/hello_world/build/zephyr/zephyr_pre0.map 
: && ccache /home/user/zephyr-sdk-0.16.8/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-g++  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/xtensa/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/user/zephyrproject/zephyr/samples/cpp/hello_world/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/home/user/zephyr-sdk-0.16.8/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0"  -L/home/user/zephyrproject/zephyr/samples/cpp/hello_world/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -lstdc++  -lgcc  -Wl,-no-pie  -DPICOLIBC_LONG_LONG_PRINTF_SCANF  -T/home/user/zephyrproject/modules/hal/espressif/zephyr/esp32/src/linker/esp32.rom.alias.ld  -T/home/user/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.ld  -T/home/user/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.api.ld  -T/home/user/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.libgcc.ld  -T/home/user/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld  -T/home/user/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld  -T/home/user/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/soc/esp32/ld/esp32.peripherals.ld  -Wl,--wrap=longjmp  --specs=picolibc.specs  -lc  -lgcc && cd /home/user/zephyrproject/zephyr/samples/cpp/hello_world/build/zephyr && /usr/bin/cmake -E true
/home/user/zephyr-sdk-0.16.8/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: /home/user/zephyr-sdk-0.16.8/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/../../picolibc/xtensa-espressif_esp32_zephyr-elf/lib/libc.a(libc_tinystdio_fdopen.c.o):(.literal.fdopen+0x18): undefined reference to `read'
[... 10 lines cut ...]
/home/user/zephyr-sdk-0.16.8/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: /home/user/zephyr-sdk-0.16.8/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/../../picolibc/xtensa-espressif_esp32_zephyr-elf/lib/libstdc++.a(basic_file.o):(.literal._ZNSt12__basic_fileIcE7seekoffExSt12_Ios_Seekdir+0x0): undefined reference to `lseek'
/home/user/zephyr-sdk-0.16.8/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: /home/user/zephyr-sdk-0.16.8/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/../../picolibc/xtensa-espressif_esp32_zephyr-elf/lib/libstdc++.a(basic_file.o):(.literal._ZNSt12__basic_fileIcE9showmanycEv+0x0): undefined reference to `fstat'
/home/user/zephyr-sdk-0.16.8/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: /home/user/zephyr-sdk-0.16.8/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/../../picolibc/xtensa-espressif_esp32_zephyr-elf/lib/libstdc++.a(basic_file.o): in function `(anonymous namespace)::xwrite(int, char const*, int)':
basic_file.cc:(.text._ZN12_GLOBAL__N_1L6xwriteEiPKci+0xb): undefined reference to `write'
[... cut ...]
**Environment (please complete the following information):**
 - OS: Linux - Ubuntu 22.04.4 LTS
 - Toolchain: Zephyr SDK 0.16.8
 - Commit SHA or Version used: 
   - zephyr-rtos repo: `dcfc3e7872542038abc51da7d6a468ac76e71613`
   - hal_espressif repo: `61b977fd2b033c656b1b2fa07b3137872236f710`

**Additional context**
Building the "regular" (C) `samples/hello_world` works fine for this board (and runs fine on the board). So it looks like the ESP32 HAL (from hal_espressif) and toolchain (from Zephyr SDK) are properly installed.

sylvioalves, 2024-07-09:
@nordicjm this issue is not only in ESP32 but seems to be in all Xtensa's.

sylvioalves, 2024-07-09:
Duplicated #66027

keith-packard, 2024-07-11:
libstdc++uses fopen/fdopen which picolibc layers on top of POSIX calls.
```
diff --git a/samples/cpp/hello_world/prj.conf b/samples/cpp/hello_world/prj.conf
index b801dee1855..5fdf8ea531e 100644
--- a/samples/cpp/hello_world/prj.conf
+++ b/samples/cpp/hello_world/prj.conf
@@ -1,2 +1,5 @@
 CONFIG_CPP=y
 CONFIG_REQUIRES_FULL_LIBCPP=y
+CONFIG_POSIX_DEVICE_IO=y
+CONFIG_POSIX_FD_MGMT=y
+CONFIG_POSIX_FILE_SYSTEM=y
```
makes this sample build.

Should picolibc using libstdc++ automatically define these?

sylvioalves, 2024-07-11:
Hi @keith-packard, I think you should address this in #66027 instead.

nashif, 2024-07-11:
closing as dupe again