zephyrproject-rtos/zephyr #75658: net: lib: http_server: close file descriptors before calling http_server_init again

blemouzy, 2024-07-09:
**Describe the bug**

HTTP server crashes after a restart

**To Reproduce**

I start to play with a modified version of `samples/net/sockets/http_server` with a `index.html` which loads a JavaScript application.
When refreshing the page with F5, the `http_server_run` function returns and `http_server_init` is called again.

https://github.com/zephyrproject-rtos/zephyr/blob/67a31ef2d781231aacb44b6a99681750a01123d5/subsys/net/lib/http/http_server_core.c#L749-L762

**Expected behavior**

No crash.

**Impact**

?

**Logs and console output**

* with `CONFIG_ZVFS_EVENTFD_MAX=1`
```
*** Booting Zephyr OS build v3.7.0-rc2 ***
[00:00:03.846,000] <inf> net_config: Initializing network
[00:00:03.846,000] <inf> net_config: IPv4 address: 192.168.0.100
[00:00:31.163,000] <inf> net_http_server: Re-starting server (-12)
[00:00:31.163,000] <err> net_http_server: eventfd failed (-12)
[00:00:31.163,000] <err> net_http_server: Failed to initialize HTTP2 server
[00:00:31.223,000] <err> eth_nxp_enet_mac: ENET_GetRxFrameSize return: 4001
[00:00:32.163,000] <err> os: ***** USAGE FAULT *****
[00:00:32.163,000] <err> os:   Illegal use of the EPSR
[00:00:32.163,000] <err> os: r0/a1:  0x801253a0  r1/a2:  0x00000000  r2/a3:  0x800013cc
[00:00:32.163,000] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x6002a65d
[00:00:32.163,000] <err> os:  xpsr:  0x4000000f
[00:00:32.163,000] <err> os: Faulting instruction address (r15/pc): 0x00000000
[00:00:32.163,000] <err> os: >>> ZEPHYR FATAL ERROR 35: Unknown error on CPU 0
[00:00:32.163,000] <err> os: Fault during interrupt handling

[00:00:32.163,000] <err> os: Current thread: 0x80125fe8 (idle)
[00:00:32.175,000] <err> os: Halting system
```
* with `CONFIG_ZVFS_EVENTFD_MAX=16`
```
*** Booting Zephyr OS build v3.7.0-rc2 ***
[00:00:04.048,000] <inf> net_config: Initializing network
[00:00:04.048,000] <inf> net_config: IPv4 address: 192.168.0.100
[00:00:31.642,000] <inf> net_http_server: Re-starting server (-12)
[00:00:31.642,000] <err> net_ctx: Port 80 is in use!
[00:00:31.642,000] <err> net_http_server: bind: 112
[00:00:31.642,000] <err> net_http_server: All services failed (1)
[00:00:31.642,000] <err> net_http_server: Failed to initialize HTTP2 server
[00:00:32.642,000] <err> os: ***** USAGE FAULT *****
[00:00:32.642,000] <err> os:   Illegal use of the EPSR
[00:00:32.642,000] <err> os: r0/a1:  0x80125670  r1/a2:  0x00000000  r2/a3:  0x800013cc
[00:00:32.642,000] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x6002a65d
[00:00:32.642,000] <err> os:  xpsr:  0x4000000f
[00:00:32.642,000] <err> os: Faulting instruction address (r15/pc): 0x00000000
[00:00:32.642,000] <err> os: >>> ZEPHYR FATAL ERROR 35: Unknown error on CPU 0
[00:00:32.642,000] <err> os: Fault during interrupt handling

[00:00:32.642,000] <err> os: Current thread: 0x801262b8 (idle)
[00:00:32.656,000] <err> os: Halting system
```

**Environment:**
 - OS: Ubuntu 20.04
 - Toolchain (e.g Zephyr SDK, ...): 0.16.3
 - Commit SHA or Version used: v3.7.0-rc2

**Additional context**

When `http_server_init` is called a second time, file descriptors (eventfd, sockets, ...) are already opened but the following code clears them:

https://github.com/zephyrproject-rtos/zephyr/blob/67a31ef2d781231aacb44b6a99681750a01123d5/subsys/net/lib/http/http_server_core.c#L77-L78

rlubos, 2024-07-09:
Not sure about the crash, but we were clearly missing some cleanup in case of errors. Can you check if https://github.com/zephyrproject-rtos/zephyr/pull/75663 helps with your case?

I'm also curious what would be to root cause for the server restart, could you perhaps enable debug logging and see what causes the restart?

blemouzy, 2024-07-09:
> Not sure about the crash, but we were clearly missing some cleanup in case of errors. Can you check if #75663 helps with your case?

It's working fine with `net/http-server-core-cleanup-on-error`: no more crash and no more "Re-starting server".

> I'm also curious what would be to root cause for the server restart, could you perhaps enable debug logging and see what causes the restart?

Please find the logs WITHOUT your patch:
```
*** Booting Zephyr OS build v3.7.0-rc2 ***
[00:00:03.845,000] <inf> net_config: Initializing network
[00:00:03.845,000] <inf> net_config: IPv4 address: 192.168.0.100
[00:00:03.845,000] <dbg> net_http_server: http_server_init: Initialized HTTP Service 192.168.0.100:8            0
[00:00:03.845,000] <dbg> net_http_server: http_server_start: Starting HTTP server
[00:00:22.697,000] <dbg> net_http_server: accept_new_client: New client from 192.168.0.250:51689
[00:00:22.697,000] <dbg> net_http_server: http_server_run: Init client #0
[00:00:22.698,000] <dbg> net_http_server: handle_http_preface: HTTP_SERVER_PREFACE_STATE.
[00:00:22.698,000] <dbg> net_http_server: handle_http1_request: HTTP_SERVER_REQUEST
[00:00:22.699,000] <dbg> net_http_server: on_header_value: Header value too long (by 47 bytes)
[00:00:22.699,000] <dbg> net_http_server: on_header_value: Header value too long (by 52 bytes)
[00:00:22.699,000] <dbg> net_http_server: handle_http1_request: HTTP_SERVER_REQUEST
[00:00:22.699,000] <dbg> net_http_server: on_header_value: Header value too long (by 2 bytes)
[00:00:22.699,000] <dbg> net_http_server: handle_http1_request: Requested URL: /
[00:00:22.699,000] <dbg> net_http_server: get_resource_detail: (http_server_tid): Got match for /
[00:00:22.699,000] <dbg> net_http_server: handle_http1_request: Connection closed client 0x80122fb0
[00:00:22.857,000] <dbg> net_http_server: accept_new_client: New client from 192.168.0.250:51690
[00:00:22.857,000] <dbg> net_http_server: http_server_run: Init client #0
[00:00:22.858,000] <dbg> net_http_server: handle_http_preface: HTTP_SERVER_PREFACE_STATE.
[00:00:22.858,000] <dbg> net_http_server: handle_http1_request: HTTP_SERVER_REQUEST
[00:00:22.858,000] <dbg> net_http_server: on_header_value: Header value too long (by 47 bytes)
[00:00:22.858,000] <dbg> net_http_server: on_header_value: Header value too long (by 2 bytes)
[00:00:22.858,000] <dbg> net_http_server: handle_http1_request: HTTP_SERVER_REQUEST
[00:00:22.858,000] <dbg> net_http_server: handle_http1_request: Requested URL: /assets/index.css
[00:00:22.858,000] <dbg> net_http_server: get_resource_detail: (http_server_tid): Got match for /assets/index.css
[00:00:22.950,000] <dbg> net_http_server: handle_http1_request: Connection closed client 0x80122fb0
[00:00:22.950,000] <dbg> net_http_server: accept_new_client: New client from 192.168.0.250:51691
[00:00:22.950,000] <dbg> net_http_server: http_server_run: Init client #0
[00:00:22.950,000] <dbg> net_http_server: handle_http_preface: HTTP_SERVER_PREFACE_STATE.
[00:00:22.950,000] <dbg> net_http_server: handle_http1_request: HTTP_SERVER_REQUEST
[00:00:22.950,000] <dbg> net_http_server: on_header_value: Header value too long (by 47 bytes)
[00:00:22.950,000] <dbg> net_http_server: on_header_value: Header value too long (by 2 bytes)
[00:00:22.950,000] <dbg> net_http_server: handle_http1_request: HTTP_SERVER_REQUEST
[00:00:22.950,000] <dbg> net_http_server: handle_http1_request: Requested URL: /assets/index.js
[00:00:22.950,000] <dbg> net_http_server: get_resource_detail: (http_server_tid): Got match for /assets/index.js
[00:00:23.096,000] <dbg> net_http_server: handle_http1_request: Connection closed client 0x80122fb0
[00:00:23.257,000] <dbg> net_http_server: accept_new_client: New client from 192.168.0.250:51693
[00:00:23.257,000] <dbg> net_http_server: http_server_run: Init client #0
[00:00:23.258,000] <dbg> net_http_server: handle_http_preface: HTTP_SERVER_PREFACE_STATE.
[00:00:23.258,000] <dbg> net_http_server: handle_http1_request: HTTP_SERVER_REQUEST
[00:00:23.258,000] <dbg> net_http_server: on_header_value: Header value too long (by 47 bytes)
[00:00:23.258,000] <dbg> net_http_server: on_header_value: Header value too long (by 2 bytes)
[00:00:23.258,000] <dbg> net_http_server: handle_http1_request: HTTP_SERVER_REQUEST
[00:00:23.258,000] <dbg> net_http_server: handle_http1_request: Requested URL: /assets/1.png
[00:00:23.258,000] <dbg> net_http_server: get_resource_detail: (http_server_tid): Got match for /assets/1.png
[00:00:23.258,000] <dbg> net_http_server: handle_http1_request: Connection closed client 0x80122fb0
[00:00:23.260,000] <dbg> net_http_server: accept_new_client: New client from 192.168.0.250:51694
[00:00:23.260,000] <dbg> net_http_server: http_server_run: Init client #0
[00:00:23.260,000] <dbg> net_http_server: handle_http_preface: HTTP_SERVER_PREFACE_STATE.
[00:00:23.260,000] <dbg> net_http_server: handle_http1_request: HTTP_SERVER_REQUEST
[00:00:23.261,000] <dbg> net_http_server: on_header_value: Header value too long (by 47 bytes)
[00:00:23.261,000] <dbg> net_http_server: on_header_value: Header value too long (by 2 bytes)
[00:00:23.261,000] <dbg> net_http_server: handle_http1_request: HTTP_SERVER_REQUEST
[00:00:23.261,000] <dbg> net_http_server: handle_http1_request: Requested URL: /assets/2.png
[00:00:23.261,000] <dbg> net_http_server: get_resource_detail: (http_server_tid): Got match for /assets/2.png
[00:00:23.261,000] <dbg> net_http_server: handle_http1_request: Connection closed client 0x80122fb0
[... cut ...]
[00:00:42.869,000] <err> os: r0/a1:  0x80123430  r1/a2:  0x00000000  r2/a3:  0x80000e44
[00:00:42.869,000] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x60021e15
[00:00:42.869,000] <err> os:  xpsr:  0x4000000f
[00:00:42.869,000] <err> os: Faulting instruction address (r15/pc): 0x00000000
[00:00:42.869,000] <err> os: >>> ZEPHYR FATAL ERROR 35: Unknown error on CPU 0
[00:00:42.869,000] <err> os: Fault during interrupt handling

[00:00:42.869,000] <err> os: Current thread: 0x80124078 (idle)
[00:00:42.869,000] <err> os: Halting system
```

rlubos, 2024-07-09:
Thanks for the logs, it was very helpful. So apparently you were affected with some other regression we had temporarily on main, which caused `zsock_poll()` to return `ENOMEM`, as `CONFIG_NET_SOCKETS_POLL_MAX` was ineffective. That's no longer the case on my branch.

However, there's apparently something more that causes the server restart crash than the socket cleanup from my PR, I can still reproduce it when I simulate that ENOMEM scenario. Need to dig more into this.

rlubos, 2024-07-10:
Ok, I've extended #75663 with additional fixes for the problems I've identified, server restarting should work fine now (at least that was the case on my side).

blemouzy, 2024-07-10:
> Ok, I've extended #75663 with additional fixes for the problems I've identified, server restarting should work fine now (at least that was the case on my side).

I still don't have any re-starting message but now, my js file takes about 15 seconds to be served:

```
[00:00:04.145,000] <inf> phy_mc_ksz8081: PHY 0 is up
[00:00:04.145,000] <inf> phy_mc_ksz8081: PHY (0) Link speed 100 Mb, full duplex

*** Booting Zephyr OS build v3.7.0-rc2-258-gb375d94f36d2 ***
[00:00:04.149,000] <inf> net_config: Initializing network
[00:00:04.149,000] <inf> net_config: IPv4 address: 192.168.0.100
[00:00:04.149,000] <dbg> net_http_server: http_server_init: Initialized HTTP Service 192.168.0.100:80
[00:00:04.149,000] <dbg> net_http_server: http_server_start: Starting HTTP server
[00:00:29.520,000] <dbg> net_http_server: accept_new_client: New client from 192.168.0.250:55946
[00:00:29.520,000] <dbg> net_http_server: http_server_run: Init client #0
[00:00:29.522,000] <dbg> net_http_server: handle_http_preface: HTTP_SERVER_PREFACE_STATE.
[00:00:29.522,000] <dbg> net_http_server: handle_http1_request: HTTP_SERVER_REQUEST
[00:00:29.522,000] <dbg> net_http_server: on_header_value: Header value too long (by 47 bytes)
[00:00:29.522,000] <dbg> net_http_server: on_header_value: Header value too long (by 76 bytes)
[00:00:29.522,000] <dbg> net_http_server: handle_http1_request: HTTP_SERVER_REQUEST
[00:00:29.522,000] <dbg> net_http_server: on_header_value: Header value too long (by 2 bytes)
[00:00:29.522,000] <dbg> net_http_server: handle_http1_request: Requested URL: /
[00:00:29.522,000] <dbg> net_http_server: get_resource_detail: (http_server_tid): Got match for /
[00:00:29.522,000] <dbg> net_http_server: handle_http1_request: Connection closed client 0x80122c28
[00:00:29.652,000] <dbg> net_http_server: accept_new_client: New client from 192.168.0.250:55947
[00:00:29.652,000] <dbg> net_http_server: http_server_run: Init client #0
[00:00:29.653,000] <dbg> net_http_server: handle_http_preface: HTTP_SERVER_PREFACE_STATE.
[00:00:29.653,000] <dbg> net_http_server: handle_http1_request: HTTP_SERVER_REQUEST
[00:00:29.653,000] <dbg> net_http_server: on_header_value: Header value too long (by 47 bytes)
[00:00:29.653,000] <dbg> net_http_server: on_header_value: Header value too long (by 2 bytes)
[00:00:29.653,000] <dbg> net_http_server: handle_http1_request: HTTP_SERVER_REQUEST
[00:00:29.653,000] <dbg> net_http_server: handle_http1_request: Requested URL: /assets/index.js
[00:00:29.653,000] <dbg> net_http_server: get_resource_detail: (http_server_tid): Got match for /assets/index.js
[00:00:43.266,000] <dbg> net_http_server: handle_http1_request: Connection closed client 0x80122c28
[00:00:43.267,000] <dbg> net_http_server: accept_new_client: New client from 192.168.0.250:55948
[00:00:43.267,000] <dbg> net_http_server: http_server_run: Init client #0
[00:00:43.267,000] <dbg> net_http_server: handle_http_preface: HTTP_SERVER_PREFACE_STATE.
[00:00:43.267,000] <dbg> net_http_server: handle_http1_request: HTTP_SERVER_REQUEST
[00:00:43.267,000] <dbg> net_http_server: on_header_value: Header value too long (by 47 bytes)
[00:00:43.267,000] <dbg> net_http_server: on_header_value: Header value too long (by 2 bytes)
[00:00:43.267,000] <dbg> net_http_server: handle_http1_request: HTTP_SERVER_REQUEST
[00:00:43.267,000] <dbg> net_http_server: handle_http1_request: Requested URL: /assets/index.css
[00:00:43.267,000] <dbg> net_http_server: get_resource_detail: (http_server_tid): Got match for /assets/index.css
[00:00:43.541,000] <dbg> net_http_server: handle_http1_request: Connection closed client 0x80122c28
...
```

But maybe it's just a browser cache issue?

I also randomly have some new error messages now:

`[00:01:00.230,000] <err> net_tcp: conn: 0x8012f8bc, new bytes 342 during TIME-WAIT state sending reset`

rlubos, 2024-07-10:
> But maybe it's just a browser cache issue?

I'd check with wireshark how the packets fly, I. e. whether the response is sent out immediately or is there indeed some unexpected delay on the Zephyr side. I don't think any of my changes should have an impact here.

> I also randomly have some new error messages now:

> [00:01:00.230,000] <err> net_tcp: conn: 0x8012f8bc, new bytes 342 during TIME-WAIT state sending reset

Can you identify (again, Wireshark could help here) what's actually being sent after the connection is closed on our side? Is it just another request? Perhaps we shouldn't be closing the connection right after serving HTTP1 request.

rlubos, 2024-07-10:
@blemouzy I've pushed one extra commit to my branch, fixing the server behavior when there's no "Connection: close" header - the connection should remain open in such case. I hope this helps for those TCP errors you see.

blemouzy, 2024-07-10:
> I'd check with wireshark how the packets fly, I. e. whether the response is sent out immediately or is there indeed some unexpected delay on the Zephyr side.

The response is indeed sent immediately:

[image]

But 13.69 seconds for a 234 KiB file looks huge:/

> I don't think any of my changes should have an impact here.

Yeah, probably not related to your changes.

> Can you identify (again, Wireshark could help here) what's actually being sent after the connection is closed on our side? Is it just another request? Perhaps we shouldn't be closing the connection right after serving HTTP1 request.

"TCP Retransmission" frames in Wireshark when the error occurs

[image]

> @blemouzy I've pushed one extra commit to my branch, fixing the server behavior when there's no "Connection: close" header - the connection should remain open in such case. I hope this helps for those TCP errors you see.

That looks worse:

```
[00:00:43.871,000] <err> net_pkt: Data buffer (1500) allocation failed.
[00:00:43.871,000] <err> net_tcp: conn: 0x801304bc packet allocation failed, len=1460
[00:00:43.971,000] <err> net_pkt: Data buffer (1340) allocation failed.
[00:00:43.971,000] <err> net_tcp: conn: 0x8013026c packet allocation failed, len=1292
[00:00:44.072,000] <err> net_pkt: Data buffer (68) allocation failed.
[00:00:44.172,000] <err> net_pkt: Data buffer (68) allocation failed.
[00:00:44.172,000] <err> net_tcp: TCP failed to allocate buffer in retransmission
[00:00:44.617,000] <err> net_pkt: Data buffer (68) allocation failed.
[00:00:44.617,000] <err> net_tcp: TCP failed to allocate buffer in retransmission
[00:00:45.235,000] <err> net_pkt: Data buffer (68) allocation failed.
[00:00:45.235,000] <err> net_tcp: TCP failed to allocate buffer in retransmission
[00:00:46.110,000] <err> net_pkt: Data buffer (68) allocation failed.
[00:00:46.110,000] <err> net_tcp: TCP failed to allocate buffer in retransmission
[00:00:47.372,000] <err> net_pkt: Data buffer (68) allocation failed.
[00:00:47.372,000] <err> net_tcp: TCP failed to allocate buffer in retransmission
[00:00:49.215,000] <err> net_pkt: Data buffer (68) allocation failed.
[00:00:49.215,000] <err> net_tcp: TCP failed to allocate buffer in retransmission
[00:00:51.930,000] <err> net_pkt: Data buffer (68) allocation failed.
[00:00:51.930,000] <err> net_tcp: TCP failed to allocate buffer in retransmission
[00:00:53.724,000] <dbg> net_http_server: client_timeout: Client 0x80123828 timeout
[00:00:53.727,000] <dbg> net_http_server: client_timeout: Client 0x80123cd0 timeout
[00:00:54.072,000] <dbg> net_http_server: client_timeout: Client 0x80124178 timeout
[00:00:54.132,000] <err> net_http_server: HTTP request handling error (-105)
[00:00:54.132,000] <dbg> net_http_server: http_server_run: Client #0 has disconnected
[00:00:54.132,000] <dbg> net_http_server: http_server_run: Client #1 has disconnected
[00:00:54.462,000] <err> net_pkt: Data buffer (68) allocation failed.
[00:00:54.462,000] <err> net_tcp: TCP failed to allocate buffer in retransmission
[00:01:00.444,000] <err> net_pkt: Data buffer (68) allocation failed.
[00:01:00.444,000] <err> net_tcp: TCP failed to allocate buffer in retransmission
```

rlubos, 2024-07-10:
Have you by any chance lowered `CONFIG_NET_BUF_TX_COUNT` (the sample defaults to 128)? That'd explain degraded performance, the TCP TX window seems to be pretty small according to wireshark. And with small buffer count and large TCP traffic, it sometimes ends up with those buffer allocation issues (which indicate that for the current net buf count the TX window should be lowered even more).

blemouzy, 2024-07-10:
> Have you by any chance lowered `CONFIG_NET_BUF_TX_COUNT` (the sample defaults to 128)? That'd explain degraded performance, the TCP TX window seems to be pretty small according to wireshark. And [...]

Indeed, I set back the value to its default value during tests...
With same tuning as http_server sample, only 0.097 seconds are needed to download the js file.

Sorry for this.

PR #75663 (2024-07-11): net: lib: http_server: Close sockets on server core errors
In case of fatal errors (during poll() or when handling listening socket), the server operation is restarted. It was missed however, that sockets opened for the server should be closed in such case. This i commit fixes it.

Fixes #75658
closed 2024-07-11 by nashif