zephyrproject-rtos/zephyr #75670: Zephyr gets bus error

sapanshah-eaton, 2024-07-09:
Zephyr OS build v3.7.0-rc1-106-g012520d2d7e8

**Describe the bug**
I am building application for STM32H563zi with Zephyr. It supports two Ethernet sockets connection. It accepts data on TCP socket and send data to UART. Data received from UART is sent back to socket.

After 10 min of operation, I am getting bus fault. 

[00:06:24.316,000] <err> os:   Precise data bus error
[00:06:24.316,000] <err> os:   BFAR Address: 0x7100
[00:06:24.316,000] <err> os: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x2000a0b8
[00:06:24.316,000] <err> os: r3/a4:  0x00007100 r12/ip:  0x00000000 r14/lr:  0x08049729
[00:06:24.316,000] <err> os:  xpsr:  0x21000000
[00:06:24.316,000] <err> os: Faulting instruction address (r15/pc): 0x0801b3c6
[00:06:24.316,000] <err> os: >>> ZEPHYR FATAL ERROR 25: Unknown error on CPU 0
[00:06:24.316,000] <err> os: Current thread: 0x2000a0b8 (unknown)
[00:06:24.382,000] <err> os: Halting system

arm-none-eabi-addr2line.exe -e build/zephyr/zephyr.elf 0x0801b3c6
C:/blr_deps/zephyrproject/zephyr/subsys/net/l2/ethernet/arp.c:443

socket thread stack is 2048, UART buffer is 2048.

Main thread stack is 9216.

Below is network config. 
CONFIG_NET_PKT_RX_COUNT=96
CONFIG_NET_PKT_TX_COUNT=96
CONFIG_NET_BUF_RX_COUNT=256
CONFIG_NET_BUF_TX_COUNT=256

aescolar, 2024-07-09:
CC @jukkar @rlubos

sapanshah-eaton, 2024-07-09:
I am also getting this error frequently 

[00:00:43.067,000] <err> eth_stm32_hal: HAL_ETH_TransmitIT tx_int_sem take timeout
[00:00:43.067,000] <err> eth_stm32_hal: eth_tx: ETH DMA error: dmaerror:4080

sapanshah-eaton, 2024-07-12:
<err> os: ***** BUS FAULT *****
[00:06:58.798,000] <err> os:   Precise data bus error
[00:06:58.798,000] <err> os:   BFAR Address: 0x7
[00:06:58.798,000] <err> os: r0/a1:  0xffffffff  r1/a2:  0x2003b680  r2/a3:  0x200146fc
[00:06:58.798,000] <err> os: r3/a4:  0x20010efc r12/ip:  0x00000000 r14/lr:  0x0801ebf3
[00:06:58.798,000] <err> os:  xpsr:  0x81000200
[00:06:58.798,000] <err> os: Faulting instruction address (r15/pc): 0x08048f4a
[00:06:58.798,000] <err> os: >>> ZEPHYR FATAL ERROR 25: Unknown error on CPU 0
[00:06:58.798,000] <err> os: Current thread: 0x20008a20 (unknown)
[00:06:58.924,000] <err> os: Halting system

This is the latest constant issue. It is coming at zephyr/subsys/net/l2/ethernet/arp.c:78

FRASTM, 2024-07-26:
@sapanshah-eaton could you please give any example of the application or sample you are running ?

hank-chiang-0607, 2024-09-16:
Hi, I also have similar problem.
Here are some information for you:
west build -b mimxrt1060_evkb samples/net/wifi

uart:~$ wifi scan
[00:01:09.586,000] <err> os: ***** BUS FAULT *****
[00:01:09.586,000] <err> os:   Precise data bus error
[00:01:09.586,000] <err> os:   BFAR Address: 0xb083b480
[00:01:09.586,000] <err> os: r0/a1:  0xb083b480  r1/a2:  0x00000000  r2/a3:  0x80002f94
[00:01:09.586,000] <err> os: r3/a4:  0x00000024 r12/ip:  0x00000001 r14/lr:  0x6000c779
[00:01:09.586,000] <err> os:  xpsr:  0x01000000
[00:01:09.586,000] <err> os: Faulting instruction address (r15/pc): 0x6001ce30
[00:01:09.586,000] <err> os: >>> ZEPHYR FATAL ERROR 25: Unknown error on CPU 0
[00:01:09.586,000] <err> os: Current thread: 0x80000b80 (shell_uart)
[00:01:09.628,000] <err> os: Halting system

hank-chiang-0607, 2024-09-16:
os information
*** Booting Zephyr OS build v3.6.0-7732-g0a9a456a22e8 ***

carlescufi, 2024-09-18:
@sapanshah-eaton please provide additional details such as stack sizes, Kconfig options and, if possible, a way to reproduce this.

kevinior, 2024-09-26:
I think I might have been seeing the same problem. It only happens with `CONFIG_ETH_STM32_HAL_API_V2`. I'll describe what I found in my investigation.

If the network cable is unplugged we usually get the `HAL_ETH_TransmitIT tx_int_sem take timeout` error message. If we got that error and the network is reconnected we get a bus error.

After a lot of debugging I worked out that when we see `tx_int_sem take timeout` the error-handling code's called to `HAL_ETH_TxFreeCallback` releases the allocated buffer(s) but the tx descriptor still has a pointer to the `struct eth_stm32_tx_context` object that was allocated on `eth_tx`'s stack.

When the network is reconnected there will eventually be a call to `HAL_ETH_ReleaseTxPacket`, which calls `HAL_ETH_TxFreeCallback` with the (now invalid) pointer to the tx context. When `HAL_ETH_TxFreeCallback` tries to dereference that pointer we get a bus error.

I tried calling `HAL_ETH_ReleaseTxPacket` instead of `HAL_ETH_TxFreeCallback` in the error-handling code but that doesn't work because the tx packet is still owned by the hardware and can't be released.

This problem reproduces reliably on our hardware (STM32H743), where we have eth_stm32_hal patched to work with a 10base-T1S PHY. That PHY is troublesome in a variety of ways, in particular it doesn't seem to communicate link errors. I can't reproduce it on a Nucleo-H753ZI board.

As a workaround I patched eth_stm32_hal to allocate tx contexts from a static array (similar to how the buffers are allocated), so that they remain valid until the HAL is finished with them.

@FRASTM should I create a separate issue for this? I don't want to hijack @sapanshah-eaton issue if it turns out to have a different cause.

erwango, 2024-09-26:
@kevinior Thank for this report. A dedicated issue would be nice indeed (and we'll close current one if it happens to fix issues reported by @sapanshah-eaton).

carlescufi, 2024-10-02:
@sapanshah-eaton please provide additional details or we will have to close this issue.

carlescufi, 2024-10-16:
Closing due to author's inactivity

closed 2024-10-16 by carlescufi