zephyrproject-rtos/zephyr #75673: kernel: timer: timer_behavior : timer_jitter_drift fails on some stm32 boards

FRASTM, 2024-07-09:
**Describe the bug**
The test tests/kernel/timer/timer_behavior/kernel.timer.timer fails.  when executed on nucleo_l073rz or nucleo_wba55cg

Even if similar to https://github.com/zephyrproject-rtos/zephyr/issues/71863  the PR does not fix for the stm32 nucleo target boards

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_l073rz ./tests/kernel/timer/timer_behavior
2. west flash
3. See error

**Expected behavior**
test passed 

**Impact**
CI error

**Logs and console output**
```
*** Booting Zephyr OS build v3.7.0-rc2-181-g145f5e61d9b5 ***                                                                              
Running TESTSUITE timer_jitter_drift                                                                                                      
===================================================================                                                                       
START - test_jitter_drift_timer_period                                                                                                    
periodic timer behavior test using built-in restart mechanism                                                                             
collecting time samples for approx 1 seconds                                                                                              
periodic timer samples gathered, calculating statistics                                                                                   
timer clock rate 32000000, kernel tick rate 10000                                                                                         
period duration statistics for 1000 samples (0 rollovers):                                                                                
  expected: 1000 us,            32000.000000 cycles                                                                                       
  min:      900.000000 us,      28800 cycles                                                                                              
  max:      1100.031250 us,     35201 cycles                                                                                              
  mean:     1000.000094 us,     32000.003000 cycles                                                                                       
  variance: 20.006255 us,       20486.404991 cycles                                                                                       
  stddev:   4.472835 us,        143.130722 cycles                                                                                         
timer start cycle 894272, end cycle 32894275,                                                                                             
total time 1000000.093750 us, expected time 1000000.000000 us,                                                                            
expected time drift 0.000000 us, difference 0.093750 us                                                                                   
RECORD: {"testcase":"jitter_drift_timer", "mechanism":"builtin", "stats_count":1000, "rollovers":0, "mean_us":1000.000094, "mean_cycles":}
                                                                                                                                          
    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_behavior/src/jitter_drift.c:324: do_test_using: (max_us <= max_us_bou)
Longest timer period too long (off by more than expected 10%)                                                                             
 FAIL - test_jitter_drift_timer_period in 1.260 seconds                                                                                   
===================================================================                                                                       
START - test_jitter_drift_timer_startdelay                                                                                                
periodic timer behavior test using explicit start with delay                                                                              
collecting time samples for approx 1 seconds                                                                                              
periodic timer samples gathered, calculating statistics                                                                                   
timer clock rate 32000000, kernel tick rate 10000                                                                                         
period duration statistics for 1000 samples (0 rollovers):                                                                                
  expected: 1000 us,            32000.000000 cycles                                                                                       
  min:      900.000000 us,      28800 cycles                                                                                              
  max:      1100.031250 us,     35201 cycles                                                                                              
  mean:     1000.000094 us,     32000.003000 cycles                                                                                       
  variance: 20.006255 us,       20486.404991 cycles                                                                                       
  stddev:   4.472835 us,        143.130722 cycles                                                                                         
timer start cycle 41661666, end cycle 73661669,                                                                                           
total time 1000000.093750 us, expected time 1000000.000000 us,                                                                            
expected time drift 0.000000 us, difference 0.093750 us                                                                                   
RECORD: {"testcase":"jitter_drift_timer", "mechanism":"startdelay", "stats_count":1000, "rollovers":0, "mean_us":1000.000094, "mean_cycle}
                                                                                                                                          
    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_behavior/src/jitter_drift.c:324: do_test_using: (max_us <= max_us_bou)
Longest timer period too long (off by more than expected 10%)                                                                             
 FAIL - test_jitter_drift_timer_startdelay in 1.260 seconds                                                                               
===================================================================                                                                       
TESTSUITE timer_jitter_drift failed. 
```

**Environment (please complete the following information):**
 - OS: ubuntu
 - Toolchain  0.16.3
 - Commit SHA or Version used : 3.7.0-rc2

golowanow, 2024-07-10:
@FRASTM, are these log lines truncated in the device log itself or just in your sample cut/paste here ?

```
RECORD: {"testcase":"jitter_drift_timer", "mechanism":"builtin", "stats_count":1000, "rollovers":0, "mean_us":1000.000094, "mean_cycles":}
```
```
RECORD: {"testcase":"jitter_drift_timer", "mechanism":"startdelay", "stats_count":1000, "rollovers":0, "mean_us":1000.000094, "mean_cycle}
```

FRASTM, 2024-07-17:
> @FRASTM, are these log lines truncated in the device log itself or just in your sample cut/paste here ?
>
> ```
> RECORD: {"testcase":"jitter_drift_timer", "mechanism":"builtin", "stats_count":1000, [...]
truncated in the copy/paste.
Here are the complete lines:

```
Running TESTSUITE timer_jitter_drift
===================================================================
START - test_jitter_drift_timer_period
periodic timer behavior test using built-in restart mechanism
collecting time samples for approx 1 seconds
periodic timer samples gathered, calculating statistics
timer clock rate 32000000, kernel tick rate 10000
period duration statistics for 1000 samples (0 rollovers):
expected: 1000 us,       	32000.000000 cycles
min:      900.000000 us, 	28800 cycles
max:      1100.031250 us, 	35201 cycles
mean:     1000.000094 us, 	32000.003000 cycles
variance: 20.006255 us, 	20486.404991 cycles
stddev:   4.472835 us, 	143.130722 cycles
timer start cycle 891072, end cycle 32891075,
total time 1000000.093750 us, expected time 1000000.000000 us,
expected time drift 0.000000 us, difference 0.093750 us
RECORD: {"testcase":"jitter_drift_timer", "mechanism":"builtin", "stats_count":1000, "rollovers":0, "mean_us":1000.000094, "mean_cycles":32000, "stddev_us":4.472835, "stddev_cycles":143, "var_us":20.006255, "var_cycles":20486, "min_us":900.000000, "min_cycles":28800, "max_us":1100.031250, "max_cycles":35201, "timer_start_cycle": 891072, "timer_end_cycle": 32891075, "total_time_us":1000000.093750, "expected_total_time_us":1000000.000000, "expected_total_drift_us":0.000000, "total_drift_us":0.093750, "expected_period_cycles":32000, "expected_period_drift_us":0.000000, "sys_clock_hw_cycles_per_sec":32000000, "CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC":32000000, "CONFIG_SYS_CLOCK_TICKS_PER_SEC":10000, "CONFIG_TIMER_TEST_PERIOD":1000, "CONFIG_TIMER_TEST_SAMPLES":1000, "MAX STD DEV":10}
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_behavior/src/jitter_drift.c:324: do_test_using: (max_us <= max_us_bound is false)
Longest timer period too long (off by more than expected 10%)
FAIL - test_jitter_drift_timer_period in 1.260 seconds
===================================================================

```
```
START - test_jitter_drift_timer_startdelay
periodic timer behavior test using explicit start with delay
collecting time samples for approx 1 seconds
periodic timer samples gathered, calculating statistics
timer clock rate 32000000, kernel tick rate 10000
period duration statistics for 1000 samples (0 rollovers):
expected: 1000 us,       	32000.000000 cycles
min:      900.000000 us, 	28800 cycles
max:      1100.031250 us, 	35201 cycles
mean:     1000.000094 us, 	32000.003000 cycles
variance: 20.006255 us, 	20486.404991 cycles
stddev:   4.472835 us, 	143.130722 cycles
timer start cycle 41658466, end cycle 73658469,
total time 1000000.093750 us, expected time 1000000.000000 us,
expected time drift 0.000000 us, difference 0.093750 us
RECORD: {"testcase":"jitter_drift_timer", "mechanism":"startdelay", "stats_count":1000, "rollovers":0, "mean_us":1000.000094, "mean_cycles":32000, "stddev_us":4.472835, "stddev_cycles":143, "var_us":20.006255, "var_cycles":20486, "min_us":900.000000, "min_cycles":28800, "max_us":1100.031250, "max_cycles":35201, "timer_start_cycle": 41658466, "timer_end_cycle": 73658469, "total_time_us":1000000.093750, "expected_total_time_us":1000000.000000, "expected_total_drift_us":0.000000, "total_drift_us":0.093750, "expected_period_cycles":32000, "expected_period_drift_us":0.000000, "sys_clock_hw_cycles_per_sec":32000000, "CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC":32000000, "CONFIG_SYS_CLOCK_TICKS_PER_SEC":10000, "CONFIG_TIMER_TEST_PERIOD":1000, "CONFIG_TIMER_TEST_SAMPLES":1000, "MAX STD DEV":10}
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_behavior/src/jitter_drift.c:324: do_test_using: (max_us <= max_us_bound is false)
Longest timer period too long (off by more than expected 10%)
FAIL - test_jitter_drift_timer_startdelay in 1.260 seconds
===================================================================

```

closed 2024-08-07 by nashif