zephyrproject-rtos/zephyr #75674: Several calls of task_wdt_init starve the application as soon as a sleep is used

bfa8, 2024-07-09:
**Description**
When the watchdog task initialization is done multiple times (multiple calls to task_wdt_init) then a sleep is called after that, the application hangs.

The application hangs in change_cpu_state_and_wait (file /zephyr/scripts/native_simulator/common/src/nce.c) on line 158:
[image]

Before, we used Zephyr 3.5.0 and called task_wdt_init multiple times, where everything worked well. Now we wanted to update to Zephyr 3.6.0 and our code didn't work anymore. There the following commit was done in Zephyr where we think the issue comes from:
https://github.com/zephyrproject-rtos/zephyr/commit/fdea8518783a5d8c4421595341f8332ac5714167

For sure, it's not best practice to initialize the watchdog multiple times (multiple calls to task_wdt_init). We would expect that task_wdt_init would return an error in case of multiple calls (since this would have an influence on the running application).

Here is a sample project where we reproduced the bug:
[multiple_calls_to_task_wdt_init.zip](https://github.com/user-attachments/files/16145180/multiple_calls_to_task_wdt_init.zip)

**How to Reproduce**
Steps to reproduce the behavior:
1. build the attached multiple_calls_to_task_wdt_init project
2. Set breakpoints on the task_wdt_init calls.
3. If you passed the second task_wdt_init call. Run the application and do a stop after some time, then you will see, that during the sleep the code get stuck in change_cpu_state_and_wait.

**Expected behavior**
Either, multiple calls to task_wdt_init are possible, and the application will execute successful OR repeated calls to task_wdt_init will return an error. 

**Environment:**
 - OS: Linux
 - Toolchain Zephyr SDK Version 0.16.5
 - Zephyr Version 3.6.0 as well as 3.7.0

**Additional context**
We solved the problem in our application by removing the second call to task_wdt_init. But the behavior is not really user-friendly like it's implemented.

aescolar, 2024-07-09:
CC @arbrauns

arbrauns, 2024-07-10:
16fc8f5295cc666923dabfff9d268d8e16dd8259/#73699 might also be relevant. It's a weird error behaviour, but I don't think calling `task_wdt_init` multiple times is inteded behaviour, it should have a check and return `-EBUSY` or something.

bfa8, 2024-07-10:
> [16fc8f5](https://github.com/zephyrproject-rtos/zephyr/commit/16fc8f5295cc666923dabfff9d268d8e16dd8259)/#73699 might also be relevant. It's a weird error behaviour, but I don't think calling `task_wdt_init` multiple times is inteded usage, it should have a check and return `-EBUSY` or something.

Yes, I completely agree.

aescolar, 2024-07-10:
@bfa8 Just as side info
What you see in that line you pointed at is just the HW models thread waiting for the CPU to go to sleep.
The CPU does not sleep in this case because there is an infinite loop in the systick timer interrupt handler calling into task_wdt_trigger() and that function setting the deadline again to the same instant.

closed 2025-01-28 by github-actions[bot]