zephyrproject-rtos/zephyr #75693: timer: kernel: cortex_m_systick: multiple minor issues in the current driver

avolkov-1221, 2024-07-10:
The current Cortex-M SysTick timer driver implementation contains some issues such as:

* the SysTick interrupt handler was incorrectly implemented as a standard C function instead of ISR_DIRECT one, may lead to the unpredictable behaviour in the current/future Zephyr's versions.

* the use of bitfields from a hardware-mapped C-struct may leads to multiple unprotected non-atomic "read-modify-write" operations. Access to all hardware registers, especially the SYST_CSR, must be atomic. I.e. dedicated **sys_readXX/sys_writeXX** functions should be used instead of C-struct style accesses.

henrikbrixandersen, 2024-07-10:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

avolkov-1221, 2024-07-10:
> Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

Done. However, I'm not sure that the template is very suitable for this type of issues.

ithinuel, 2024-07-10:
>  The timer should work regardless of Zephyr version or compiler's version/optimisation level used.

This would imply that there are cases where it does not but there is no reproduction steps.
Is it a speculative assumption or have you effectively encountered issues since the lasts fixes that happened a few months back?

> the use of bitfields from a hardware-mapped C-struct may leads to multiple unprotected non-atomic "read-modify-write" operations. Access to all hardware registers, especially the SYST_CSR, must be [...]

This struct is provided by CMSIS and this is used as the reference for cortex-m core peripherals.
All the fields are defined as volatile and accesses are therefore protected from spurious optimisation.

> the SysTick interrupt handler was incorrectly implemented as a standard C function instead of ISR_DIRECT one, [...]

Can you please elaborate on what makes this implementation incorrect?
As far as I can tell, the `ISR_DIRECT` adds:
- The interrupt attribute.
- Tracing infrastructure which I don't think is desirable on systick do to the frequency at which it triggers.

As far as I understand, the interrupt attribute isn't strictly required either since the hardware handles the exception entry/exit conforming with the AAPCS. This is not going to change (not even slightly) for those cores.

avolkov-1221, 2024-07-10:
Hi, thank you for the fast response.

> Hi, thank your for reporting your concerns.
> 
> > The timer should work regardless of Zephyr version or compiler's version/optimisation level used.
> 
> This would imply that there are cases where it does not but there is no reproduction steps. Is it a speculative assumption or have you effectively encountered issues since the lasts fixes that happened a few months back?

For now it's mainly assumption, but I suspect that #66578 may be related to the missing, due to incorrect "clear by read",  underun check. In any case, due to SYST_CSR nature, the CMSIS style is not well fitted for it: you can't read only a part of the register, so COUNTFLAG will be always cleared.

> This struct is provided by CMSIS and this is used as the reference for cortex-m core peripherals. All the fields are defined as volatile and accesses are therefore protected from spurious [...]

Yes , I know  where the decls come from, and already have seen the generated code:  indirectly disabled optimisation where it suppose to be, non-atomic "read-write-modify".

AFAIR it was long discussion in the early 2010x in LKML about exactly the same topic: usage volatile in structs. Broken optimisation and cache flaw was one of the reasons for refusing to use it in the modern Kernel.

> Can you please elaborate on what makes this implementation incorrect? As far as I can tell, the `ISR_DIRECT` adds:
>
> * The interrupt attribute.
> * Tracing infrastructure which I don't think is [...]
1. Trace is not so interesting for me, as power management provided by ISR_DIRECT_PM() et al. 
2. The handler must be defined at least as  'void handler (void)', not 'void handler(arg)'. 
    Sorry, but were the  'arg' come from in the direct IRQ handler?
    BTW I've an experience with terrible code generated by the newer gcc, for incorrectly declared functions.
    And it was not problem of gcc, but ancient code. Will we wait for a similar situation?

> As far as I understand, the interrupt attribute isn't strictly required either since the hardware handles the exception entry/exit conforming with the AAPCS. This is not going to change (not even slightly) for those cores.

Interrupts attribute mark the function as a special one, and at least supposed that all used registers were preserved on stack. Usual calling conventions mayor may not do this. Could you guaranteed that any feature let's say ARM Cortex-M v1X ABI will not change the behaviour for the price/performance/something else?

ithinuel, 2024-07-11:
> For now it's mainly assumption, but I suspect that https://github.com/zephyrproject-rtos/zephyr/issues/66578 may be related to the missing […]

I see. This issue will be closed shortly as it was resolved in https://github.com/zephyrproject-rtos/zephyr/pull/63344 and therefore cannot be reproduced.

> AFAIR it was long discussion in the early 2010x in LKML about exactly the same topic […]

I don't have the background nor the bandwidth to investigate this history and evaluate how relevant it is for the case at hand. I would rather stick with what CMSIS offers since it is meant to be the reference. If it is not fit for purpose, this is where a fix should be provided rather than relying on a Zephyr specific solution.

> 1. Trace is not so interesting for me, as power management provided by ISR_DIRECT_PM() et al.

Then, using Zephyr's macro may not be a good solution.

> 2. The handler must be defined at least as 'void handler (void)', not 'void handler(arg)'.

That is correct indeed. A PR could be made to change that signature and add the missing interrupt attribute.

> Could you guaranteed that any feature let's say ARM Cortex-M v1X ABI will not change the behaviour for the price/performance/something else?

I cannot speak for the future but if a new architecture version were to break this, there'd be a lot of other thing to address in the kernel, let alone this driver.
As far as I can tell, for now armv6m, armv7m and armv8m both base and main line are compatible with this approach.

ithinuel, 2024-07-11:
> Then please take it as axiome that "volatile" in structs is evil, and is not recommended for production releases.

I may have missed the guideline. Would you be able to point me to the reference in Zephyr’s documentation please?

> Sorry for Captain Obvious mode, but the generic reference != the best for everyone.

I agree, we are not aiming at that thankfully. This reference specificly designed for Cortex-M will be best for Zephyr’s target :)

> Has been done actually. The PR is https://github.com/zephyrproject-rtos/zephyr/pull/75695

Thanks, I’ll review that momentarily.

commit 213da72d (2024-08-14): timer: cortex_m_systick: use direct interrupt handler instead of C-function

The current driver implements the global defined "systick" interrupt
callback "sys_clock_isr" as a standard C function with an argument.
However, ARM's direct interrupt handlers do not have any arguments;
they must be declared as "void handler(void)".

Additionally, the direct handler should include some missing special
header/footer.

Fixes: #75693
closed 2024-08-14 by MaureenHelm