zephyrproject-rtos/zephyr #75704: z_unpend_thread_no_timeout

Hadatko, 2024-07-10:
**Notes (delete this)**
Github Discussions (https://github.com/zephyrproject-rtos/zephyr/discussions)
are available to first verify that the issue is a genuine Zephyr bug and not a
consequence of Zephyr services misuse.

This issue list is only for bugs in the main Zephyr code base
(https://github.com/zephyrproject-rtos/zephyr/). If the bug is for a project
fork (such as NCS) specific feature, please open an issue in the fork project
instead.

**Describe the bug**
A clear and concise description of what the bug is.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - What have you tried to diagnose or workaround this issue?
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - ...

[image]

Usage ALWAYS_INLINE in definition in sched.c file makes our compiler complaning for missing implementation of that function in timer.c

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

Maybe others compilers are not complaning. Currently we are using unsupported architecture

**Expected behavior**
A clear and concise description of what you expected to happen.

Removing ALWAYS_INLINE from source definition fixed issue.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

[image]

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows) Linux
 - Toolchain (e.g Zephyr SDK, ...) CGT from TI
 - Commit SHA or Version used
VERSION_MAJOR = 3
VERSION_MINOR = 6
PATCHLEVEL = 99
VERSION_TWEAK = 0
EXTRAVERSION =

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

nashif, 2024-07-10:
@Hadatko  your bug text has lots of boilerplate and template text, please clarify what the bug is exaxtly, starting with the title.

Hadatko, 2024-07-10:
Hi, maybe you could use markdown brackets for disabling description in posted file. Like here: https://github.com/EmbeddedRPC/erpc/blob/develop/.github/ISSUE_TEMPLATE/bug_report.md?plain=1
I saw this also in other projects.

That is just an suggestion.
The desription:
[image]
Usage ALWAYS_INLINE in definition in sched.c file makes our compiler complaning for missing implementation of that function in timer.c
Maybe others compilers are not complaning. Currently we are using unsupported architecture
[image]
Removing ALWAYS_INLINE from source definition fixed issue.

PR #76122 (2024-07-27): kernel: Remove ALWAYS_INLINE from z_unpend_thread_no_timeout()
A little bit of kernel housekeeping.

1. Removes the declaration for z_sched_abort() as that routine no longer exists.

2. Removes the ALWAYS_INLINE attribute from the definition of the routine z_unpend_thread_no_timeout() to fix an unresolved symbol error that was occurring with some compilers.  (Fixes #75704)
closed 2024-07-27 by nashif