zephyrproject-rtos/zephyr #75710: crypto: secp256r1: crypto.secp256r1.p256-m_raw does not build

katgiadla, 2024-07-10:
**Describe the bug**
The test `crypto/secp256r1/crypto.secp256r1.p256-m_raw` does not build.

Observed for: 
- `nrf5340dk/nrf5340/cpuapp/ns`
- `nrf9160dk/nrf9160/ns`

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf5340dk/nrf5340/cpuapp` connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/crypto/secp256r1 -p nrf5340dk/nrf5340/cpuapp/ns --device-testing --device-serial /dev/ttyACM1 -v --inline-logs --west-flash="--erase"`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
-- Including generated dts.cmake file: /zephyr/zephyr/twister-out/nrf5340dk_nrf5340_cpuapp_ns/tests/crypto/secp256r1/crypto.secp256r1.p256-m_raw/zephyr/dts.cmake

warning: MBEDTLS_PSA_CRYPTO_C (defined at modules/mbedtls/Kconfig.tls-generic:485,
modules/mbedtls/Kconfig.tls-generic:485) was assigned the value 'y' but got the value 'n'. Check
these unsatisfied dependencies: ((!BUILD_WITH_TFM && MBEDTLS_BUILTIN && MBEDTLS_CFG_FILE = "config-
tls-generic.h" && MBEDTLS) || (!BUILD_WITH_TFM && MBEDTLS_BUILTIN && MBEDTLS_CFG_FILE = "config-tls-
generic.h" && MBEDTLS && 0)) (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_MBEDTLS_PSA_CRYPTO_C and/or look up
MBEDTLS_PSA_CRYPTO_C in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

warning: The choice symbol MBEDTLS_PSA_CRYPTO_EXTERNAL_RNG (defined at modules/mbedtls/Kconfig.tls-
generic:468, modules/mbedtls/Kconfig.tls-generic:468) was selected (set =y), but no symbol ended up
as the choice selection. See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_MBEDTLS_PSA_CRYPTO_EXTERNAL_RNG and/or look
up MBEDTLS_PSA_CRYPTO_EXTERNAL_RNG in the menuconfig/guiconfig interface. The Application
Development Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of
the manual might be helpful too.

Parsing /zephyr/zephyr/zephyr/Kconfig
Loaded configuration '/zephyr/zephyr/zephyr/boards/nordic/nrf5340dk/nrf5340dk_nrf5340_cpuapp_ns_defconfig'
Merged configuration '/zephyr/zephyr/zephyr/tests/crypto/secp256r1/prj.conf'
Merged configuration '/zephyr/zephyr/zephyr/tests/crypto/secp256r1/p256-m_raw.conf'
Merged configuration '/zephyr/zephyr/twister-out/nrf5340dk_nrf5340_cpuapp_ns/tests/crypto/secp256r1/crypto.secp256r1.p256-m_raw/zephyr/misc/generated/extra_kconfig_options.conf'

warning: MBEDTLS_PSA_P256M_DRIVER_ENABLED (defined at modules/mbedtls/Kconfig.tls-generic:509, modules/mbedtls/Kconfig.tls-generic:509) has direct dependencies (MBEDTLS_PSA_CRYPTO_C && MBEDTLS_BUILTIN && MBEDTLS_CFG_FILE = "config-tls-generic.h" && MBEDTLS) || (MBEDTLS_PSA_CRYPTO_C && MBEDTLS_BUILTIN && MBEDTLS_CFG_FILE = "config-tls-generic.h" && MBEDTLS && 0) with value n, but is currently being y-selected by the following symbols:
 - MBEDTLS_PSA_P256M_DRIVER_RAW (defined at modules/mbedtls/Kconfig.tls-generic:517, modules/mbedtls/Kconfig.tls-generic:517), with value y, direct dependencies (MBEDTLS_BUILTIN && MBEDTLS_CFG_FILE = "config-tls-generic.h" && MBEDTLS) || (MBEDTLS_BUILTIN && MBEDTLS_CFG_FILE = "config-tls-generic.h" && MBEDTLS && 0) (value: y), and select condition MBEDTLS_BUILTIN && MBEDTLS_CFG_FILE = "config-tls-generic.h" && MBEDTLS (value: y)

error: Aborting due to Kconfig warnings

CMake Error at zephyr/cmake/modules/kconfig.cmake:389 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  zephyr/cmake/modules/zephyr_default.cmake:132 (include)
  zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:75 (include)
  zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:97 (include_boilerplate)
  zephyr/cmake/package_helper.cmake:126 (find_package)
```

**Environment (please complete the following information):**
- OS: Ubuntu 20.04.1 LTS
- Toolchain: Zephyr SDK 0.16.8
- Commit SHA or Version used: `v3.7.0-rc2-227-g65f63622e273`

**Additional context**
Probably introduced by 83cd9f5a01664e82d3a0cb062b356b9a4de3e755

hakehuang, 2024-07-13:
NXP has the same issue on lpcxpresso55s69/lpc55s69/cpu0/ns

nashif, 2024-07-14:
marking as release blocker due to countless build errors on multiple platforms.

PR #75717 (2024-07-15): modules: mbedtls: fix MBEDTLS_PSA_P256M_DRIVER_RAW Kconfig option
Make it depend on `MBEDTLS_PSA_P256M_DRIVER_ENABLED` instead of selecting it.

This fixes the build of
`tests/crypto/secp256r1/crypto.secp256r1.p256-m_raw` on filtered-out platforms since PR #75441 (`83cd9f5a01664e82d3a0cb062b356b9a4de3e755`).

Fixes #75710.
closed 2024-07-15 by nashif