zephyrproject-rtos/zephyr #75746: Fail to build samples/sysbuild/hello_world

aaronyegx, 2024-07-11:
**Describe the bug**

**To Reproduce**

**Expected behavior**

**Logs and console output**

**Environment (please complete the following information):**

aaronyegx, 2024-07-11:
Hi @gmarull , I encounter the building issue on samples/sysbuild/hello_world. Could you help check if I miss anything? Thanks.

henrikbrixandersen, 2024-07-11:
The command `west build -b nrf54h20dk/nrf54h20/cpuapp --sysbuild samples/sysbuild/hello_world -- -DSB_CONFIG_REMOTE_BOARD='"nrf54h20dk/nrf54h20/cpurad"'` builds fine under GNU/Linux. I assume this must be a Windows CMD issue.

gmarull, 2024-07-11:
The issue you are observing is likely due to https://github.com/zephyrproject-rtos/zephyr/issues/69552, which has not been fixed yet. On Linux the escaping trick `'"` works fine, maybe not on Windows. You can still make use of twister configurations as suggested in the sample, that is, run:

```
west build -b nrf54h20dk/nrf54h20/cpuapp -T sample.sysbuild.hello_world.nrf54h20dk_cpuapp_cpurad samples/sysbuild/hello_world
```

I'll close this issue as there's another tracking the problem, please, add your details there if necessary.

closed 2024-07-11 by gmarull

krissrex, 2026-08-17:
On Windows, try `-DSB_CONFIG_REMOTE_BOARD='\"nrf54h20dk/nrf54h20/cpurad\"'`