zephyrproject-rtos/zephyr #75758: ST LIS2DUX12 driver: will not correctly read values on big-endian HW

ragrus-nbl, 2024-07-11:
**Describe the bug**
ST LIS2DUX12 accelerometer driver makes unnecessary "little-endian to CPU endian" conversion. This will lead to incorrect readings on big-endian HW.

I believe these `sys_le16_to_cpu()` conversions are not needed: [three here](https://github.com/zephyrproject-rtos/zephyr/blob/v3.7.0-rc2/drivers/sensor/st/lis2dux12/lis2dux12.c#L165) and [one there](https://github.com/zephyrproject-rtos/zephyr/blob/v3.7.0-rc2/drivers/sensor/st/lis2dux12/lis2dux12.c#L188).

That's because values are converted to CPU endianness in the hal_st code: [here](https://github.com/zephyrproject-rtos/hal_st/blob/b77157f6bc4395e398d90ab02a7d2cbc01ab2ce7/sensor/stmemsc/lis2dux12_STdC/driver/lis2dux12_reg.c#L637) and [there](https://github.com/zephyrproject-rtos/hal_st/blob/b77157f6bc4395e398d90ab02a7d2cbc01ab2ce7/sensor/stmemsc/lis2dux12_STdC/driver/lis2dux12_reg.c#L679) correspondingly.

I can't reproduce/check this in practice as I don't have a big-endian HW with this sensor.

**To Reproduce**
Read any sensor channel (e.g., SENSOR_CHAN_DIE_TEMP) on big-endian HW.

**Environment**
 - Zephyr 3.7.0-rc2

avisconti, 2024-07-25:
Well, the BE/LE stuff is always confusing me.

The STDC drivers (included in zephyr as stmemsc in hal_st module) has indeed the purpose to abstract from the platform details. So, they should in fact already solve the BE/LE related issue, and I would like to check on some BE platform  (I do not have any) to see whether this is really the case. But I agree that from zephyr driver standpoint that calls (i.e ```sys_le16_to_cpu()```) should not be there as the stmemsc HAL should already take care of the byte swapping as I mentioned earlier...

Let me think more about it. Thanks for pointing this out.

erwango, 2024-07-31:
Removing `STM32` label as this is not related to STM32 SoC.

commit 9ea4cb96 (2024-08-27): drivers/sensor/st: Fix wrong data byte swap for be

A 16-bit value built using byte shifts and ORs from a given
couple of lsb and msb bytes will result to be the same on both
little-endian and big-endian architectures, e.g.

    uint8_t lsb, msb;
    int16_t val;

    /* val is the same number on both le and be archs, but has
       different layout in memory */
    val = (msb << 8) | lsb;

All the xyz_raw_get() APIs of stmemsc sensor module build the sensor
data using the above method and DO NOT hence require (it actually leads
to wrong values on big-endian machines) to use any le/be swap rou [... cut]
commit d800ccea (2024-08-31): drivers/sensor/st: Fix wrong data byte swap for be

A 16-bit value built using byte shifts and ORs from a given
couple of lsb and msb bytes will result to be the same on both
little-endian and big-endian architectures, e.g.

    uint8_t lsb, msb;
    int16_t val;

    /* val is the same number on both le and be archs, but has
       different layout in memory */
    val = (msb << 8) | lsb;

All the xyz_raw_get() APIs of stmemsc sensor module build the sensor
data using the above method and DO NOT hence require (it actually leads
to wrong values on big-endian machines) to use any le/be swap rou [... cut]
closed 2024-08-27 by nashif