zephyrproject-rtos/zephyr #75785: Bluetooth: Regression in various BAP and CAP qualification test cases

sjanc, 2024-07-11:
**Describe the bug**
There is regression in BT stack that is causing issues with BAP and CAP qualification test cases.
This seems to affect 80+ test cases.

**To Reproduce**
Execute BAP/UCL/STR/BV-556-C qualification test case.

**Expected behavior**
Test pass.

**Impact**
Bluetooth qualification process is affected.

**Additional context**
With bisecting first bad commit seems to be "Bluetooth: Controller: Refactor of ull_adv_sync_pdu_set_clear()" but reverting it on top of main doesn't fix the issue.
I also noticed that checking out before mentioned commit but not updating modules with 'west update' doesn't fix test.

In fail case ISO_SDU_Length is 0, while in pass it has test data. Both logs attached.

[fail_BAP_UCL_STR_BV_556_C_2024_07_11_17_04_23.zip](https://github.com/user-attachments/files/16179705/fail_BAP_UCL_STR_BV_556_C_2024_07_11_17_04_23.zip)

[pass_BAP_UCL_STR_BV_556_C_2024_07_11_16_54_24.zip](https://github.com/user-attachments/files/16179706/pass_BAP_UCL_STR_BV_556_C_2024_07_11_16_54_24.zip)

Thalley, 2024-07-11:
Link to the PR that caused this regression: https://github.com/zephyrproject-rtos/zephyr/pull/71660

Thalley, 2024-07-11:
Can't assign but @Tronil is the author of the regression :)

Thalley, 2024-07-11:
@carlescufi this is the regression @sjanc was talking about earlier today. Should we consider this a high priority issue due to number of regression, or medium due to ISO/LE Audio being considered unstable?

Tronil, 2024-07-30:
FYI this is unlikely to be caused by #71660; It did introduce a stack overflow but that was later fixed. It does *not* have any influence on SDU lengths, it only changes periodic advertising.

Also according to @sjanc (on discord):
I tried to revert all controller related patches up to one that checking out before fix the issue
but it doesn't help

Thalley, 2024-07-30:
> FYI this is unlikely to be caused by #71660; It did introduce a stack overflow but that was later fixed. It does _not_ have any influence on SDU lengths, it only changes periodic advertising.

I think we tracked down the issue via `git bisect`. @sjanc please confirm. 

> Also according to @sjanc (on discord): I tried to revert all controller related patches up to one that checking out before fix the issue but it doesn't help

The fix referring to here is the stack overflow fix, which didn't fix the issues seen here.

sjanc, 2024-07-30:
Yes, this was done with bisecting and is reproducible in my environment...
"scripts: twisterlib: filter coverage information" (hash b1290cb86cb89b7897f6de9e2907bd2bc97d2388) is last commit that works for me

Just note that this might be more subtle issue since, eg modules also seems to affect this (eg there were mbedtls updates too), but as I played with this nothing seems to help unless I get back to above commit.

sjanc, 2024-08-01:
Hi,

Okay, so I spent some more time looking at this issue.

If I leave controller image on "scripts: twisterlib: filter coverage information" and bisect only application core I end up with following first bad commit "Bluetooth: conn: Allocate TX context JIT".

However, checkout out before this commit, applying BT_HCI_TX_STACK_SIZE size fix and rebuilding both cores... doesn't fix the issue.

TBH I'm not sure on what is going on.

cvinayak, 2024-08-03:
@sjanc Thank you for analyzing the issue.

Definitely more eyes are needed, is there any chance you can describe a test case that can be manually tested at my desk to reproduce the qualification test case failure? It could (without my expertise with autopts) motivate me to investigate/bisect at my desk.

sjanc, 2024-08-19:
Hi,

I double check this with SoftDevice based controller and first bad commit is again "Bluetooth: conn: Allocate TX context JIT".

sjanc, 2024-08-19:
I was able to debug this...   https://github.com/zephyrproject-rtos/zephyr/pull/77242

PR #77242 (2024-08-21): test:bluetooth:tester: Fix user_data size in pools
There was hardcoded value for user_data which is no longer valid. Use proper CONFIG_BT_CONN_TX_USER_DATA_SIZE for that.

Fixes #75785
closed 2024-08-21 by fabiobaltieri