zephyrproject-rtos/zephyr #75789: DNS name is not getting resolved second time after Network disconnection and reconnection

ashutoshpandey-eaton, 2024-07-11:
Running an IOT application on NUCLEO-H743ZI which uses MQTT, **first time DNS name is getting resolved** and the application runs fine.
**In between (when the application is running) when ethernet cable is removed and attached after ~5-10 seconds, then DNS name is not getting resolved. and getting error -11.** from zsock_getaddrinfo [getaddrinfo.c] 

When debugged more found following call sequence:

zsock_getaddrinfo () [getaddrinfo.c]  >>z_zsock_getaddrinfo_internal() >>exec_query()>>dns_get_addr_info() [ [getaddrinfo.c] ]>>dns_resolve_name[resolve.c]>>dns_write() [resolve.c]>>net_context_sendto[net_context.c]>>
context_sendto>>context_setup_udp_packet>>net_context_create_ipv4_new() returns -22 [net_context.c]>>
net_ipv4_is_addr_unspecified() _then print message:_ "DROP: src addr is unspecified" from net_context.c

**error status:**
exec_query() returns -11 [DNS_EAI_SYSTEM] as dns_get_addr_info() is returning -2. dns_write()[resolve.c] returns -22.
NOTE:CONFIG_NET_IPV4 is used.

rlubos, 2024-07-12:
This is no longer reproducible on `main`, but can indeed be reproduced with DHCPv4-assigned address on Zephyr v3.6.0.

The root cause DNS fails is lack of IPv4 address on the interface, and IPv4 address is missing because DHCP starts in a wrong state after network interface DOWN/UP toggle (which happens when you plug out/in cable). This was fixed with https://github.com/zephyrproject-rtos/zephyr/commit/0f56974c9d61bafcbfae813fa17919d5e40f8cac, I've triggered a backport PR towards 3.6 branch: https://github.com/zephyrproject-rtos/zephyr/pull/75820

rlubos, 2024-07-16:
Fixed with https://github.com/zephyrproject-rtos/zephyr/pull/75820

closed 2024-07-16 by rlubos

prayassamriya, 2024-07-16:
> This is no longer reproducible on `main`, but can indeed be reproduced with DHCPv4-assigned address on Zephyr v3.6.0.
> 
> The root cause DNS fails is lack of IPv4 address on the interface, and IPv4 address is missing because DHCP starts in a wrong state after network interface DOWN/UP toggle (which happens when you plug out/in cable). This was fixed with [0f56974](https://github.com/zephyrproject-rtos/zephyr/commit/0f56974c9d61bafcbfae813fa17919d5e40f8cac), I've triggered a backport PR towards 3.6 branch: #75820

@rlubos is there a plan to release another version of 3.6?

rlubos, 2024-07-16:
> @rlubos is there a plan to release another version of 3.6?

I'm not aware of any bugfix releases planned.

prayassamriya, 2024-07-16:
> I'm not aware of any bugfix releases planned.

okay, I am wondering why are we maintaining 3.6 backport branch. may be that is the standard process for each release, I am not aware of?

rlubos, 2024-07-16:
> okay, I am wondering why are we maintaining 3.6 backport branch. may be that is the standard process for each release, I am not aware of?

https://docs.zephyrproject.org/latest/releases/#periodic-releases

Specifically:
> Periodic and non-LTS releases are maintained with updates, bug fixes and security related updates for at least two cycles

prayassamriya, 2024-07-19:
> https://docs.zephyrproject.org/latest/releases/#periodic-releases
>
> Specifically:

Thanks for sharing

PR #75820 (2024-07-15): [Backport v3.6-branch] net: lib: dhcpv4: goto INIT on IF down, not RENEWING
Backport 0f56974c9d61bafcbfae813fa17919d5e40f8cac from #73118.

Fixes #75789