zephyrproject-rtos/zephyr #75795: doc: Documentation issue in 'boards/nxp/vmu_rt1170/doc/index'

allegrocantabile, 2024-07-11:
**Describe the bug**

[VMU-RT1170 product page link and User's Guide link do not contain material]

"https://www.nxp.com/part/VMU-RT1170" points to the correct address, but the content shows "page not found"

"https://cognipilot.org/cerebri/boards/nxp_vmu_rt1170/" points to the correct address, but shows 404 error

**Google Chrome**

* Page: `boards/nxp/vmu_rt1170/doc/index`
* Version: main
* SHA-1: 9a73c32852c5460327499c0c6cb3e07a5f95c80e

glenn-andrews, 2024-07-15:
Welp, I though this would be an easy fix, but both NXP and CogniPilot have scrubbed all mention for VMU products from their web pages.

danieldegrasse, 2024-11-01:
It looks like this would have been fixed by #75863. @allegrocantabile please ping me here if that is not the case and you'd like this issue reopened.

closed 2024-11-01 by danieldegrasse

glenn-andrews, 2024-11-01:
I spoke to NXP. There were supposed to be real documentation pages coming, but I've not seen them yet and the engineer who told me about them ghosted me afterwards.

glenn-andrews, 2024-11-01:
By an amazing coincidence, he just emailed me. There's been a delay and they should have the pages up in a month or so.

dleach02, 2024-11-01:
@glenn-andrews I reached out to that team. 

NXP is a big company so lots of different things going on. For the benefit of other readers, the vmu_rt1170 board is still in development and hasn't reached official production release. NXP has processes for when web pages and stuff get posted and normally they occur when the board is officially announced. 

The enablement we have in Zephyr is considered early enablement or early access. We probably, for this board, should have just had a more skeleton docs page but I suspect they didn't know the original pages they referenced were not permanent (again, big company).