zephyrproject-rtos/zephyr #75813: Real time performace impacted by ethernet traffic

aneejosep, 2024-07-12:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/75424

<div type='discussions-op-text'>

<sup>Originally posted by **aneejosep** July  4, 2024</sup>
I am evaluating Zephyr for one of the upcoming product development and found that the real time performance is greatly affected when there is traffic in the network interface. 

The way I do my test is as follows.

- A timer to signal a semaphore every 1ms.
- A high priority task to receive this signal and add trace point.
- Adding trace point is nothing but to get the current time and compute the difference from the previous point (ideally it should be 1ms) and store it in an array.
- Record 100K points and after tests are done, print the values to console.
- Use these values to plot a histogram.

I see that when no network traffic the results are ok. But when there is network traffic, the values even shift more than 500μs on an STM32 board. On an NXP board it is even worse like 3+ ms jitter.

[image]

For NXP, an issue #75624 was created and fixed by [this pull request](https://github.com/zephyrproject-rtos/zephyr/pull/75625).

For STM32, it has to be fixed by the block in the HAL driver as mentioned in the answer in the discussion.
</div>

KozhinovAlexander, 2024-07-17:
Please link ethernet driver used from Zephyr repository. In my opinion there are two different implementations.

marwaiehm-st, 2024-09-02:
I modified the function "eth_iface_init" to  allow users to configure the Ethernet RX thread by adding the preemptive threading (to reduce jitter) :  
https://github.com/marwaiehm-st/zephyr/commit/26422e31897f6b9afdf9bac80fd2d76a72986261

```c
		k_thread_create(&dev_data->rx_thread, dev_data->rx_thread_stack,
				K_KERNEL_STACK_SIZEOF(dev_data->rx_thread_stack),
				rx_thread, (void *) dev, NULL, NULL,
				IS_ENABLED(CONFIG_NET_TC_THREAD_PREEMPTIVE) ?
				K_PRIO_PREEMPT(CONFIG_ETH_STM32_HAL_RX_THREAD_PRIO) :
				K_PRIO_COOP(CONFIG_ETH_STM32_HAL_RX_THREAD_PRIO),
				0, K_NO_WAIT);
``` 

I tested on the board **stm32h747i_disco/stm32h747xx/m7**

- without modifications:  using the cooperative thread
==> the jitter is around **450μs** for 1ms task when there is Ethernet traffic.

[image]

- with modification and CONFIG_NET_TC_THREAD_PREEMPTIVE=y: using the preemptive
==> the jitter is **90μs** when there is Ethernet traffic.

[image]

The jitter is below 100μs, **is this acceptable, or does it need to be less than that** ?

KozhinovAlexander, 2024-09-03:
> I modified the function "eth_iface_init" to  allow users to configure the Ethernet RX thread by adding the preemptive threading (to reduce jitter) :  
> https://github.com/marwaiehm-st/zephyr/commit/26422e31897f6b9afdf9bac80fd2d76a72986261
> 
> ```c
> 		k_thread_create(&dev_data->rx_thread, dev_data->rx_thread_stack,
> 				K_KERNEL_STACK_SIZEOF(dev_data->rx_thread_stack),
> 				rx_thread, (void *) dev, NULL, NULL,
> 				IS_ENABLED(CONFIG_NET_TC_THREAD_PREEMPTIVE) ?
> 				K_PRIO_PREEMPT(CONFIG_ETH_STM32_HAL_RX_THREAD_PRIO) :
> 				K_PRIO_COOP(CONFIG_ETH_STM32_HAL_RX_THREAD_PRIO),
> 				0, K_NO_WAIT);
> ``` 
> 
> 
> I tested on the board **stm32h747i_disco**
> 
> - without modifications:  using the cooperative thread
> ==> the jitter is around **450μs** for 1ms task when there is Ethernet traffic.
> 
> [image]
> 
> 
> 
> - with modification and CONFIG_NET_TC_THREAD_PREEMPTIVE=y: using the preemptive
> ==> the jitter is **90μs** when there is Ethernet traffic.
> 
> [image]
> 
> 
> The jitter is below 100μs, **is this acceptable, or does it need to be less than that** ?
> 
> 
> 
Reducing the hitter by factor of 5 is great. Did you created a PR also - it would be very nice. I can test it then.

marwaiehm-st, 2024-09-03:
> > I modified the function "eth_iface_init" to  allow users to configure the Ethernet RX thread by adding the preemptive threading (to reduce jitter) :
> > [marwaiehm-st@26422e3](https://github.com/marwaiehm-st/zephyr/commit/26422e31897f6b9afdf9bac80fd2d76a72986261)
> > ```c
> > 		k_thread_create(&dev_data->rx_thread, dev_data->rx_thread_stack,
> > 				K_KERNEL_STACK_SIZEOF(dev_data->rx_thread_stack),
> > 				rx_thread, (void *) dev, NULL, NULL,
> > 				IS_ENABLED(CONFIG_NET_TC_THREAD_PREEMPTIVE) ?
> > 				K_PRIO_PREEMPT(CONFIG_ETH_STM32_HAL_RX_THREAD_PRIO) :
> > 				K_PRIO_COOP(CONFIG_ETH_STM32_HAL_RX_THREAD_PRIO),
> > 				0, K_NO_WAIT);
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > I tested on the board **stm32h747i_disco**
> > 
> > * without modifications:  using the cooperative thread
> >   ==> the jitter is around **450μs** for 1ms task when there is Ethernet traffic.
> > 
> > ![without_modif](https://private-user-images.githubusercontent.com/161579084/363784472-eec2063e-a053-4ce8-ae5f-0a339e566ec2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUzNDkwNzAsIm5iZiI6MTcyNTM0ODc3MCwicGF0aCI6Ii8xNjE1NzkwODQvMzYzNzg0NDcyLWVlYzIwNjNlLWEwNTMtNGNlOC1hZTVmLTBhMzM5ZTU2NmVjMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwM1QwNzMyNTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jMmYwZTdlMWU3OTkzMWY5NTdmNDkxY2Y4ZjA2MzMxYzY5MzhkZDg0NGVmZmE5N2ZjZTcwZTQ0NDllYTBiZDFjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.q7C9zTLCNPfjVcXeQIA1f16MPUu9kl-eDYKgPV_b_uU)
> > 
> > * with modification and CONFIG_NET_TC_THREAD_PREEMPTIVE=y: using the preemptive
> >   ==> the jitter is **90μs** when there is Ethernet traffic.
> > 
> > ![with_modifica](https://private-user-images.githubusercontent.com/161579084/363788450-42bee47e-4aea-4efc-885d-3cd5dc8bdaa1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUzNDkwNzAsIm5iZiI6MTcyNTM0ODc3MCwicGF0aCI6Ii8xNjE1NzkwODQvMzYzNzg4NDUwLTQyYmVlNDdlLTRhZWEtNGVmYy04ODVkLTNjZDVkYzhiZGFhMS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwM1QwNzMyNTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01ZjI2MDEwYjYzYjljOGJiODQ5YjE4MjFlN2M2NjgwOTBlNjZmMTc2ZjNlZDRhMjA2NTEyZDhiYjM2NDNiYzZkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.oV5U8ywirD6_zXv7Otd7JBHrFbqu33vZ57kISOBiuDE)
> > The jitter is below 100μs, **is this acceptable, or does it need to be less than that** ?
> 
> Reducing the hitter by factor of 5 is great. Did you created a PR also - it would be very nice. I can test it then.

This is the [PR](https://github.com/zephyrproject-rtos/zephyr/pull/77904)

KozhinovAlexander, 2024-09-03:
> This is the [PR](https://github.com/zephyrproject-rtos/zephyr/pull/77904)

Thanks a lot. Could you also please describe, how you did the measurements. I would like to reproduce it and see, if I get same values. Your plots of measurements are looking very promising.

marwaiehm-st, 2024-09-03:
I follwed the description of the issue and added the modifications to the zperf sample to test with it

- Test Setup

- [ ] Hardware Setup:

Board: STM32H747I-DISCO
Ethernet Connection: Connect the board and the computer to a network switch .

- Configuration

- [ ]  Conf Settings:

Enable preemptive threading for the Ethernet RX thread on the file prj.conf 
(**to reproduce the issue, do not enable preemptive threading**)
```shell
CONFIG_NET_TC_THREAD_PREEMPTIVE=y
``` 
Set the IP adresses 
```shell
CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.168.0.111"
CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.168.0.110"
``` 
- [ ] Real-Time Task :

Implement a high-priority task that waits for a semaphore signaled by a timer every 1ms.
Record the jitter by measuring the time difference between consecutive semaphore signals.
Example: I added this code to main file of the test zperf

```c
#define NUM_POINTS 80000

K_SEM_DEFINE(timer_sem, 0, 1);

static uint32_t trace_points[NUM_POINTS];
static uint32_t prev_time;
static uint32_t index = 0;

void timer_handler(struct k_timer *dummy) {
    k_sem_give(&timer_sem);
}

K_TIMER_DEFINE(my_timer, timer_handler, NULL);

void high_priority_task(void) {
    uint32_t current_time;
    uint32_t time_diff;

    while (index < NUM_POINTS) {
        k_sem_take(&timer_sem, K_FOREVER);
        current_time = k_cycle_get_32();
        time_diff = current_time - prev_time;
        trace_points[index++] = time_diff;
        prev_time = current_time;
    }

    // Print the values to the console
    for (int i = 0; i < NUM_POINTS; i++) {
        printk("%u\n", trace_points[i]);
    }
}

K_THREAD_DEFINE(high_priority_thread, 1024, high_priority_task, NULL, NULL, NULL, 1, 0, 0);
``` 
Add to the main function of the zperf test:
```c
prev_time = k_cycle_get_32();
k_timer_start(&my_timer, K_MSEC(1), K_MSEC(1));
``` 

- Running the Test

- [ ] Generate Ethernet Traffic :

Use a network  computer to send continuous Ethernet traffic to the board.
Use **iperf** to generate traffic:, "192.168.0.111" is the ip adress of the board
On the host (IP 192.168.0.110) run iperf:
```shell
iperf -c  192.168.0.111 -u -w 1024 -t 120 -l 1024
``` 

- [ ] Collect the Data
Use a serial terminal  minicom to capture the output of the real-time task.
Save the output from the serial terminal to a file.

- [ ] Analyze the Data
I used a Python script to load the data from the file and plot the histogram.

KozhinovAlexander, 2024-09-03:
Great description. Please give me time at least till the end of the week, s.t. I can test it on my h745.
Meanwhile you can better prepare your PR.

theom4, 2024-09-10:
Why don't you try another TCP/IP stack such as Mongoose?

KozhinovAlexander, 2024-09-10:
> Why don't you try another TCP/IP stack such as Mongoose?

This topic does not covers main problem in the issue. Please open a duscussion thread.

Mongoose has no Apache 2.0 license support, which is required by Zephyr:
- Mongoose licensing: https://mongoose.ws/licensing/
- Zephyr licensing approach: https://docs.zephyrproject.org/latest/LICENSING.html

Some components were itegrated with license other than Apache 2.0, but in long term they shall be replaced, cause no commertial products otherwise are possible.

FRASTM, 2024-09-18:
closing as the https://github.com/zephyrproject-rtos/zephyr/pull/77904 is merged

closed 2024-09-18 by FRASTM