zephyrproject-rtos/zephyr #75825: Tests: Bluetooth: Missing ICS in Bluetooth Host qualification file

theob-pro, 2024-07-12:
**Describe the bug**

The following ICS may be enabled:

- GAP: 17b/4: LE security mode 3 level 3
- GAP: 20A/6: Security Manager OOB
- GAP: 27/7: Writable Appearance
- GATT: 3/24: Enabling Robust Caching

**Impact**

People using the Zephyr qualification file to qualify their product may not qualify some feature even tho the feature is supported by the Zephyr Bluetooth Host.

Thalley, 2024-07-15:
3/16 Characteristic Value Reliable Writes should also be supported

theob-pro, 2024-07-15:
> 3/16 Characteristic Value Reliable Writes should also be supported

Yeah, good catch, I missed it

sjanc, 2024-07-22:
Regarding "GAP: 20A/6: Security Manager OOB"  I'm actually not sure what those are suppose to mean (also couple other already set), since this is not allowed to be present in AD/SCAN_RSP as those are ...well OOB  :-D      (I'll set it for time being though)

Regarding "GAP: 23/7: Connected Isochronous Stream Terminate procedure" I'm also bit confused (same for 23/6) as those are "C.1: Mandatory IF LL 9/32 "Connected Isochronous Stream - Peripheral", otherwise Excluded."  And since this is host only there is no LL 9/32 ...      It seems that intention was to test procedure so I'll set it for time being.

Regarding "GATT: 3/25a: Enabling Robust Caching", this is GATT client configuration, is this supported? Or assumption is that application handles that?

Thalley, 2024-07-30:
> Regarding "GAP: 20A/6: Security Manager OOB" I'm actually not sure what those are suppose to mean (also couple other already set), since this is not allowed to be present in AD/SCAN_RSP as those are ...well OOB :-D (I'll set it for time being though)

Might be missing something here - What does this have to do with advertising or scan response data?

> Regarding "GAP: 23/7: Connected Isochronous Stream Terminate procedure" I'm also bit confused (same for 23/6) as those are "C.1: Mandatory IF LL 9/32 "Connected Isochronous Stream - Peripheral", [...]

Indeed, sounds weird. Nevertheless it's a feature that we support. 

> Regarding "GATT: 3/25a: Enabling Robust Caching", this is GATT client configuration, is this supported? Or assumption is that application handles that?

Just looked through `gatt.c`, and we do not have any client support for this as the client. I've removed it again from https://github.com/zephyrproject-rtos/zephyr/pull/75821

sjanc, 2024-07-30:
> Might be missing something here - What does this have to do with advertising or scan response data?

Those are AD/SR data types in ICS   (some are used in OOB channel only but are listed here)

Thalley, 2024-07-30:
> Those are AD/SR data types in ICS (some are used in OOB channel only but are listed here)

Found the reference: 
[image]

It's just whether we support setting these flags, which Zephyr definitely does.

sjanc, 2024-07-30:
nah, those ICS are for types defined in CSS

Thalley, 2024-07-30:
> nah, those ICS are for types defined in CSS

What do you mean? The 20A/6 refers to this section in this document

sjanc, 2024-07-30:
my mistake, you already linked to CSS

what I mean is that 20A is for peripheral (Table 20A: Peripheral Link Layer Advertising Data Types) while it lists data types that are not allowed in AD/SR

Thalley, 2024-07-30:
> while it lists data types that are not allowed in AD/SR

Why are these not allowed?

sjanc, 2024-07-30:
Cause those are for OOB, there is table on page 12 which shows where each data type is allowed

Thalley, 2024-07-30:
> Cause those are for OOB, there is table on page 12 which shows where each data type is allowed

I see. Thanks. So the OOB data flags is only something we can expose out of band. What's preventing us from just printing or logging it? That's OOB. 

Question is whether enabling it prevents us from passing any tests. I imagine that PTS may request the OOB data which we can probably send via BTP, so I don't see it as an issue to enable

sjanc, 2024-07-30:
I don't think it affects any of PTS tests as PTS is not testing OOB channels, but after thinking more about this (and checking some documentation regarding qualification) I think those should be enabled in ICS anyway (especially that SMP OOB pairing is supported).

closed 2024-08-29 by carlescufi