zephyrproject-rtos/zephyr #75843: POSIX kconfigs show up although POSIX APIs is not being used

nashif, 2024-07-13:
**Describe the bug**
Building a hello world result in many POSIX kconfigs being enabled or showing in the resulting .config

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b qemu_x86 samples/hello_world
2. grep ^CONFIG_POSIX build/zephyr/.config
CONFIG_POSIX_AEP_CHOICE_NONE=y
CONFIG_POSIX_OPEN_MAX=4
CONFIG_POSIX_PAGE_SIZE=0x1000
CONFIG_POSIX_LIMITS_RTSIG_MAX=0
CONFIG_POSIX_MAX_FDS=4
CONFIG_POSIX_MAX_OPEN_FILES=4

**Expected behavior**

When a feature is disabled, it should not contaminate resulting .config with unrelated options. All Kconfigs of that feature shall be guarded

**Impact**

Confusion and potential leakge of features...

**Logs and console output**

grep ^CONFIG_POSIX build/zephyr/.config
CONFIG_POSIX_AEP_CHOICE_NONE=y
CONFIG_POSIX_OPEN_MAX=4
CONFIG_POSIX_PAGE_SIZE=0x1000
CONFIG_POSIX_LIMITS_RTSIG_MAX=0
CONFIG_POSIX_MAX_FDS=4
CONFIG_POSIX_MAX_OPEN_FILES=4

**Environment (please complete the following information):**

 - OS:  Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: 3.7.-rc2

nashif, 2024-07-13:
also note many deprecated options that do not start with CONFIG_POSIX...

```
-- Generated zephyr.dts: /home/nashif/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/nashif/zephyrproject/zephyr/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /home/nashif/zephyrproject/zephyr/build/zephyr/dts.cmake

warning: MAX_THREAD_BYTES (defined at arch/Kconfig:377) was assigned the value '3' but got the value
''. Check these unsatisfied dependencies: USERSPACE (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_MAX_THREAD_BYTES and/or look up
MAX_THREAD_BYTES in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

warning: BOUNDS_CHECK_BYPASS_MITIGATION (defined at kernel/Kconfig:942) was assigned the value 'y'
but got the value 'n'. Check these unsatisfied dependencies: USERSPACE (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_BOUNDS_CHECK_BYPASS_MITIGATION and/or look
up BOUNDS_CHECK_BYPASS_MITIGATION in the menuconfig/guiconfig interface. The Application Development
Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual
might be helpful too.

Parsing /home/nashif/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/nashif/zephyrproject/zephyr/boards/qemu/nios2/qemu_nios2_defconfig'
Merged configuration '/home/nashif/zephyrproject/zephyr/tests/kernel/common/prj.conf'
Configuration saved to '/home/nashif/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/nashif/zephyrproject/zephyr/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: /home/nashif/bin/zephyr-sdk-0.16.8/nios2-zephyr-elf/nios2-zephyr-elf/bin/ld.bfd (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/nashif/bin/zephyr-sdk-0.16.8/nios2-zephyr-elf/bin/nios2-zephyr-elf-gcc
-- Using ccache: /usr/bin/ccache
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nashif/zephyrproject/zephyr/build
-- west build: running target run
[1/126] Preparing syscall dependency handling

[2/126] Generating include/generated/zephyr/version.h
-- Zephyr version: 3.7.0-rc2 (/home/nashif/zephyrproject/zephyr), build: v3.7.0-rc2-487-g2e012668e904
[125/126] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           RESET:          0 GB         32 B      0.00%
           FLASH:      104654 B     131040 B     79.86%
             RAM:       15500 B     131040 B     11.83%
        IDT_LIST:          0 GB         2 KB      0.00%
Generating files from /home/nashif/zephyrproject/zephyr/build/zephyr/zephyr.elf for board: qemu_nios2
[125/126] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: nios2
	cpu->env.reset_addr: 		0
	cpu->env.exception_addr: 	400020
***** delaying boot 500ms (per build configuration) *****
*** Booting Zephyr OS build v3.7.0-rc2-487-g2e012668e904 (delayed boot 500ms) ***
Running TESTSUITE atomic
===================================================================
START - test_atomic
 PASS - test_atomic in 0.000 seconds
===================================================================
START - test_atomic_overflow
 PASS - test_atomic_overflow in 0.000 seconds
===================================================================
START - test_threads_access_atomic
 PASS - test_threads_access_atomic in 0.400 seconds
===================================================================
TESTSUITE atomic succeeded
Running TESTSUITE bitarray
===================================================================
START - test_bitarray_alloc_free
Testing bit array alloc and free with predefined patterns
Testing bit array alloc and free with divisor 1
Testing bit array alloc and free with divisor 2
[... 959 lines cut ...]
# CONFIG_POSIX_MQUEUE is not set
# CONFIG_POSIX_PUTMSG is not set
# CONFIG_POSIX_SIGNAL is not set
# CONFIG_POSIX_SYSCONF is not set
# CONFIG_POSIX_SYSLOG is not set
# CONFIG_POSIX_UNAME is not set
# CONFIG_PTHREAD is not set
# CONFIG_PTHREAD_BARRIER is not set
# CONFIG_PTHREAD_COND is not set
# CONFIG_PTHREAD_IPC is not set
# CONFIG_PTHREAD_KEY is not set
# CONFIG_PTHREAD_MUTEX is not set
# CONFIG_PTHREAD_RWLOCK is not set
# CONFIG_PTHREAD_SPINLOCK is not set
# CONFIG_TIMER is not set
CONFIG_TIMER_DELAYTIMER_MAX=0
CONFIG_SEM_NAMELEN_MAX=0
CONFIG_SEM_VALUE_MAX=0
# end of Deprecated POSIX options
# end of POSIX Options

#
# POSIX Shell Utilities
                                                                                                                               1039,1        77%
```

cfriedt, 2024-07-13:
`CONFIG_POSIX_AEP_CHOICE_NONE=y` is expected (it's the default choice, saying not to enable POSIX).

Some of the aforementioned options are deprecated (not going to change until they are removed).

Others simply represent constants.

cfriedt, 2024-07-13:
> also note many deprecated options that do not start with CONFIG_POSIX...

... which is one reason why they are deprecated.

cfriedt, 2024-07-13:
This is hardly a bug... deprecated Kconfig options following the regular deprecation process?

cfriedt, 2024-07-13:
Also, FYI, CONFIG_POSIX_API will be deprecated after v3.7.0

nashif, 2024-07-13:
> `CONFIG_POSIX_AEP_CHOICE_NONE=y` is expected (it's the default choice, saying not to enable POSIX).
> 
ok, so this is what is replacing CONFIG_POSIX_API?

> It might be possible to move these
> 
> ```
> CONFIG_POSIX_OPEN_MAX=4
> CONFIG_POSIX_PAGE_SIZE=0x1000
> ```

I do not know about moving and if that is needed, what we need is some guard somewhere or make them depend on something else and show up only when the parent option is enabled.

> 
> These are deprecated (not going to change until they are removed).
> 
> ```
> CONFIG_POSIX_LIMITS_RTSIG_MAX=0
> CONFIG_POSIX_MAX_FDS=4
> CONFIG_POSIX_MAX_OPEN_FILES=4
> ```

No need to change, but didnt those depend on POSIX_API before they were deprecated? Kconfig.deprecated could maybe depend on POSIX_API so they do not show up in each generated .config.

nashif, 2024-07-13:
> This is hardly a bug... deprecated Kconfig options following the regular deprecation process?

when kconfig options show up whether the feqture is enabled or not, it is a bug. Kconfigs of features need to be contained and guarded.

cfriedt, 2024-07-13:
> > `CONFIG_POSIX_AEP_CHOICE_NONE=y` is expected (it's the default choice, saying not to enable POSIX).
> 
> ok, so this is what is replacing CONFIG_POSIX_API?

CONFIG_POSIX_AEP_CHOICE_NONE=y

> No need to change, but didnt those depend on POSIX_API before they were deprecated? Kconfig.deprecated could maybe depend on POSIX_API so they do not show up in each generated .config.

That could work. I'll make a PR.

cfriedt, 2024-07-15:
Made some additional notes in the PR, but `CONFIG_POSIX_API` can't be used for gating anything (it was also not previously used for that). It was mainly used for default-y-selection of common options and for adjusting the default include search path.