zephyrproject-rtos/zephyr #75848: doc: Documentation issue in 'services/device_mgmt/mcumgr'

Rob-Grady, 2024-07-13:
This section is extremely confusing, and much of it is incorrect. It really should be rewritten.

I would recommend describing the subsystem from the point of view of the device it is running on. The current documentation is constantly switching between 'remote' and local systems without actually telling the reader. Documentation for the 'client' should be removed or at least placed is a separate area.  The client is not part of zephyr and all the talk about it only adds to the confusion of what you are really talking about.  

**Environment**

* Page: `services/device_mgmt/mcumgr`
* Version: main
* SHA-1: 1906ab33da8970f4098bcacba11cd5f35e572754

nordicjm, 2024-07-15:
Where does it mention client? And yes there is a zephyr client https://github.com/zephyrproject-rtos/zephyr/tree/main/subsys/mgmt/mcumgr/smp_client and groups https://github.com/zephyrproject-rtos/zephyr/tree/main/subsys/mgmt/mcumgr/grp/img_mgmt_client and https://github.com/zephyrproject-rtos/zephyr/tree/main/subsys/mgmt/mcumgr/grp/os_mgmt_client

nordicjm, 2024-09-10:
Ping @Rob-Grady

nordicjm, 2024-11-01:
Inactivity for over 3 months, closing.

closed 2024-11-01 by nordicjm