zephyrproject-rtos/zephyr #75878: tests/lib/cpp/cxx fail to build durign cmake step on esp32c6_devkitc

aescolar, 2024-07-15:
**Describe the bug**
* tests/lib/cpp/cxx/cpp.main.cpp14 
* tests/lib/cpp/cxx/cpp.main.cpp17 
* tests/lib/cpp/cxx/cpp.main.cpp20 
* tests/lib/cpp/cxx/cpp.main.cpp2A 
* tests/lib/cpp/cxx/cpp.main.cpp2B 
* tests/lib/cpp/cxx/cpp.main.cpp98 
* tests/lib/cpp/cxx/cpp.main.minimal 
* tests/lib/cpp/cxx/cpp.main.newlib 
* tests/lib/cpp/cxx/cpp.main.newlib_nano 
* tests/lib/cpp/cxx/cpp.main.picolibc 
Fail to build for esp32c6_devkitc

**To Reproduce**
Steps to reproduce the behavior:
1.  twister -p esp32c6_devkitc -T tests/lib/cpp/cxx/
4. See error

**Expected behavior**
No build failures, and test passes

**Impact**
Failures in main CI

**Logs and console output**
https://github.com/zephyrproject-rtos/zephyr/runs/27431582117
```
Cmake build failure
Loading Zephyr default modules (Zephyr base).
...
CMake Error at /__w/zephyr/zephyr/cmake/modules/extensions.cmake:428 (add_library):
  Cannot find source file:

    /__w/zephyr/zephyr/soc/espressif/esp32c6/power.c
```

**Environment (please complete the following information):**

 - OS: Linux, Ubuntu 22.04, CI
 - Toolchain: 0.16.8
 - Commit SHA or Version used: Main as of now fd0e46506fabd4c0c91ecb336a39cb6a6aa2b270 (3.7.0-rc3)

**Additional context**

raffarost, 2024-07-15:
Hi @aescolar,
The first Zephyr support of ESP32C6 indeed ended up mistakenly including HAS_PM and HAS_POWEROFF.
We have a PR open for review that will add support to PM and fix these issues:

 https://github.com/zephyrproject-rtos/zephyr/pull/75278

aescolar, 2024-07-15:
Marking as blocker, so we can have a green CI for 3.7.0 after merging the fix.

PR #75880 (2024-07-16): soc/espressif/esp32c6: Do not set HAS_PM or HAS_POWEROFF
The source files required for this features are not present in the tree.
So if CONFIG_PM or CONFIG_POWEROFF are enabled, there would be a cmake failure.

Let's just indicate these features are not supported in kconfig.

Fixes #75878
closed 2024-07-16 by nashif