zephyrproject-rtos/zephyr #75898: ARC: nsim/nsim_vpx5: build broken

evgeniy-paltsev, 2024-07-15:
**Describe the bug**

After recent nsim SoCs & boards reorganization (#72862) the `SOC_SERIES_*` config is missing for vpx5 SoC which leads to cmake errors when building against nsim/nsim_vpx5 configuration.

This was noted after addition of `nsim/nsim_vpx5` to internal CI.

For example:

```
Loading Zephyr default modules (Zephyr base).
-- Application: /verification_workspace/zephyr/samples/philosophers
-- CMake version: 3.22.1
-- Found Python3: /usr/bin/python (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /root/.cache/zephyr
-- Zephyr version: 3.7.0-rc3 (/verification_workspace/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: nsim, qualifiers: nsim_vpx5
-- Found host-tools: zephyr 0.16.8 (/opt/toolchains/zephyr-sdk-0.16.8)
-- Found host-tools: zephyr 0.16.8 (/opt/toolchains/zephyr-sdk-0.16.8)
-- Found dts preprocessor: /opt/toolchains/zephyr-sdk-0.16.8/aarch64-zephyr-elf/bin/aarch64-zephyr-elf-gcc (Zephyr SDK 0.16.8)
-- Found Dtc: /opt/toolchains/zephyr-sdk-0.16.8/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /verification_workspace/zephyr/boards/snps/nsim/arc_classic/nsim_nsim_vpx5.dts
-- Generated zephyr.dts: /verification_workspace/twister-out/nsim_nsim_vpx5/samples/philosophers/sample.kernel.philosopher.fifos/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /verification_workspace/twister-out/nsim_nsim_vpx5/samples/philosophers/sample.kernel.philosopher.fifos/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /verification_workspace/twister-out/nsim_nsim_vpx5/samples/philosophers/sample.kernel.philosopher.fifos/zephyr/dts.cmake
Parsing /verification_workspace/zephyr/Kconfig
Loaded configuration '/verification_workspace/zephyr/boards/snps/nsim/arc_classic/nsim_nsim_vpx5_defconfig'
Merged configuration '/verification_workspace/zephyr/samples/philosophers/prj.conf'
Merged configuration '/verification_workspace/twister-out/nsim_nsim_vpx5/samples/philosophers/sample.kernel.philosopher.fifos/zephyr/misc/generated/extra_kconfig_options.conf'

warning: ARCV2_INTERRUPT_UNIT (defined at drivers/interrupt_controller/Kconfig:8) was assigned the
value 'y' but got the value 'n'. Check these unsatisfied dependencies: ARC (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_ARCV2_INTERRUPT_UNIT and/or look up
ARCV2_INTERRUPT_UNIT in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

warning: ARCV2_TIMER (defined at drivers/timer/Kconfig.arcv2:6) was assigned the value 'y' but got
the value 'n'. Check these unsatisfied dependencies: ARC (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_ARCV2_TIMER and/or look up ARCV2_TIMER in
the menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration
Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful too.

Configuration saved to '/verification_workspace/twister-out/nsim_nsim_vpx5/samples/philosophers/sample.kernel.philosopher.fifos/zephyr/.config'
Kconfig header saved to '/verification_workspace/twister-out/nsim_nsim_vpx5/samples/philosophers/sample.kernel.philosopher.fifos/zephyr/include/generated/zephyr/autoconf.h'
CMake Error at /verification_workspace/zephyr/cmake/modules/arch_v2.cmake:36 (message):
  ARCH not defined.  Check that BOARD=nsim, is selecting an appropriate SoC
  in Kconfig, SoC=nsim_vpx5, and that the SoC is selecting the correct
  architecture.
Call Stack (most recent call first):
  /verification_workspace/zephyr/cmake/modules/zephyr_default.cmake:132 (include)
  /verification_workspace/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /verification_workspace/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
```

**Expected behavior**
No build error should happen

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: MWDT 2024.06
 - Commit SHA: recent main, 1ed04e7b49c5bec8e09f81480b7f95ae51b59b10

nashif, 2024-07-17:
@evgeniy-paltsev is you want this to go into 3.7, it needs to be marked as a release blocker with justification.

evgeniy-paltsev, 2024-07-17:
Justification for 3.7:

The postponing fix for next release will leave the `nsim/nsim_vpx5` platform completely broken for 3.7 release and LTS branch as well (so we'd need to backport fix to LTS branch later)

As the fix itself is trivial (3 code line changes) and it only changes files of one Synopsys SoC configuration (so it can't broke anything else) I'd propose to merge fix to 3.7

PR #75896 (2024-07-17): arc: nsim: vpx5: fix SoC config issue which leads to cmake error
After recent nsim SoCs & boards reorganization the `SOC_SERIES_*` config is missing for vpx5 SoC which leads to cmake errors when building against nsim/nsim_vpx5 configuration.

Fixes: #72862
Fixes: #75898
closed 2024-07-17 by nashif