zephyrproject-rtos/zephyr #75907: RFC: platform hooks to replace SYS_INIT in soc/board code

nashif, 2024-07-15:
## Introduction

We currently have different way for initializing platforms, the most popular method is to call some code during the init process that initializes the platforms (soc or board) using SYS_INIT. There are a few different ways this is being done now, for example on ARM we have `CONFIG_PLATFORM_SPECIFIC_INIT` which calls a function `z_arm_platform_init` in __start before jumping to prep_c. Looking at some implementations of this function, it is not different from what is being done using SYS_INIT. Most ARM platforms just call `SystemInit();` 

All SYS_INIT calls tend to happen at the same spot during the init process and there is nothing special about this, i.e. no dependencies beside the fact that it needs to be run very early and before device and service initialization.

This RFC introduces a new way for initializing platform and moves away from SYS_INIT and arch specific implementation (on ARM) to a more generic solution using platform hooks that are always executed at the same spot. Multiple hooks will be provided to allow early and late initialization.

### Problem description

Long term, we are planning to deprecate the use of SYS_INIT and move away from priority based initialization and instead use dependency based init process. Platform initialization is fixed and does not need to be part of what is primarily used for "services". There is visible misuse of SYS_INIT here that can be replaced with something consistent and easy to implement and debug across all architectures. 
SYS_INIT is not easy to follow and debug,  priorities are arbitrary and in some cases picked randomly....

### Proposed change

Use pre-defined hooks at different stages of the boot process to allow maximum customization in a consistent way. Move all existing hooks to one unified design on all architectures.

## Detailed RFC

- provide a hook interface defined in a header, the "API" will be something that can be implemented and not called.
- identify the hook slots during the boot process and define a hook for each slot:
  - pre-c hook: something that can be called before we jump to cstart, similar to `z_arm_platform_init` right now
  - pre-kernel: very early in cstart and before we initialize device (where we have `INIT_LEVEL_EARLY` right now)
  - post -kernel: after kernel was initialized, close to where we have `INIT_LEVEL_POST_KERNEL` right now.

Likely there are other strategic spots, some boards/socs have defined interesting priorties and places to call soc/board code, however, some of these probably can be moved one of the above.

We have two options for defining such hooks:

- Use kconfig, board/soc can use kconfig and such hooks would only run if the kconfig is defined.
- Use weak symbols

We have many platforms that have such hooks (using SYS_INIT) in both soc and board code, meaning that we should consider providing the hooks for both, so instead of having one layer, for example using `platform_init_..()`, we might need to use `soc_init...()` and `board_init...()`. 

### Proposed change (Detailed)
see https://github.com/zephyrproject-rtos/zephyr/pull/62595

### Dependencies
Not many identified so fat, we need to make the change and convert all SYS_INIT call to hooks, this is currenlty WIP. We have to identify special cases and dependencies and fix those.

### Concerns and Unresolved Questions
- What hooks and where? How many hook types do we need, how flexible do we want to be.
- Kconfig vs weak symbols
- Soc/Board or generic platform hooks
- Dependency on devices and how should this be dealt with.

## Alternatives
Continue using SYS_INIT.....

nashif, 2024-07-17:
turns out ARC has something similar to ARM, enabled with `CONFIG_ARC_EARLY_SOC_INIT` which is runs whe the systems comes out of reset...

bjarki-andreasen, 2024-08-13:
I prefer the naming schema:
```
init_<target>_post_c()
init_<target>_pre_kernel()
init_<target>_pre_main()
```
`target` being: "soc", "board", "pm", "arch" etc.
This should allow us to clearly define the hooks based on target user and level :)

closed 2024-09-09 by carlescufi