zephyrproject-rtos/zephyr #75909: CONFIG_NET_NATIVE_OFFLOADED_SOCKETS fails after DTLS handshake

brandon-exact, 2024-07-15:
**Describe the bug**

I am trying to get `CONFIG_NET_NATIVE_OFFLOADED_SOCKETS` to work with lwm2m and I have noticed that I am able to do the handshake successfully then it seems to get stuck in a loop.

**To Reproduce**

I modified the lwm2m_client sample to use the offloaded sockets API as shown. I also am using the leshan public server. I added the device to it using the default credentials in the sample.
```diff
brandon@brandon:~/fork/zephyr/samples/net/lwm2m_client (main)$ git diff
diff --git a/samples/net/lwm2m_client/boards/native_sim.conf b/samples/net/lwm2m_client/boards/native_sim.conf
index 3e21f837f7c..45c5ead6689 100644
--- a/samples/net/lwm2m_client/boards/native_sim.conf
+++ b/samples/net/lwm2m_client/boards/native_sim.conf
@@ -1,5 +1,18 @@
 CONFIG_DNS_RESOLVER=y
-CONFIG_DNS_SERVER_IP_ADDRESSES=y
-CONFIG_DNS_SERVER1="192.0.2.2"
 CONFIG_LWM2M_DNS_SUPPORT=y
-CONFIG_NET_CONFIG_MY_IPV4_GW="192.0.2.2"
+CONFIG_ETH_NATIVE_POSIX=n
+CONFIG_NET_DRIVERS=y
+CONFIG_NET_SOCKETS=y
+CONFIG_NET_SOCKETS_OFFLOAD=y
+CONFIG_NET_NATIVE_OFFLOADED_SOCKETS=y
+CONFIG_HEAP_MEM_POOL_SIZE=1024
+
+# IPv6 DAD requires lower level network interface access, below exposed socket-level access
+CONFIG_NET_IPV6_DAD=n
+
+
+CONFIG_DEBUG=y
+
+CONFIG_MBEDTLS_DEBUG=y
+CONFIG_MBEDTLS_MEMORY_DEBUG=y
+CONFIG_MBEDTLS_LOG_LEVEL_DBG=y
\ No newline at end of file
diff --git a/samples/net/lwm2m_client/prj.conf b/samples/net/lwm2m_client/prj.conf
index 0911fdf34e1..7c3425405d6 100644
--- a/samples/net/lwm2m_client/prj.conf
+++ b/samples/net/lwm2m_client/prj.conf
@@ -39,3 +39,6 @@ CONFIG_NET_CONFIG_MY_IPV4_GW="192.0.2.2"
 # Longer endpoint name might be returned in a registration reply
 CONFIG_COAP_EXTENDED_OPTIONS_LEN=y
 CONFIG_COAP_EXTENDED_OPTIONS_LEN_VALUE=40
+
+
+CONFIG_LWM2M_APP_SERVER="coaps://leshan.eclipseprojects.io:5684"
```
```bash
cd zephyr/samples/net/lwm2m_client/build
```

**Logs and console output**

It is able to resolve the domain and perform the handshake.
[image]
It seems that a [thread just calls this function forever](https://github.com/zephyrproject-rtos/mbedtls/blob/2f24831ee13d399ce019c4632b0bcd440a713f7c/library/ssl_msg.c#L5774). I copied the last few steps of the handshale...
```
00:00:00.930,003] <wrn> mbedtls: ssl_tls.c:4525: client state: MBEDTLS_SSL_SERVER_FINISHED
[00:00:00.930,003] <wrn> mbedtls: ssl_tls.c:8645: => parse finished
[00:00:00.930,003] <wrn> mbedtls: ssl_tls.c:8364: => calc finished tls
[00:00:00.930,003] <dbg> mbedtls: zephyr_mbedtls_debug: ssl_tls.c:8381: dumping 'finished output' (32 bytes)
[00:00:00.930,003] <dbg> mbedtls: zephyr_mbedtls_debug: ssl_tls.c:8381: 0000:  a7 05 47 34 84 1a 4d 78 5e 0e 95 b7 09 e0 fe e2  ..G4..Mx^.......
[00:00:00.930,003] <dbg> mbedtls: zephyr_mbedtls_debug: ssl_tls.c:8381: 0010:  30 e3 e1 a6 fe 4d ef 1d 74 21 87 71 ef 6a cf c4  0....M..t!.q.j..
[00:00:00.930,003] <inf> mbedtls: ssl_tls.c:8391: dumping 'calc finished result' (12 bytes)
[00:00:00.930,003] <inf> mbedtls: ssl_tls.c:8391: 0000:  e5 b4 83 5f 9d 51 b3 61 fe c2 eb c6              ..._.Q.a....
[00:00:00.930,003] <wrn> mbedtls: ssl_tls.c:8395: <= calc finished
[00:00:00.930,003] <wrn> mbedtls: ssl_msg.c:4189: => read record
[00:00:00.930,003] <wrn> mbedtls: ssl_msg.c:2155: => fetch input
[00:00:00.930,003] <wrn> mbedtls: ssl_msg.c:2190: next record in same datagram, offset: 14
[... 71 lines cut ...]
    at /home/brandon/fork/zephyr/subsys/net/lib/sockets/sockets_tls.c:3157
#3  0x0805e16c in ztls_poll_update_ctx (ctx=ctx@entry=0x80bf280 <tls_contexts>, pfd=0x80c1090 <sock_fds>, pev=0xf40fc214)
    at /home/brandon/fork/zephyr/subsys/net/lib/sockets/sockets_tls.c:3242
#4  0x0805e2f2 in tls_sock_ioctl_vmeth (obj=0x80bf280 <tls_contexts>, request=259, args=<optimized out>) at /home/brandon/fork/zephyr/subsys/net/lib/sockets/sockets_tls.c:3726
#5  0x0804d90e in zvfs_fdtable_call_ioctl (vtable=0x80b4fa0 <tls_sock_fd_op_vtable>, obj=0x80bf280 <tls_contexts>, request=259) at /home/brandon/fork/zephyr/include/zephyr/sys/fdtable.h:189
#6  0x0805b465 in zsock_poll_internal (fds=0x80c1090 <sock_fds>, nfds=3, timeout=...) at /home/brandon/fork/zephyr/subsys/net/lib/sockets/sockets.c:2442
#7  0x0805b57b in z_impl_zsock_poll (fds=0x80c1090 <sock_fds>, nfds=3, poll_timeout=410) at /home/brandon/fork/zephyr/subsys/net/lib/sockets/sockets.c:2489
#8  0x0807326a in zsock_poll (timeout=<optimized out>, nfds=3, fds=0x80c1090 <sock_fds>)
    at /home/brandon/fork/zephyr/samples/net/lwm2m_client/build/zephyr/include/generated/zephyr/syscalls/socket.h:416
#9  socket_loop (p1=0x0, p2=0x0, p3=0x0) at /home/brandon/fork/zephyr/subsys/net/lib/lwm2m/lwm2m_engine.c:830
#10 0x0804d790 in z_thread_entry (entry=0x80730e6 <socket_loop>, p1=0x0, p2=0x0, p3=0x0) at /home/brandon/fork/zephyr/lib/os/thread_entry.c:48
#11 0x080579e7 in posix_arch_thread_entry (pa_thread_status=0x80cf7bc <engine_thread_stack+2536>) at /home/brandon/fork/zephyr/arch/posix/core/thread.c:61
#12 0x080a5812 in nct_thread_starter (arg_el=0x80d2590) at /home/brandon/fork/zephyr/scripts/native_simulator//common/src/nct.c:364
#13 0xf7cfc157 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:447
#14 0xf7d90e08 in clone3 () at ../sysdeps/unix/sysv/linux/i386/clone3.S:111
```
**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows): linux
 - Toolchain (e.g Zephyr SDK, ...): 0.16.8
 - Commit SHA or Version used: v3.7.0-rc3

**Additional context**

It connects to my prod server fine without DTLS.

rlubos, 2024-07-16:
@brandon-exact Please check https://github.com/zephyrproject-rtos/zephyr/pull/75931

brandon-exact, 2024-07-16:
that fixed it!

PR #75931 (2024-07-27): net: sockets: tls: Fix poll update event check
In case POLLIN is set, and no new application data has been detected, the ztls_poll_update_ctx() should only return -EAGAIN if no other events are available for the socket. Otherwise, the function may end up busy-looping, in case for example POLLOUT is also monitored for the socket.

Current check verifying that was wrong, as it caused to function to return -EAGAIN even if some other events could've been reported for the socket.

Fixes #75909
PR #76369 (2024-08-03): [Backport v3.7-branch] net: sockets: tls: Fix poll update event check
Backport bfe958a7f1f683fd6261794352142c0672a9ed22 from #75931.

Fixes #75909
closed 2024-07-27 by nashif