zephyrproject-rtos/zephyr #75914: BLE MESH:compile error in proxy_cli.c

zhaishiqiang, 2024-07-16:
**Describe the bug**

For proxy feature,we always use provisioner as proxy client and node as proxy server.I perionally think that macro CONFIG_BT_MESH_GATT_PROXY means proxy server feature and macro CONFIG_BT_MESH_PROXY_CLIENT means proxy client feature. So for provisioner only enable CONFIG_BT_MESH_PROXY_CLIENT should be okey.But actually, there occur a compile error in function is_hash_equal in proxy_cli.c. Because there is no member variable "identity" in struct bt_mesh_subnet_keys when calling function bt_mesh_encrypt. For our environment of provisioner, the CONFIG_BT_MESH_GATT_PROXY is not enabled, so there is no member verisiable "identity" in struct bt_mesh_subnet_keys.

**Expected behavior**

I have review the mesh spec, and can not find any description about provisioner should act as proxy client or both proxy client and proxy server.So we expect that provisioner should compile and work success when only enable proxy client macro.

Finally, I want to known the solution of the flow questions:
1.Whether the point of "CONFIG_BT_MESH_GATT_PROXY means proxy server feature and macro CONFIG_BT_MESH_PROXY_CLIENT means proxy client feature" is corrent?
2.If 1 is corrent, can we only enable CONFIG_BT_MESH_PROXY_CLIENT for provisioner?
3.If we want only enable proxy client feature for provisioner, what should we do?

Thank you very much.

PavelVPV, 2024-08-28:
Hi @zhaishiqiang, very sorry for such late response, I was on a long leave. I've submitted a PR to fix the issue you found. Please have a look.

You are right, there is no dependency between Proxy Client and Server in the spec. This is just a programming error. You are correct that `CONFIG_BT_MESH_GATT_PROXY` enables Proxy Server and `CONFIG_BT_MESH_PROXY_CLIENT` enables Proxy Client. And yes, they should work independently. So if you want to enable Proxy Client in your provisioner, `CONFIG_BT_MESH_PROXY_CLIENT` should be sufficient.

zhaishiqiang, 2024-08-30:
Dear @PavelVPV 
Thanks for your help.

PR #77670 (2024-08-30): bluetooth: mesh: Fix proxy client compilation
When building proxy client without proxy server, it fails to find the `identity` field in the `struct bt_mesh_subnet_keys` because the latter only compiles when Proxy server is enabled. It should be possible to compile Proxy Client without Proxy server.

This commit compiles the field if any of GATT features is enabled.

Fixes #75914
commit 1fa27888 (2024-08-30): bluetooth: mesh: Fix proxy client compilation

When building proxy client without proxy server, it fails to find
the `identity` field in the `struct bt_mesh_subnet_keys` because the
latter only compiles when Proxy server is enabled. It should be
possible to compile Proxy Client without Proxy server.

This commit compiles the field if any of GATT features is enabled.

Fixes #75914
closed 2024-08-30 by nashif