zephyrproject-rtos/zephyr #75918: BLE MESH: mesh proxy server fails due to failed ext adv param update for mesh gatt advs

m-alperen-sener, 2024-07-16:
**Description**

MESH GATT advertisements attempts fails after any  bt_le_ext_adv_update_param(..) call.
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/mesh/adv_ext.c#L158

**To Reproduce**

Failing PTS test cases: with latest AutoPTS
Most of the MESH/SR/PROX tests and MESH/SR/RPR/SCN tests.
A few of them:
- MESH/SR/RPR/SCN/BV-08-C
- MESH/SR/RPR/SCN/BV-05-C
- MESH/SR/RPR/SCN/BV-04-C
- MESH/SR/PROX/BV-22-C
- MESH/SR/PROX/BV-23-C
- MESH/SR/PROX/BV-24-C
- MESH/SR/PROX/BV-04-C

**Expected behavior**

Passing PTS tests

**Impact**

Ble Mesh RPR server and Proxy Server is not qualifiable

**Logs and console output**

[00:00:33.811,035] <err> bt_mesh_adv_ext: Failed updating adv params : -105
[00:00:33.811,065] <wrn> bt_mesh_gatt: Failed to advertise using type 0x02 (err -105)
[00:00:33.811,096] <err> bt_mesh_gatt: Advertising proxy failed (err: -105)
[00:00:33.813,171] <err> bt_mesh_adv_ext: Failed updating adv params : -105
[00:00:33.813,171] <wrn> bt_mesh_gatt: Failed to advertise using type 0x02 (err -105)
[00:00:33.813,201] <err> bt_mesh_gatt: Advertising proxy failed (err: -105)
[00:00:35.372,711] <wrn> bt_mesh_crypto: Enc beacon: 0x02, 0x0000000b
[00:00:38.806,823] <wrn> bt_conn: Send auto LE param update failed (err -105)
[00:00:52.816,009] <err> bt_mesh_adv_ext: Failed updating adv params : -105
[00:00:52.816,040] <err> bt_mesh_beacon: Failed to send beacon: err -105
[00:00:52.816,375] <err> bt_mesh_adv_ext: Failed updating adv params : -105
[00:00:52.816,375] <wrn> bt_mesh_gatt: Failed to advertise using type 0x02 (err -105)
[00:00:52.816,406] <err> bt_mesh_gatt: Advertising proxy failed (err: -105)
[00:01:02.816,650] <err> bt_mesh_adv_ext: Failed updating adv params : -105
[00:01:02.816,680] <err> bt_mesh_beacon: Failed to send beacon: err -105
[00:01:02.817,108] <err> bt_mesh_adv_ext: Failed updating adv params : -105
[00:01:02.817,108] <wrn> bt_mesh_gatt: Failed to advertise using type 0x02 (err -105)
[00:01:02.817,108] <err> bt_mesh_gatt: Advertising proxy failed (err: -105)
[00:01:12.817,352] <err> bt_mesh_adv_ext: Failed updating adv params : -105
[00:01:12.817,382] <err> bt_mesh_beacon: Failed to send beacon: err -105
[00:01:12.817,718] <err> bt_mesh_adv_ext: Failed updating adv params : -105
[00:01:12.817,718] <wrn> bt_mesh_gatt: Failed to advertise using type 0x02 (err -105)
[00:01:12.817,749] <err> bt_mesh_gatt: Advertising proxy failed (err: -105)
[00:01:20.683,929] <err> bt_mesh_adv_ext: Failed updating adv params : -105
[00:01:20.683,959] <wrn> bt_mesh_gatt: Failed to advertise using type 0x02 (err -105)
[00:01:20.683,959] <err> bt_mesh_gatt: Advertising proxy failed (err: -105)
[00:01:22.817,962] <err> bt_mesh_adv_ext: Failed updating adv params : -105
[00:01:22.817,993] <err> bt_mesh_beacon: Failed to send beacon: err -105
[00:01:22.818,572] <err> bt_mesh_adv_ext: Failed updating adv params : -105
[00:01:22.818,572] <wrn> bt_mesh_gatt: Failed to advertise using type 0x02 (err -105)
[00:01:22.818,572] <err> bt_mesh_gatt: Advertising proxy failed (err: -105)

PavelVPV, 2024-11-18:
@m-alperen-sener , I guess it is due to #77241?

m-alperen-sener, 2024-11-18:
we can close this issue then, thanks.

closed 2024-11-18 by m-alperen-sener