zephyrproject-rtos/zephyr #75919: BLE MESH: dfum and mbtm fails on PTS tests

m-alperen-sener, 2024-07-16:
**Describe the bug**
Following test cases fails during PTS testing with AutoPTS
DFUM/CL/FU/BV-01-C
DFUM/CL/FU/BV-02-C
MBTM/SR/BT/BV-10-C

**To Reproduce**
Compile the tester app with those additional configuration and run the test cases listed above:
https://github.com/auto-pts/auto-pts/blob/master/autopts/bot/iut_config/zephyr.py#L233-L243

**Expected behavior**
PTS passes

**Logs and console output**
2024-07-15 10:56:43,224 LT1-thread root ERROR client.py                 1012  _run_test_case            : Error opcode in response!
Traceback (most recent call last):
  File "/home/*****/auto-pts/autopts/client.py", line 996, in _run_test_case
    response = test_case.run_next_step()
  File "/home/*****/auto-pts/autopts/ptsprojects/testcase.py", line 664, in run_next_step
    response = step(*args)
  File "/home/*****/auto-pts/autopts/ptsprojects/testcase.py", line 684, in run_wid
    my_response = self.handle_mmi_generic(wid, description, style,
  File "/home/*****/auto-pts/autopts/ptsprojects/testcase.py", line 605, in handle_mmi_generic
    response = self.generic_wid_hdl(wid, description, test_case_name)
  File "/home/*****/auto-pts/autopts/wid/mmdl.py", line 32, in mmdl_wid_hdl
    return generic_wid_hdl(wid, description, test_case_name, [__name__])
  File "/home/*****/auto-pts/autopts/wid/wid.py", line 26, in generic_wid_hdl
    response = _generic_wid_hdl(wid, description, test_case_name, module_names)
  File "/home/*****/auto-pts/autopts/wid/wid.py", line 22, in _generic_wid_hdl
    return wid_hdl(WIDParams(wid, description, test_case_name))
  File "/home/*****/auto-pts/autopts/wid/mmdl.py", line 2367, in hdl_wid_990
    btp.mmdl_dfu_update_firmware_start(addrs, slot_idx, slot_size, block_size, chunk_size, fwid, metadata)
  File "/home/*****/auto-pts/autopts/pybtp/btp/mmdl.py", line 2074, in mmdl_dfu_update_firmware_start
    (rsp,) = iutctl.btp_socket.send_wait_rsp(*MMDL['dfu_update_firmware_start'], data)
  File "/home/*****/auto-pts/autopts/pybtp/iutctl_common.py", line 272, in send_wait_rsp
    raise BTPError("Error opcode in response!")
autopts.pybtp.types.BTPError: Error opcode in response!

PavelVPV, 2025-02-01:
@m-alperen-sener , @sjanc , do these tests still fail?

m-alperen-sener, 2025-02-03:
Only MBTM/SR/BT/BV-10-C seems to be fixes but all DFU Clients tests still fail.

PavelVPV, 2025-02-10:
All 3 tests passes on my desk with:
- autopts revision: https://github.com/auto-pts/auto-pts/commit/fcc03e8757bc76c6f698d95a7fc98a56c729e961
- zephyr revision: https://github.com/zephyrproject-rtos/zephyr/commit/55d3deb030bf10198b675fe6ce2effe738f9c01b
- extra configs: https://github.com/auto-pts/auto-pts/blob/fcc03e8757bc76c6f698d95a7fc98a56c729e961/autopts/bot/iut_config/zephyr.py#L236C1-L247C56

closed 2025-02-10 by PavelVPV

sjanc, 2025-02-11:
Hmm not sure why bot is not picking proper overlay when building for those tests..

It fails for me too when run with  client bot locally, will investigate

m-alperen-sener, 2025-02-11:
The order of overlays applied seems troublesome if you look at the build log;

Example;
`Parsing /work/zephyr/tests/bluetooth/tester/Kconfig
Loaded configuration '/work/zephyr/boards/nordic/nrf54l15dk/nrf54l15dk_nrf54l15_cpuapp_defconfig'
Merged configuration '/work/zephyr/tests/bluetooth/tester/prj.conf'
Merged configuration '/work/zephyr/tests/bluetooth/tester/boards/nrf54l15dk_nrf54l15_cpuapp.conf'
Merged configuration '/work/zephyr/tests/bluetooth/tester/overlay-mesh.conf'
Merged configuration '/work/zephyr/tests/bluetooth/tester/prj.conf'
Merged configuration '/work/zephyr/tests/bluetooth/tester/build/tester/zephyr/.config.sysbuild'
Configuration saved to '/work/zephyr/tests/bluetooth/tester/build/tester/zephyr/.config'`

As you see in that build the prj.conf file is merged twice I don't know why but you might have a similar issue with the bot.

PavelVPV, 2025-02-11:
Manual build. I didn't use the bot.