zephyrproject-rtos/zephyr #75933: 3.7 Release notes - pending areas

aescolar, 2024-07-16:
This issue is to track and ping maintainers of areas which may lack release notes entries for 3.7.
Maintainers you are very welcome to reply here if there is nothing of significance to mentioned in the release notes and tick the entry in this list

### Architectures & platforms
- [ ] arc @ruuddw @abrodkin @evgeniy-paltsev
- [ ] arm @ithinuel
- [ ] NIOS-2 @nashif 
- [ ] x86 @edersondisouza 
- [ ] BeagleBoard @jadonk
- [ ] Espressif @sylvioalves 
- [ ] Ezurio @rerickson1
- [ ] GD32 @cameled @nandojve 
- [ ] Google: @fabiobaltieri 
- [ ] Infineon @ifyall 
- [ ] Intel x86 @edersondisouza 
- [ ] Intel Xtensa @dcpleung 
- [ ] Intel ISH @kwd-doodling 
- [ ] Intel Agilex @gdengi 
- [ ] Litex @tgorochowik @kgugala @fkokosinski 
- [ ] Microchip MEC @jvasanth1 
- [ ] Microchip RISC-V @fkokosinski @kgugala 
- [ ] Microchip SAM @nandojve 
- [ ] Nuvoton NPCX @MulinChao @ChiHuaL 
- [ ] Nuvoton Numicro @cyliangtw
- [ ] nRF @anangl @masz-nordic 
- [ ] opentitan @snematbakhsh 
- [ ] Panasonic @pideu-sj 
- [ ] Renesas SmartBond @ioannis-karachalios @andrzej-kaczmarek 
- [ ] Renesas RA @soburi @KhiemNguyenT 
- [ ] Renesas RZ @tgorochowik 
- [ ] Renesas R-Car @aaillet @lorc 
- [ ] Synopsys  @ruuddw 
- [ ] TI Simplelink / K3 @vaishnavachath
- [ ] Xen @povergoing 
- [ ] Xilinx @henrikbrixandersen 

### Drivers and sensors
- [ ] Auxiliary Display @thedjnK
- [ ] Audio @lyakh @lgirdwood @marc-hb @kv2019i
- [ ] bbram @yperess 
- [ ] Coredump @mrkhldn 
- [ ] Crypto @ceolin
- [ ] DAC @martinjaeger 
- [ ] DAI @kv2019i @LaurentiuM1234
- [ ] EDAC @finikorg 
- [ ] FPGA @cfriedt 
- [ ] Fuel gauge @aaronemassey @teburd
- [ ] GNSS @bjarki-andreasen 
- [ ] IEEE 802.15.4 @fgrandel @jukkar @rlubos 
- [ ] LED @Mani-Sadhasivam @simonguinot 
- [ ] Mbox @carlocaione 
- [ ] Memory management @dcpleung @carlocaione 
- [ ] MSPI @swift-tk 
- [ ] PECI @albertofloyd 
- [ ] PCI @dcpleung 
- [ ] Retained memory @nordicjm
- [ ] SMBUS @finikorg
- [ ] syscon @carlocaione 
- [ ] System timer @andyross 
- [ ] Time Aware GPIO @akanisetti
- [ ] Virtualization @tbursztyka 
- [ ] W1 @str4t0m

### Libraries / Subsystems
- [ ] ACPI @najumon1980
- [ ] Benchmarks @peter-mitsis
- [ ] Binary Descriptors @yonsch
- [ ] Bluetooth controller @cvinayak 
- [ ] Bluetooth mesh @PavelVPV @alxelax @jhedberg 
- [ ] Bluetooth classic @lylezhu2012
- [ ] C++ @stephanosio 
- [ ] C library @stephanosio 
- [ ] Cache @carlocaione 
- [ ] CI @stephanosio 
- [ ] DAP @jfischer-no 
- [ ] Debug: @nashif 
- [ ] Demand Paging @dcpleung 
- [ ] Device Driver Model @gmarull @tbursztyka 
- [ ] DFU @de-nordic @nordicjm 
- [ ] DSP subsystem @stephanosio 
- [ ] EC Host Commands @niedzwiecki-dawid 
- [ ] Emulation @yperess 
- [ ] Filesystem @de-nordic 
- [ ] Filesystems: FatFs reentrant support: @Ox11 
- [ ] Formatted output: @nordic-krch 
- [ ] Heap management @npitre @andyross 
- [ ] IPC @carlocaione @arnopo 
- [ ] JSON @d3zd3z 
- [ ] mgmt: updatehub @nandojve 
- [ ] Modbus @jfischer-no 
- [ ] OpenAMP @carlocaione 
- [ ] OSDP @sidcha 
- [ ] Picolibc : @keith-packard 
- [ ] POSIX API @cfriedt @ycsin 
- [ ] Retention @nordicjm
- [ ] RTIO @teburd 
- [ ] Task watchdog @martinjaeger 
- [ ] Test framework /ztest @nashif 
- [ ] Tracing @nashif 
- [ ] Twister @nashif 
- [ ] USB susbystem & drivers @jfischer-no @tmon-nordic
- [ ] USB-C:  @sambhurst
- [ ] Userspace @dcpleung 
- [ ] VFS @de-nordic 
- [ ] ZBus @rodrigopex

aescolar, 2024-07-16:
@bjarki-andreasen,  @rlubos,  @simonguinot,  @carlocaione,  @carlocaione,  @swift-tk,  @albertofloyd,  @nordicjm, @carlocaione,  @andyross,  @akanisetti, @tbursztyka,  @str4t0m, @najumon1980, @peter-mitsis, @yonsch, @cvinayak,  @jhedberg,  @lylezhu2012, @stephanosio,  @stephanosio,  @carlocaione,  @stephanosio,  @jfischer-no,  @tbursztyka,  @nordicjm,  @stephanosio,  @niedzwiecki-dawid,  @de-nordic, @Ox11,  @nordic-krch,  @andyross,  @arnopo,  @d3zd3z,  @jfischer-no,  @carlocaione,  @sidcha,  @keith-packard,  @ycsin,  @nordicjm, @tmon-nordic, @sambhurst, @de-nordic,  @rodrigopex,

rodrigopex, 2024-07-16:
@aescolar, Done #75938.

jadonk, 2024-07-16:
https://github.com/zephyrproject-rtos/zephyr/pull/75949

ithinuel, 2024-07-16:
Arm https://github.com/zephyrproject-rtos/zephyr/pull/75957

dcpleung, 2024-07-16:
Demand paging and Intel Xtensa (audio DSP) are at https://github.com/zephyrproject-rtos/zephyr/pull/75972.

For memory management and userspace, there are mostly internal changes which should be invisible to areas outside of them.

cyliangtw, 2024-07-17:
@aescolar, 
it's done #75980

teburd, 2024-07-17:
rtio release notes #76013

aaronemassey, 2024-07-17:
fuel gauge release notes https://github.com/zephyrproject-rtos/zephyr/pull/76009

de-nordic, 2024-07-17:
DFU, FS, VFS:
https://github.com/zephyrproject-rtos/zephyr/pull/76018
https://github.com/zephyrproject-rtos/zephyr/pull/76019

keith-packard, 2024-07-17:
picolibc unchanged since 3.6.0 (still at 1.8.6)

rerickson1, 2024-07-17:
Ezurio release notes: https://github.com/zephyrproject-rtos/zephyr/pull/76028

bjarki-andreasen, 2024-07-18:
GNSS release notes: #76049

edersondisouza, 2024-07-18:
Nothing worth mention for x86.

cfriedt, 2024-07-19:
POSIX API doc cleanup in #76114 and release notes in #76112

cfriedt, 2024-07-19:
FPGA release notes in #76116

simonguinot, 2024-07-19:
LED release notes: https://github.com/zephyrproject-rtos/zephyr/pull/76051 and https://github.com/zephyrproject-rtos/zephyr/pull/76091.

fkokosinski, 2024-07-22:
Renesas RZ/T2M: https://github.com/zephyrproject-rtos/zephyr/pull/76152.

evgeniy-paltsev, 2024-07-22:
ARC architecture and Synopsys platforms: #76183

lylezhu2012, 2024-07-23:
Bluetooth classic release notes: #76197.

str4t0m, 2024-07-25:
1-Wire release notes: #76288.

aescolar, 2024-07-26:
Closing as the release is being done right now.

closed 2024-07-26 by aescolar