zephyrproject-rtos/zephyr #75963: Zephyr PM APIs confusion

fabiobaltieri, 2024-07-16:
## Introduction

Hi, I've been working with few chips using PM and PM_DEVICE recently and found some lack of clarity in the PM APIs and how/when they should be used.

### Problem description

There's three main PM config options that control the system behavior, let's see if I got them right:

- `PM`: enables CPU power states, basically a callback that says "hey CPU, I won't need you for a while, go to sleep"
- `PM_DEVICE`: controls whether the device suspend/resume APIs are available and especially whether those `xxx_pm_action` driver callback are built or not
- `PM_DEVICE_RUNTIME`: adds reference counters on the stuff above, with a different API, if the device enables the runtime features either with `pm_device_runtime_enable` or `zephyr,pm-device-runtime-auto`
- `PM_DEVICE_SYSTEM_MANAGED`: enables the stuff in [subsys/pm/device_system_managed.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/pm/device_system_managed.c), which basically means that now the system calls suspend and resume on all devices every time the CPU goes to sleep for too long, unless they are busy or flagged for pm runtime operation

So if you are running on a system that is using `PM` and `PM_DEVICE`, the default setting is `PM_DEVICE_SYSTEM_MANAGED=y` which means that all of your devices implementing PM are going to be suspended and resume continuously as the CPU goes to sleep and resume. I guess this would be fine if the PM device API were meant to be used for that reason, like preparing drivers to wake up the CPU early if needed, but the problem is that this is not how they are used all the time.

If one does a quick grep for `PM_DEVICE_DT_INST_DEFINE` in `drivers/` there's a lot of usage of this for devices external of the CPU, displays, flash memories, GNSS, LED controllers, sensors, let's ignore all the input ones that I did while (clearly) not understanding what these were for. All of these clearly don't want to be suspended every time the CPU sleeps for a few milliseconds, but that's what happens, it's the same API. You may argue it's API misuse, but then if the device callback were meant to be hard bound to the CPU going in and out of sleep, why would `PM_DEVICE` be decoupled from `PM`?

Now one can disable `PM_DEVICE_SYSTEM_MANAGED`, but now some of the drivers that do want that callback when the CPU suspends are not going to get it anymore, and there's no way out.

So one can use `PM_DEVICE_RUNTIME` instead and ensure to all the devices that are NOT bound to the CPU have runtime PM enabled and are controlled by the application, but then this [disables](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/pm/Kconfig#L150) `PM_DEVICE_SYSTEM_MANAGED` by default... why?!

Now, `PM_DEVICE_RUNTIME`, how's that supposed to be used? There's load of drivers calling `pm_device_runtime_enable` on themselves so that they use the runtime API automatically once enabled, but then that means that once a driver gets that call introduced it changes from being active by default to suspended by default and MUST be enabled manually by the application, which inevitably causes a regression downstream. So maybe it [should not](https://github.com/zephyrproject-rtos/zephyr/pull/75862) be used in the driver? And we should rely on `zephyr,pm-device-runtime-auto` or the application calling `pm_device_runtime_enable` on the device? It seems to make more sense, but then how does one handle `pm_device_init_suspended` (which is often paired with self `pm_device_runtime_enable`).

### Proposed change

These APIs and use cases needs to be clear as a start, right now it seems like there's a bit of a mess in the code base, the PM callback are used with different scope.

The whole `PM_DEVICE_SYSTEM_MANAGED` story seems to be the immediate problem, at least it is to me right now. I think that this should not be a single option that enables/disables cpu-device hooks, maybe it should be a `PM_DEVICE_FLAG_` bit? `PM_DEVICE_FLAG_SYSTEM_MANAGED` or something?

Looking for initial feedback on this, trying to confirm if there's something I'm missing.

fabiobaltieri, 2024-07-16:
cc @ceolin @nashif @teburd @tmleman @JordanYates @gmarull @carlescufi

bjarki-andreasen, 2024-07-16:
Since I discovered this function `pm_device_driver_init`, at least from a device driver perspective it is pretty straight forward, just include `#include <zephyr/pm/device.h>` and implement the pm_action callback implementing ON and RESUME, while hiding the SUSPEND and OFF actions behind `#ifdef CONFIG_PM_DEVICE`. No idea if this is documented anywhere, I just listened to a talk on YouTube :)

gmarull, 2024-07-16:
A partial attempt to split PM_DEVICE from PM_DEVICE_RUNTIME was made a while ago https://github.com/zephyrproject-rtos/zephyr/pull/62160. PM_DEVICE should be ditched entirely, it's just nonsense. We should also consider introducing a "PM v2", and deprecate "PM v1".

fabiobaltieri, 2024-07-16:
> A partial attempt to split PM_DEVICE from PM_DEVICE_RUNTIME was made a while ago #62160. PM_DEVICE should be ditched entirely, it's just nonsense. We should also consider introducing a "PM v2", and deprecate "PM v1".

Interesting, one of the point I wrote and then dropped was asking whether DEVICE and DEVICE_RUNTIME really make sense in the way they are currently implemented. So what was the idea? `PM_DEVICE` for CPU bound and `PM_DEVICE_RUNTIME` for application controlled?

gmarull, 2024-07-16:
> Interesting, one of the point I wrote and then dropped was asking whether DEVICE and DEVICE_RUNTIME really make sense in the way they are currently implemented. So what was the idea? `PM_DEVICE` for [...]

`PM_DEVICE` -> suspend/resume devices when CPU goes to idle (ie, nonsense the way it's done)
`PM_DEVICE_RUNTIME` -> suspend/resume devices based on consumer needs

They share code, but for no good reasons IMHO, it makes things confusing (and incompatible, how can the same callback be shared for both cases?)

bjarki-andreasen, 2024-07-16:
`PM_DEVICE` from my understanding is just the resume/suspend hooks for devices? and `PM_DEVICE_RUNTIME` tracks usage, it is entirely "above" `PM_DEVICE` I believe.

ceolin, 2024-07-16:
> `PM_DEVICE` from my understanding is just the resume/suspend hooks for devices? and `PM_DEVICE_RUNTIME` tracks usage, it is entirely "above" `PM_DEVICE` I believe. Maybe the source code paints a different story

That is correct, `PM_DEVICE` is about enable the callback hook in devices. `PM_DEVICE_SYSTEM_MANAGED` and `PM_DEVICE_RUNTIME` control how device power management happens in devices.

ceolin, 2024-07-16:
> 
> `PM_DEVICE` -> suspend/resume devices when CPU goes to idle (ie, nonsense the way it's done) 
> `PM_DEVICE_RUNTIME` -> suspend/resume devices based on consumer needs

That is  simply **wrong**.  `PM_DEVICE_SYSTEM_MANAGED` is the option that suspend and resume devices. Also this behavior can be tuned, shallow power states can decide to not trigger power management action in devices using `
zephyr,pm-device-disabled`

> 
> They share code, but for no good reasons IMHO, it makes things confusing (and incompatible, how can the same callback be shared for both cases?)

The callback is for responding to an action (suspend / resume / off ).. Why can't it be shared ?

ceolin, 2024-07-16:
> ## Introduction
> Hi, I've been working with few chips using PM and PM_DEVICE recently and found some lack of clarity in the PM APIs and how/when they should be used.
>
> ### Problem description
> There's [...]

This option was introduced to get rid of the atrocity that was `PM_DEVICE_RUNTIME_EXCLUSIVE` that was enabled by default when `PM_DEVICE_RUNTIME` was enabled and to facilitate the migration was asked to keep the old behavior, so this option is disabled when `PM_DEVICE_RUNTIME` is enabled.

The idea is having direct option to select which behavior you want.

- `PM_DEVICE_SYSTEM_MANAGED` -> Devices that are not doing device runtime pm are suspended when the SoC gets idle (and the power state used has not disabled it)
- `PM_DEVICE_RUNTIME` -> suspend / resume devices at runtime and not bound to system idle.

`PM_DEVICE` -> Enable the basic infra-struct, it is used by both behaviors. 

This one we may remove or make a hidden option to avoid confusion and simplify things.

> Now, `PM_DEVICE_RUNTIME`, how's that supposed to be used? There's load of drivers calling `pm_device_runtime_enable` on themselves so that they use the runtime API automatically once enabled, but [...]

Per device ? Per specific power states ?

> 
> Looking for initial feedback on this, trying to confirm if there's something I'm missing.

ceolin, 2024-07-16:
> A partial attempt to split PM_DEVICE from PM_DEVICE_RUNTIME was made a while ago #62160. PM_DEVICE should be ditched entirely, it's just nonsense. We should also consider introducing a "PM v2", and deprecate "PM v1".

There is nothing blocking the code to change. The problem was not ditching `PM_DEVICE` the problem was that you had tried to completely remove the support for suspend/resume devices when the system is idle (what is now controlled by `PM_DEVICE_SYSTEM_MANAGED`).

henrikbrixandersen, 2024-07-16:
No matter how this all fits together, I think a good step would be to add more documentation on how the different options/behaviors are a) intended to be used by an application developer and b) intended to be implemented by the various driver types (on-chip/off-chip peripherals, SoCs).

ceolin, 2024-07-17:
> No matter how this all fits together, I think a good step would be to add more documentation on how the different options/behaviors are a) intended to be used by an application developer and b) intended to be implemented by the various driver types (on-chip/off-chip peripherals, SoCs).

Yes that is very reasonable.

gmarull, 2024-07-17:
> That is simply **wrong**. `PM_DEVICE_SYSTEM_MANAGED` is the option that suspend and resume devices. Also this behavior can be tuned, shallow power states can decide to not trigger power management [...]

This continues to be a flawed design.

> 
> > They share code, but for no good reasons IMHO, it makes things confusing (and incompatible, how can the same callback be shared for both cases?)
> 
> The callback is for responding to an action (suspend / resume / off ).. Why can't it be shared ?

Because they're entirely different scenarios?

JordanYates, 2024-07-17:
> Now, PM_DEVICE_RUNTIME, how's that supposed to be used? There's load of drivers calling pm_device_runtime_enable on themselves

That is an incorrect implementation, just because `PM_DEVICE_RUNTIME` is enabled, it doesn't mean the user wants it enabled on that particular device. Worse, there is no way to stop it from happening without modifying the driver.

> And we should rely on zephyr,pm-device-runtime-auto or the application calling pm_device_runtime_enable on the device?

Yes

> It seems to make more sense, but then how does one handle pm_device_init_suspended (which is often paired with self pm_device_runtime_enable).

As @bjarki-andreasen mentioned, this was the rationale behind `pm_device_driver_init`. It "magically" intialises the device into the lowest possible power state without the driver needing to think about it.

fabiobaltieri, 2024-07-17:
> That is an incorrect implementation, just because `PM_DEVICE_RUNTIME` is enabled, it doesn't mean the user wants it enabled on that particular device. Worse, there is no way to stop it from happening without modifying the driver.

Cool, that sounded wrong from the start, and unsurprisingly when I added that to the basic gpio input drivers I had users reaching out because their application lost the device all of a sudden.

So let's hypothetically say that I'm going drop those calls from all the input drivers, we end up with few use cases for `PM_RUNTIME` users:
- Nordic users, no `PM`, the device is now always active, does not take pm_runtime apis anymore, the users did not log and check the error codes, seemingly everything still works, user ends up with a very hard to troubleshoot power consumption regression
- `PM` user, let's say stm32, exact same story, but now the device starts suspending and unsuspending with the CPU, user ends up with a device working intermittently AND a very hard to troubleshoot power consumption regression

But even under normal development, someone implements PM support for a device, applications running with `PM` and `PM_DEVICE` had everything working, now they update the rtos and all of a sudden a device starts working intermittently. Maybe they'll find `PM_DEVICE_SYSTEM_MANAGED`, manage to shut it off, but maybe their system relied on those calls for something else. Sucks to be you, only way to decouple the two is to enable PM runtime. Don't need usage counter? Maybe you have a suspend reason mask to control the device state (we do). Sucks even more to be you.

> That is correct, `PM_DEVICE` is about enable the callback hook in devices. `PM_DEVICE_SYSTEM_MANAGED` and `PM_DEVICE_RUNTIME` control how device power management happens in devices.

Well that's not quite the whole story, the combination of `PM`, `PM_DEVICE`, `PM_DEVICE_SYSTEM_MANAGED`, how the driver is implemented and how the devices are configured in the tree control how power management happen. And some combination of those results in totally nonsense behavior. And that may change as the drivers evolve and depending on what the author believes the right way of using the API is.

Listen from what I see and read from these comments, with everyone suggesting different tricks to try and workaround the state of the API, I think it's pretty evident that the API is unclear. I think Gerard was right and the two use cases should be logically split. Don't care if they share the code or not, the implementation is not the point, but we need hooks for CPU devices (which should be ONE option, not a magic combination of three), we need an API for end devices, and the whole story should not have combinations that results in calls being done automatically by the system and random point, and especially should not be causing regression every time someone implements PM on a driver.

This should not be a thing were few people knows the tricks to make the thing work, the API needs to be defensive to misuse.

JordanYates, 2024-07-17:
> So let's hypothetically say that I'm going drop those calls from all the input drivers, we end up with few use cases for PM_RUNTIME users:
>   * Nordic users, no PM, the device is now always active, [...]

Introducting minor power consumption regressions to fix incorrect implementations hasn't been a showstopper before AFAICT, why start worrying now :)

The STM32 behavior sounds like a real pain. And at the risk of sounding self-centered, its somewhat self-inflicted (not you personally, but `PM` in general). If `PM` has no way of determining which devices should be controlled when the CPU changes states, that sounds like a `PM` problem, nothing to do with `PM_DEVICE_RUNTIME`.

bjarki-andreasen, 2024-07-17:
I believe PM_DEVICE and PM_DEVICE_RUNTIME in isolation are really useful.  The application has full control of all devices, and devices sharing busses have a useful counter to keep the bus powered while any device is active. Power domains are also automatically managed to make sure all required domains are on for any resumed device.

Any "automatic" power management of devices which is not a direct result of pm_runtime_get() or put() from the application is a bug IMO.

All most users want is to be able to power a device before using it, and power it of when done. Only get() and put() should be of interrest to the application regarding device PM

fabiobaltieri, 2024-07-17:
> Introducting minor power consumption regressions to fix incorrect implementations hasn't been a showstopper before AFAICT, why start worrying now :)

I have absolutely no problem doing this, the problem is that if we clean this up we should:
1. do it for the whole code base, get rid of every antipattern, otherwise we'll always get more, but more importantly
2. introduce a regression and break the user ONCE, which hope may be the outcome of this conversation

> If `PM` has no way of determining which devices should be controlled when the CPU changes states, that sounds like a `PM` problem, nothing to do with `PM_DEVICE_RUNTIME`.

Yes, and the problem is that now it relies on the runtime bit to work.

> I believe PM_DEVICE and PM_DEVICE_RUNTIME in isolation are really useful.

Yes, I think that there should be only two options: one for CPU power management which enables everything you need: power states and cpu related power states handlers (why on earth you would want only half of that?!), and one for application controlled, I what we use runtime now. Except for people who uses the normal API because haven't been bitten by PM messing it up for them.

ceolin, 2024-07-17:
> But even under normal development, someone implements PM support for a device, applications running with `PM` and `PM_DEVICE` had everything working, now they update the rtos and all of a sudden a device starts working intermittently.

Sorry what you mean here ? I didn't get the "update the rtos" part

> Maybe they'll find `PM_DEVICE_SYSTEM_MANAGED`, manage to shut it off, but maybe their system relied on those calls for something else. Sucks to be you, only way to decouple the two is to enable PM [...]

So, here what exactly is the need ? We have the following options:

- `CONFIG_PM_DEVICE` -> Enable support for pm hooks in devices. Only devices that do pm define it. Implicitly enable `CONFIG_PM_DEVICE_SYSTEM_MANAGED` **if** `CONFIG_PM_DEVICE_RUNTIME` is **not** enabled.
- `CONFIG_PM_DEVICE_SYSTEM_MANAGED` -> Suspend / Resume devices when the system is idle. Only in devices that don't have device runtime pm enabled. It is disabled by default if `CONFIG_PM_DEVICE_RUNTIME` is enabled. It is possible to select which power states (idle states) trigger these actions, but it  will do it for **all** devices that support device power management **and** don't have device runtime enabled **and** are not busy.
- `CONFIG_PM_DEVICE_RUNTIME` -> Enable support for device runtime pm, by default it disables `CONFIG_PM_DEVICE_SYSTEM_MANAGED`. It imposes additional resources in all devices that have support for device power management.

These options are **NOT** mutually exclusive, it is possible to have both enabled, `CONFIG_PM_DEVICE_SYSTEM_MANAGED` and `CONFIG_PM_DEVICE_RUNTIME`.

From a device that implements power management, it can:

1) Rely on system to suspend / resume when it is idle
2) Enable device runtime PM and controls when it should be suspended / resumed

May a device that implements power management don't need any of these behaviors ? Calling directly `pm_device_action_cb`  is risky and should not be used since it may be in use and lead to inconsistencies in device runtime or busy status for example.

Some initial thoughts based on this discussion:

1) Split the logic between device system managed and device runtime. Drivers to implement two actions callbacks:
   - `typedef int (*pm_device_runtime_action_cb_t)(const struct device *dev, enum pm_device_action action);`
   - `typedef int (*`)(const struct device *dev, enum pm_device_action action, const struct pm_state_info *state);` -> the state here is the one the system is going to sleep in case of `suspend` or is the one it woke up in case of `resume`.

They may be exactly the same in most cases but it gives the opportunity for wake up devices do something different without need to use notifiers. It also provides more flexibility for devices, since they can implement only one of these callbacks. For example, a driver can implement only the device runtime callback and it will not be suspend / resumed by the system even if it has not enabled device runtime pm.

2) Having different callbacks we can get rid of `PM_DEVICE`. Basically you enable `CONFIG_PM` and devices hooks are available.

3) The available options will be:
    - `CONFIG_PM` -> System power management and device hooks.
    - `CONFIG_PM_DEVICE_RUNTIME` -> Implies in additional resources per device to support runtime power management.
    - `CONFIG_PM_DEVICE_SYSTEM_MANAGED` -> When disabled the subsystem will never call `pm_device_action_cb_t`

4) At start keep the same logic between system manged / runtime pm, if a device has runtime pm enabled the system will not call `pm_device_runtime_action_cb_t`

I am playing around with these ideas to see how effective they are. Early comments are welcomed :)

fabiobaltieri, 2024-07-18:
@ceolin Thanks, yes the split logic you are describing resonates with what I had in mind.

```c
typedef int (*)(const struct device *dev, enum pm_device_action action, const struct pm_state_info *state);
```

Great, wondering if this even needs `action` or could just get away with `state`, maybe renamed to `next_state` to make it clear, bit of an implementation detail really.

> 2. Having different callbacks we can get rid of `PM_DEVICE`.

Really like the idea of removing some option, make it less likely to pick suboptimal configurations (why would you want only half power management...)

I would go as far as considering renaming the options to better describe the intent, maybe `PM` could be `PM_SYSTEM` and `CONFIG_PM_DEVICE_RUNTIME`, well it could be `PM_DEVICE` but that would be confusing as hell.

> It also provides more flexibility for devices, since they can implement only one of these callbacks.

Yes that's what I'd like, the "update the rtos" bit I mentiond is, for example, if https://github.com/zephyrproject-rtos/zephyr/pull/75678 getes merged as is and PM users upgrades (ITE, NPCX, STM32...) they would either get the device autosuspending with the CPU, or the device starting as suspended. If we have split callbacks and get rid of `pm_device_runtime_enable` then that would not happen.

Actually at this point `pm_device_runtime_enable` does not make much sense, if a device has the callback, that's it. Maybe we can instead have only the devicetree property `zephyr,pm-device-runtime-auto`, which would be more like `zephyr,pm-device-init-suspended`. Now there's no risk of a driver being updated and changing behavior (if we actually drop that API), and one would only control the device PM initial status with devicetree. (@JordanYates what do you think?)

> The available options will be:
> * CONFIG_PM_DEVICE_SYSTEM_MANAGED

Do we even need this at this point? If you have  cpu states and system PM callbacks, is there a case where you woud NOT want them called? I think this should go and at this point it's safe to always have it enabled. Unless you are thinking about saving some code by dropping those loops.

Thanks again for going through this.

JordanYates, 2024-07-18:
> Maybe we can instead have only the devicetree property `zephyr,pm-device-runtime-auto`, which would be more like `zephyr,pm-device-init-suspended`. Now there's no risk of a driver being updated and [...]

Dropping `pm_device_runtime_enable` would be fine with me, I've never had a reason to use it because it implicitly locks the code to platforms that have that node. The proposed change to the devicetree property isn't great because if you're using `pm_device_driver_init` (you should be), the device can boot into either `suspended` or `off`, depending on the rest of the devicetree.

fabiobaltieri, 2024-07-18:
@JordanYates I'm not familiar with the power domain stuff, but good point, that should be considered as well.

So what's the "correct" pattern in your experience? Unconditionally implement the `pm_control` callback with suspend/resume if `PM_DEVICE` and `TURN_ON/OFF` all the time and always initialize with `pm_device_driver_init`?

There's one driver in the tree using that right now and your name is all over it :-) I'd like if we can converge to a "right way" of using the PM API and then fix up the code base so all drivers adopt it.

JordanYates, 2024-07-19:
> @JordanYates I'm not familiar with the power domain stuff, but good point, that should be considered as well.
>
> So what's the "correct" pattern in your experience? Unconditionally implement the [...]

Yes, that is the correct usage. Historically I was using limited upstream sensor drivers, so I wasn't pushing many changes, but that is changing now. I have a few more PR's pending that use the same pattern:
 * #75886
 * #75920
 * #75921

ceolin, 2024-07-19:
There is something that be misleading without `pm_device_runtime_enable`. If a driver **not** initialize `suspended` or `off` (it is active). What is going to be its reference count ? if we start with `1` this will never get balanced, if we start with `0` the first time someone calls `pm_device_runtime_get` it will trigger the driver pm callback asking to resume. Not to say that is confusing having a device supporting runtime pm that is active but its reference count is 0.

Dropping this function would be ok if  we assume that all devices that support runtime pm are using it and them initialize suspended.

bjarki-andreasen, 2024-07-19:
@ceolin For devices which draw tons of power, say GNSS and cellular modems, The drivers I have written initialize these as suspended, as starting them is wasteful if they are not to be used immediately (which is often the case for trackers, power gnss, get fix, suspend gnss, power cellular, send fix). Aside from waste, starting multiple power hungry devices may overwhelm the power supply of a board, which may not be designed to run with multiple of these devices at the same time.

IMO, if PM_DEVICE is enabled, devices should be initialized suspended, and the application should be powering them as they become needed.

ceolin, 2024-07-19:
> Great, wondering if this even needs `action` or could just get away with `state`, maybe renamed to `next_state` to make it clear, bit of an implementation detail really.

That is not necessarily the next state, if the system is resuming the next is always `ACTIVE` it may be more useful to have know from which state it is resuming. But I am not entirely sure if that is needed.

> Really like the idea of removing some option, make it less likely to pick suboptimal configurations (why would you want only half power management...)
>
> I would go as far as considering renaming the [...]

That is a valid question. This is a long time existent option in Zephyr I am not sure if there aren't use cases where an application uses only cpu states that don't cause execution context lost and because of this the application can just disable this behavior.

ceolin, 2024-07-19:
> @ceolin For devices which draw tons of power, say GNSS and cellular modems, The drivers I have written initialize these as suspended, as starting them is wasteful if they are not to be used [...]

100% agree.

fabiobaltieri, 2024-07-19:
> For devices which draw tons of power, say GNSS and cellular modems, The drivers I have written initialize these as suspended, as starting them is wasteful if they are not to be used immediately

Sure, they may also put the device in a brownout reset loop on a weak battery, where you'd rather start with everything off, detect the condition and stop, blink some scary red led or something, that said, this behavior is giant hassle for samples, demos and prototyping.

I think it would be ideal to have the driver being flagged for init suspended in the devicetree, so the user has a choice. If the reworked PM APIs incorporated that for good it'd be ideal, at that point I don't see any use case for enabling that in runtime.

ceolin, 2024-07-22:
I will be out on vacation next week, but when I get back I will come up with a draft addressing the ideas discussed here.

fabiobaltieri, 2024-07-22:
Thanks a lot Flavio, and everyone else for the inputs so far, just about to start the 4.0 cycle, could not have been better timing to figure this out.

albertofloyd, 2024-07-25:
cc @scottwcpg

teburd, 2024-07-25:
I’d note that perhaps with the updated sensor API power management could be done better by the driver itself. But this is something I need to explore

fabiobaltieri, 2024-07-26:
@JordanYates (and everyone really) one more question: I've bumped info few drivers using PM runtime for their internal state, `gpio/gpio_stm32.c`, `i2c/i2c_ll_stm32.c`, `i2c/i2c_nrfx_twim.c`... there's quite a few.

Do you think that's a legitimate use of the API? It feels wrong to me, can't see why those driver needs the pm runtime stuff to control their internal state rather than just calling what they need to call directly, the internal state is then exposed with the application that can end up messing with it.

JordanYates, 2024-07-26:
> @JordanYates (and everyone really) one more question: I've bumped info few drivers using PM runtime for their internal state, `gpio/gpio_stm32.c`, `i2c/i2c_ll_stm32.c`, `i2c/i2c_nrfx_twim.c`... [...]

This is a question that needs to be asked at the API level, not the driver level, otherwise users cannot write portable code.
For example, I think we have general agreement that drivers implementing  the flash API should be calling the PM functions internally, as in #73246.

Does the same behavior make sense for I2C? For the standard API maybe, but with the RTIO drivers I'm not sure there is a great location to call the PM functions automatically.

The question with "the drivers just calling what they need to call directly" is how do they handle a PM request in the middle of their internal transitions? Going through the PM API's means drivers don't need to worry about details like that.

teburd, 2024-07-26:
> This is a question that needs to be asked at the API level, not the driver level, otherwise users cannot write portable code. For example, I think we have general agreement that drivers implementing [...]

With RTIO so long as there's work in the queue of I/O work to do, the driver could be on (get on first work item, put after last). Arguably more efficient even then the current implementations which would get/put after each i2c_transfer() equivalent transaction

ceolin, 2024-07-29:
> This is a question that needs to be asked at the API level, not the driver level, otherwise users cannot write portable code. For example, I think we have general agreement that drivers implementing [...]

There is exactly the point, specially with think on concurrent usage of a driver, e.g two different threads using it. The PM runtime API helps to keep the driver in a consistent state

ceolin, 2024-08-19:
So, I am working through items listed here. Let me try to summarize the problems
and possible changes:

### Confusion / Complexity due many configuration (`CONFIG_PM`, `CONFIG_PM_DEVICE`, `CONFIG_PM_DEVICE_SYSTEM_MANAGED`,` CONFIG_PM_DEVICE_RUNTIME`)

   The proposal is to have only two symbols, `CONFIG_PM` and `CONFIG_PM_DEVICE_RUNTIME`.
   These options can be used together or independently.

   `CONFIG_PM` will cover system idle states and also what is covered now by `CONFIG_PM_DEVICE`
   and `CONFIG_PM_DEVICE_SYSTEM_MANAGED`.

### Drivers don't have clear knowlodge if they are being asked to suspend/resume  because the system is idle or due device runtime pm (pm_device_runtime_get/put)

   Adding two more actions, `PM_DEVICE_ACTION_RUNTIME_SUSPEND` and `PM_DEVICE_ACTION_RUNTIME_RESUME`. These actions will be used *only* by device runtime pm (pm_device_runtime_get/put).

   Most cases, if a driver support both methods (system managed device pm and device runtime pm) the driver
   will handle them the same way:

   ```
        switch (action) {
        case PM_DEVICE_ACTION_RESUME:
                __fallthrough;
        case PM_DEVICE_ACTION_RUNTIME_RESUME:
                ... 
                break;
        case PM_DEVICE_ACTION_SUSPEND:
                __fallthrough;
        case PM_DEVICE_ACTION_RUNTIME_SUSPEND:
                ...
                break;
   ```

   But it allows, for the ones that need it, handle these actions differently or to support only one model.

###  Devices may need to take special actions depending on which power state the system is suspending. e.g if the SoC is transitioning to `PM_STATE_SUSPEND_TO_RAM`, a device that is used as wake up source may need to take a special action like change pin control.

   Changing the action callback to

   ```
   typedef int (*pm_device_action_cb_t)(const struct device *dev,
				     enum pm_device_action action,
				     const struct pm_state_info *soc_state);
   ```

   where `soc_state` is the state that the system is going to when action == `PM_DEVICE_ACTION_SUSPEND`
   and it the state that the system is returning from when action == `PM_DEVICE_ACTION_RESUME`.

   In the `PM_STATE_SUSPEND_TO_RAM` example, it would be something like:

   ```
   pm_device_system_action_run(dev, PM_DEVICE_ACTION_SUSPEND, PM_STATE_SUSPEND_TO_RAM);
   pm_device_system_action_run(dev, PM_DEVICE_ACTION_RESUME, PM_STATE_SUSPEND_TO_RAM);
   ```

### Improper usage of `pm_device_action_run`

   All runtime pm should be done using device runtime pm (pm_device_runtime_get/put), changing
   a device state manually may result in incosistencies.

   We should remove or make it private. Any usage of this function out of the pm subsys must change
   to use device runtime pm.
   

@fabiobaltieri @JordanYates @bjarki-andreasen @teburd at all ^

ceolin, 2024-08-19:
@mmahadevan108 @erwango ^

fabiobaltieri, 2024-08-30:
@ceolin thanks for the proposal, I think it's a step forward splitting system and runtime, would go as far as suggesting renaming the current actions from `PM_DEVICE_ACTION_RESUME` to `PM_DEVICE_ACTION_SYSTEM_RESUME` (same for suspend).

Since this is going to require reworking the existing drivers, I'm wondering, would it be possible to remove the `pm_device_runtime_enable` API and only use `zephyr,pm-device-runtime-auto`?

This would remove the use case where a driver with no PM support works normally in an application with PM_RUNTIME enabled, then the driver implements the PM api and existing applications have a regression because now the driver initializes in suspend state.

bjarki-andreasen, 2024-08-30:
With the proposal to solely use `pm_device_runtime_get/put`, we can optimize the device drivers to just have two actions: `resume` and `suspend` (enabled/disabled).

A device can only do two things, it can be resumed, it can be suspended. That is all device drivers should implement. Why they are asked to be active or not is not relevant to them.

With runtime being used to control the devices, which is a necessity in a system which has busses and power domains (IE all of them), the actions `PM_DEVICE_ACTION_TURN_OFF` and `PM_DEVICE_ACTION_TURN_ON` no longer make sense since a domain can't just be put down. Power domains go off when all child devices are suspended, and can only be kept alive by an extra `get()`.

We can shrink down PM_DEVICE (which covers current PM_DEVICE and PM_DEVICE_RUNTIME) into a single call in drivers and power domains:
```
int (*pm_action_callback)(const struct device *dev, bool enable);
```

ceolin, 2024-08-30:
> @ceolin thanks for the proposal, I think it's a step forward splitting system and runtime, would go as far as suggesting renaming the current actions from `PM_DEVICE_ACTION_RESUME` to `PM_DEVICE_ACTION_SYSTEM_RESUME` (same for suspend).

I can do it if it makes more clear. I had only introduced `PM_DEVICE_ACTION_RUNTIME_*` to distinguish from the system managed actions, but we can make it even more explicit.

> Since this is going to require reworking the existing drivers, I'm wondering, would it be possible to remove the `pm_device_runtime_enable` API and only use `zephyr,pm-device-runtime-auto`?
>
> This [...]

Sounds good to me. Will take a look if there is any bigger implication doing it. @JordanYates  ^

> 
> (apologies for the delay in the reply, got caught into other things)

No worries, thanks for coming back to this.

bjarki-andreasen, 2024-09-02:
> There is a difference between a device being suspended because it is not active and having the its power cut (because of the power domain it is was suspended). The thing is that you can have N devices under a power domain that is active because there is one device still active. When this device is suspended the domain is also suspended which means that the power supply for all N devices was cut, in this situation all N-1 devices that were already suspend may need to take additional action about it.

Right, with this in mind, we have 3 states, and 4 events (actions) that can be received according to this "state machine":
* State: OFF
  * Event: ON -> initialize device and enter INACTIVE state
* State: INACTIVE
  * Event: RESUME -> resume device
  * Event: OFF -> deinit device
* State: ACTIVE
  * Event: SUSPEND -> suspend device

Is there anything missing from the above "state machine"?

erwango, 2024-09-03:
> Right, with this in mind, we have 3 states, and 4 events (actions) that can be received according to this "state machine":

What about handling S2RAM state ?
- Device has been previously initialized either with boot time configuration (from ROM) or user configuration (saved in RAM) but registers are erased due to power loss which happened while device was either INACTIVE (ACTIVE also maybe ? Tbc)
- Event is RESUME

bjarki-andreasen, 2024-09-03:
> What about handling S2RAM state ?
>
> * Device has been previously initialized either with boot time configuration (from ROM) or user configuration (saved in RAM) but registers are erased due to power [...]

S2RAM matches the INACTIVE state. From here it can be resumed when the system exits S2RAM. if power is lost, all state information is lost, so the device will implicitly be in OFF state and receive ON event at boot.

erwango, 2024-09-03:
> S2RAM matches the INACTIVE state. From here it can be resumed when the system exits S2RAM.

So you mean: up to the device to know that registers have to be reinitialized on RESUME ? I don't see it too easy without API, as registers loss can happen asynchronously.
For instance:
- Device goes from ACTIVE to INACTIVE for some reason.
- Core lives its life and goes RUN>S2IDLE>RUN>S2RAM>RUN>S2IDLE>RUN (no information sent to device about these transitions > Maybe I'm wrong here ?)
- RESUME is received by device (still in INACTIVE). How can it know it has to restore registers ?

bjarki-andreasen, 2024-09-03:
> Device goes from ACTIVE to INACTIVE for some reason.

Devices can't change states unprompted, they just follow whatever the application or some other higher level subsystem tells them.

> * Core lives its life and goes RUN>S2IDLE>RUN>S2RAM>RUN>S2IDLE>RUN (no information sent to device about these transitions > Maybe I'm wrong here ?)

The application or a higher level subsystem will be adjusting the device's state to adhere to whatever system state/power mode/power target the application or higher level subsystem provides.

> RESUME is received by device (still in INACTIVE). How can it know it has to restore registers ?

This is not possible :) if at any time it lost power, it would have received the OFF event and have de initialized its registers, it will receive ON followed by RESUME if the application/higher level subsystem resumes it using `get()`

erwango, 2024-09-05:
> Devices can't change states unprompted, they just follow whatever the application or some other higher level subsystem tells them.

Sure. I meant "Device goes from ACTIVE to INACTIVE" ... "based on some external directive" (coming from application or some other higher level subsystem)

> The application or a higher level subsystem will be adjusting the device's state to adhere to whatever system state/power mode/power target the application or higher level subsystem uses.

So, you mean that, in any case, device is informed of system state change ?

> This is not possible :) if at any time it lost power, it would have received the OFF event and have de initialized its registers, it will receive ON followed by RESUME if the application/higher level subsystem resumes it using get()

To be exact here, device can lose power while remaining suspended. That's my understanding of S2RAM. Application will resume from where it was suspended and we don't expect it would reinit devices (with for instance uart run time configuration). Otherwise, why bothering to save and restore CPU context. See :
`The CPU context is usually the minimum set of CPU registers which content must be restored on resume to let the platform resume its execution from the point it left at the time of suspension.`
So device is expected to be able to restore its state (registers configuration) on its resume following a S2RAM operation on system side.

Note that ambiguity is allowed as we don't have much documentation on S2RAM feature: https://docs.zephyrproject.org/latest/search.html?q=S2RAM&check_keywords=yes&area=default

bjarki-andreasen, 2024-09-05:
> > The application or a higher level subsystem will be adjusting the device's state to adhere to whatever system state/power mode/power target the application or higher level subsystem uses.
> 
> So, you mean that, in any case, device is informed of system state change ?

I define a system state as a high level state which dictates the power states of multiple devices, some being active, some being suspended, some being off. Any single device has no idea what any other device is doing (unless it on a bus in which case the device does keep the bus it is on resumed if needed).

With this definition, the device is not informed of the system state, only the state the system requires the device to be in.

> To be exact here, device can lose power while remaining suspended. That's my understanding of S2RAM.

I understand suspended to mean the device preserves its configuration, its just not active. For a UART peripheral, this means disabling the clock to it, while maintaining power to its HW registers. Apply the clock again and it resumes as if nothing happened. Loosing power however, what I describe as `OFF`, means no clock, and no power to registers. The UART is effectively hard reset to boot values. Applying a clock to it as this point does nothing, it has to be reinitialized from ROM/RAM by the application.

S2RAM would disable the clock to the UART, or, it could store the last runtime configuration in persistent RAM, completely power down the UART, then reinit it with its last configuration read from persistent RAM once resumed.

How exactly the kernel/system continues after S2RAM, and where previous configurations/states are stored, is platform specific/ambiguous I think.

ceolin, 2024-09-06:
> Right, with this in mind, we have 3 states, and 4 events (actions) that can be received according to this "state machine":
>
>     * State: OFF
>
>       * Event: ON -> initialize device and enter [...]
The states are what we have now (we have though an additional state for the transition between active / suspended). The actions is that are causing confusion since the driver can't distinguish between device runtime pm and system managed pm. For many drivers it just not matter it will do the same thing for equivalent actions but for devices that just work in one of them different actions make sense.

ceolin, 2024-09-06:
> I define a system state as a high level state which dictates the power states of multiple devices, some being active, some being suspended, some being off. Any single device has no idea what any [...]

That is what I had in mind too, but not the reality right now. A device is just called with a turn off action if it is under a power domain and AFAIK we are not mapping SoC power domains so devices are just be calling with suspend action.

But the logic is what I had in mind.

I think to be productive here we have to tackle it in different entries. The first problem is the confusion between system managed device power management and device runtime pm. For this we can either having multiple callbacks or multiple actions. The latter is less expensive and is proposed in https://github.com/zephyrproject-rtos/zephyr/issues/75963#issuecomment-2297356146

The other problem is the excessive build options. It seems that the solution is to have two symbols, `CONFIG_PM` (that is `CONFIG_PM` + `CONFIG_PM_DEVICE` + `CONFIG_PM_DEVICE_SYSTEM_MANAGED`) and `CONFIG_PM_DEVICE_RUNTIME` (that is `CONFIG_PM_DEVICE` + `CONFIG_PM_DEVICE_RUNTIME`)

bjarki-andreasen, 2024-09-06:
> AFAIK we are not mapping SoC power domains so devices are just be calling with suspend action.

We could be mapping them though :)

>I think to be productive here we have to split the problem. The first problem is the confusion between system managed device power management and device runtime pm. For this we can either having multiple callbacks or multiple actions. The latter is less expensive and is proposed in https://github.com/zephyrproject-rtos/zephyr/issues/75963#issuecomment-2297356146

I don't think we should mix system power states like S2RAM, which is an ambiguous high level state dictating the states of multiple devices and domains, like clocks, RAM, cores, and indeed some devices like RTCs, UARTs, GPIO wakeup sources etc., with device power states.

Devices can be off, be suspended, be active. Devices should not be informed what the ambiguous high level state of the system is at any time, its not relevant.

>The other problem is the excessive build options. It seems that the solution is to have two symbols, CONFIG_PM (that is CONFIG_PM + CONFIG_PM_DEVICE + CONFIG_PM_DEVICE_SYSTEM_MANAGED) and CONFIG_PM_DEVICE_RUNTIME (that is CONFIG_PM_DEVICE + CONFIG_PM_DEVICE_RUNTIME)

do we agree that `CONFIG_PM` would depend on `CONFIG_PM_DEVICE_RUNTIME`? and `CONFIG_PM_DEVICE` does not really belong in `CONFIG_PM`, since it is effectively `CONFIG_PM_DEVICE_RUNTIME`?

An application should be fully power manageable using only `CONFIG_PM_DEVICE_RUNTIME`, without any higher level subsystem dictating system power states.

bjarki-andreasen, 2024-09-06:
> The states are what we have now (we have though an additional state for the transition between active / suspended). The actions is that are causing confusion since the driver can't distinguish [...]

Do you have an example where a device would behave differently depending on if it is managed by device runtime pm or system managed pm?

ceolin, 2024-09-06:
> > AFAIK we are not mapping SoC power domains so devices are just be calling with suspend action.
> 
> We could be mapping them though :)

Yep :)

> 
> > I think to be productive here we have to split the problem. The first problem is the confusion between system managed device power management and device runtime pm. For this we can either having multiple callbacks or multiple actions. The latter is less expensive and is proposed in [#75963 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/75963#issuecomment-2297356146)
> 
> I don't think we should mix system power states like S2RAM, which is an ambiguous high level state dictating the states of multiple devices and domains, like clocks, RAM, cores, and indeed some devices like RTCs, UARTs, GPIO wakeup sources etc., with device power states.

That isn't exactly the point here ? The idea is not mixing power states, devices will continue to have states like "ACTIVE", "SUSPENDED" and "OFF". What is changing is that more actions can lead the device to change to these states.

The reason is because a SoC power state can trigger it. For example, in STM32L562xx the RNG device can be "ACTIVE" or "SUSPENDED" when the SoC state is "RUN" or "SLEEP" but it is "OFF" for all other states like "STOP0...2".

>
> Devices can be off, be suspended, be active. Devices should not be informed what the ambiguous high level state of the system is at any time, its not relevant.

Agree about device states, but the system state can be relevant because in some situations like if the device is used as a wake up source and needs to change its pinctrl state.

> 
> > The other problem is the excessive build options. It seems that the solution is to have two symbols, CONFIG_PM (that is CONFIG_PM + CONFIG_PM_DEVICE + CONFIG_PM_DEVICE_SYSTEM_MANAGED) and CONFIG_PM_DEVICE_RUNTIME (that is CONFIG_PM_DEVICE + CONFIG_PM_DEVICE_RUNTIME)
> 
> do we agree that `CONFIG_PM` would depend on `CONFIG_PM_DEVICE_RUNTIME`? and `CONFIG_PM_DEVICE` does not really belong in `CONFIG_PM`, since `CONFIG_PM_DEVICE` effectively is `CONFIG_PM_DEVICE_RUNTIME`?

Nope. `CONFIG_PM` is related to SoC power states, `CONFIG_PM_DEVICE_RUNTIME` is for devices. `CONFIG_PM` does not depend on `CONFIG_PM_DEVICE_RUNTIME`.

`CONFIG_PM_DEVICE` is not `CONFIG_PM_DEVICE_RUNTIME`, the way it is now it is the base for `CONFIG_PM_DEVICE_RUNTIME` and `CONFIG_PM_DEVICE_SYSTEM_MANAGED`. 

The idea is simplify it, `CONFIG_PM_DEVICE` can go away and we have `CONFIG_PM` that changes SoC states and also trigger device actions when this happens since some SoC power states power off  some devices.

`CONFIG_PM_DEVICE_RUNTIME` you get `pm_device_runtime_get/put` and is independent of SoC power states (`CONFIG_PM`).

Having more actions allows a device  to distinguish these methods.
> 
> A system should be fully power manageable from an application using only `CONFIG_PM_DEVICE_RUNTIME`, without any higher level subsystem dictating system power states.

I am not understanding it. What you mean with system power states here ? and about the application ? You are proposing that the SoC does not go any sleep state when it is idle unless it is explicitly requested by the application ?

Applications always have full control over the system. The application can set power constraints, latency request and events to change the idle behavior.

ceolin, 2024-09-06:
> Do you have an example where a device would behave differently depending on if it is managed by device runtime pm or system managed pm?

Devices that are used as wake up source may need to change its configuration properly when the SoC is deep sleeping but they don't need to do this when the system is active or in a shallow sleep state.

I don't think that is a common use case though. The biggest  thing here is that a device that is supposed to work only with device runtime pm but the the application was built without supporting it (`CONFIG_PM_DEVICE_RUNTIME=n`) won't be accidentally suspend when the SoC gets idle.

bjarki-andreasen, 2024-09-07:
I can not make sense of the following:

> The biggest thing here is that a device that is supposed to work only with device runtime pm but the application was built without supporting it (`CONFIG_PM_DEVICE_RUNTIME=n`) won't be accidentally suspend when the SoC gets idle.

All a device can and should do, is implement the pm callback, how can `CONFIG_PM_DEVICE_RUNTIME=n` affect the device driver? (unless you are specifically pointing out the case where a device driver itself is controlling its bus using runtime PM).

> Devices that are used as wake up source may need to change its configuration properly when the SoC is deep sleeping but they don't need to do this when the system is active or in a shallow sleep state.

I would argue that for single device drivers, configuring them as wakeup sources from a device pm perspective just means keeping them resumed while the system is off. Configuring whatever trigger they can produce, like an RTCs alarm, or a sensors trigger, is a different story entirely.

bjarki-andreasen, 2024-09-07:
>>I don't think we should mix system power states like S2RAM, which is an ambiguous high level state dictating the states of multiple devices and domains, like clocks, RAM, cores, and indeed some devices like RTCs, UARTs, GPIO wakeup sources etc., with device power states.

> That isn't exactly the point here ? The idea is not mixing power states, devices will continue to have states like "ACTIVE", "SUSPENDED" and "OFF". What is changing is that more actions can lead the [...]

Okay, we seem to agree here, I just don't understand why we need to add more actions than the 4 we support now. Maybe I'm misunderstanding that? The RNG device here would only receive the actions `RESUME`, `SUSPEND`, `ON` and `OFF`, it does not need to know what the SoC state is right? It only needs to know what state it is requested to be in. Its up the some higher level subsystem to manage when the RNG should be active in, according to the SoC state, not the device itself.

> I am not understanding it. What you mean with system power states here ? and about the application ? You are proposing that the SoC does not go any sleep state when it is idle unless it is [...]

I am proposing that the application dictates which devices should be resumed or suspended, and SoC should try to enter the lowest power state it can, while keeping the devices the application wants to use powered.

For example:
* Application wants to use RNG device
* Application resumes RNG device using pm_device_runtime_get()
* SoC checks which minimum power state is required to resume RNG device (clocks, memory domains, etc.)
* SoC enters said minimum power state
* pm_device_runtime subsystem calls RNG pm callback with action RESUME

With this system design, the application is in full control of system, only dictating which devices it needs at any given time. Then:

* Application done with RNG device
* Application suspends RNG device using pm_device_runtime_put()
* pm_device_runtime subsystem calls RNG pm callback with action SUSPEND
* SoC checks which minimum power state it can to to now that RNG device is suspended
* SoC enters said minimum power state

To clarify here, the system state can not trigger suspend, the system state depends on the application triggering suspend.

ceolin, 2024-09-09:
> I can not make sense of the following:
>
>
> All a device can and should do, is implement the pm callback, how can `CONFIG_PM_DEVICE_RUNTIME=n` affect the device driver? (unless you are specifically [...]

Yes that is correct. Zephyr also provides a simpler way for device power management that is suspend/resume devices when the SoC is idle (which state triggers this action is configurable). 

> 
> > Devices that are used as wake up source may need to change its configuration properly when the SoC is deep sleeping but they don't need to do this when the system is active or in a shallow sleep state.
> 
> I would argue that for single device drivers, configuring them as wakeup sources from a device pm perspective just means keeping them resumed while the system is off. Configuring whatever trigger they can produce, like an RTCs alarm, or a sensors trigger, is a different story entirely. This is done from the application using the RTC and Sensor/Sensing APIs respectively.

ceolin, 2024-09-09:
> > I am not understanding it. What you mean with system power states here ? and about the application ? You are proposing that the SoC does not go any sleep state when it is idle unless it is explicitly requested by the application ?
> > Applications always have full control over the system. The application can set power constraints, latency request and events to change the idle behavior.
> 
> I am proposing that the application dictates which devices should be resumed or suspended, and SoC should try to enter the lowest power state it can, while keeping the devices the application wants to use powered.
> 
> For example:
> 
>     * Application wants to use RNG device
> 
>     * Application resumes RNG device using pm_device_runtime_get()
> 
>     * SoC checks which minimum power state is required to resume RNG device (clocks, memory domains, etc.)
> 
>     * SoC enters said minimum power state
> 
>     * pm_device_runtime subsystem calls RNG pm callback with action RESUME
> 
> 
> With this system design, the application is in full control of system, only dictating which devices it needs at any given time. Then:
> 
>     * Application done with RNG device
> 
>     * Application suspends RNG device using pm_device_runtime_put()
> 
>     * pm_device_runtime subsystem calls RNG pm callback with action SUSPEND
> 
>     * SoC checks which minimum power state it can to to now that RNG device is suspended
> 
>     * SoC enters said minimum power state
> 
> 
> To clarify here, the system state can not trigger suspend, the system state depends on the application triggering suspend. Devices themselves just get told to be resumed/suspended, they are oblivious to any higher level dependencies.

I overall liked it, we have APIs to set power constraints in all power states that trigger power loss in a device but this API has to be explicitly called. Automate it with device runtime sounds a good approach.

I think this behavior is a good one indeed, there will require that we have this information for all devices in DT though.

@bjarki-andreasen the other problem is when the target is not using device runtime pm, I agree that this is the recommended method but there are a lot of targets using system managed device pm, it needs less resource and it is simpler.  There are use cases where Zephyr's application is not in the control. Open sound firmware is one of those, the host OS can arbitrarily forces Zephyr to sleep to any power state.

I think what you have described is what we want, we can agree about it when device runtime is enabled. The rest of proposals here is for when the target is using system managed device power management. For a device the change will be handle these two different events or ignore them if device runtime pm is the only method it is supposed to work.

hubertmis, 2024-09-13:
The PM_DEVICE_RUNTIME framework should be supplemented with requirements on implicit INACTIVE->ACTIVE->INACTIVE state changes.

We should not require a device driver API user to call `pm_device_runtime_get()` / `_put()` functions even if PM_DEVICE_RUNTIME is enabled.

The API user should be able to perform the following single-step sequence:

1. Get a random number from an entropy driver

The entropy device driver should handle this sequence by:

1. A random number request
    1. Resuming the device
    2. Generating the number
    3. Suspending the device

The above sequence should be performed regardless `PM_DEVICE_RUNTIME` is enabled or not.

The only reason for enabling `PM_DEVICE_RUNTIME` is when the API user would like to perform a few operations in a row:

1. Get one random number
2. Get another random number

The entropy driver would handle this sequence of requests like this:

1. First request
    1. Resume
    2. Generate the number
    3. Suspend
2. Second request
    1. Resume
    2. Generate the number
    3. Suspend

We can clearly spot that 1.3 and 2.1 steps are a waste. `PM_DEVICE_RUNTIME` allows us to remove this waste. The application should be modified to:

1. `pm_device_runtime_get()`
2. Get one random number
3. Get another random number
4. `pm_device_runtime_put()`

And then the entropy driver:

1. `ACTION_RESUME`
    1. Resume
2. Get the first number
    1. Generate the number
3. Get the second number
    1. Get the number
4. `ACTION_SUSPEND`
    5. Suspend

However, without using `PM_DEVICE_RUNTIME`, the application would still be fully functional. Just its quality (performance, power efficiency) would be degraded

hubertmis, 2024-09-13:
> Currently, if pm device runtime is not enabled, everything is resumed at boot, and remains resumed.

That's not entirely true. Nordic device drivers work according to the principles I described: they implicitly resume and suspend on each request. We should not force users to enable `PM_DEVICE_RUNTIME` with its overhead to keep this functionality.

> I think this is correct behavior: "I don't need power management" -> power management is not enabled, explicitly nor implicitly.

So maybe we should create a new Kconfig option or connect implicit power management in the device drivers with `CONFIG_PM`. In the router case you could state "I don't need power management" - `CONFIG_PM=n` what would disable both CPU and implicit device drivers' power management.

> Only the application knows which devices and how said devices will be used at any given time, therefore the application should inform some pm subsystem like pm device runtime when a device is needed and not.

But not only the application is a user of device drivers API. Other device drivers and middleware would also require to use pm device runtime implemented. E.g. Application stores a file, so the filesystem uses a memory driver which uses an SPI driver which uses a GPIO driver. If we required explicit calls, all elements in the chain would need to implement pm device runtime calls.

If we make these calls implicit, we don't need to update all middleware and higher-level drivers at once.

[30 more replies]