zephyrproject-rtos/zephyr #76000: coap: block_number in coap_get_block1_option() is too small

yashi, 2024-07-17:
**Describe the bug**

[RFC 7959 states](https://datatracker.ietf.org/doc/html/rfc7959#section-2.1) that the Block Number can be more than 8 bits.

```
           0
           0 1 2 3 4 5 6 7
          +-+-+-+-+-+-+-+-+
          |  NUM  |M| SZX |
          +-+-+-+-+-+-+-+-+

           0                   1
           0 1 2 3 4 5 6 7 8 9 0 1 2 3 4 5
          +-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
          |          NUM          |M| SZX |
          +-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+

           0                   1                   2
           0 1 2 3 4 5 6 7 8 9 0 1 2 3 4 5 6 7 8 9 0 1 2 3
          +-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
          |                   NUM                 |M| SZX |
          +-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+

                       Figure 1: Block Option Value
 ```

However, `block_number` in [`coap_get_block1_option()`](https://docs.zephyrproject.org/latest/doxygen/html/group__coap.html#ga796c9fcf9f447ccf7def2da091e7e5f5) and [`coap_get_block2_option()`](https://docs.zephyrproject.org/latest/doxygen/html/group__coap.html#ga161875609f4a1ce5806625f97cd90a39) is declared as `uint8_t`.

**To Reproduce**
Try `samples/net/sockets/coap_server/` with `coap_get_option_int()` replaced with `coap_get_block1_option()`

**Expected behavior**

`block_number` to be `unsigned int`.

**Impact**
Block number is truncated when using the APIs.

**Environment (please complete the following information):**

 - OS: Linux Debian Sid
 - Toolchain: Zephyr SDK v0.16.6
 - Commit SHA: v3.7.0-rc3-25-gcb17f5f5a4af

boaks, 2024-07-23:
Fixing that by changing the pointer type will be an API break.
Maybe it's easier to document the limit and  reference to use the `coap_block_context` variants for larger transfers?

rlubos, 2024-07-24:
> Fixing that by changing the pointer type will be an API break. Maybe it's easier to document the limit and reference to use the `coap_block_context` variants for larger transfers?

The CoAP API is still considered "unstable", and we'll announce this rather trivial change in the migration guide. I'd personally prefer just to fix this instead having to document limitations of not-well-thought API functions.

pdgendt, 2024-07-24:
I agree, especially since the `coap_append_block[1|2]_option` functions allow to set these values.

boaks, 2024-07-24:
Just in the case, the API gets changed, adding the "more" flag is also helpful.

rlubos, 2024-07-24:
> Just in the case, the API gets changed, adding the "more" flag is also helpful.

That's a good point, it seems that `coap_get_block2_option()` has purposely limited the functionality to requests only:

` Ignore the has_more flag as it should always be zero on queries.`

@SeppoTakalo Since you've introduced this particular function, any objections to introduce a pointer to `has_more` flag, to make the function more generic?

SeppoTakalo, 2024-07-24:
No objections.
I think the API has been formed a bit too much on what LwM2M client needed.

As it is still unstable, I would prefer someone to refactor it to be generic CoAP API that would be usable on both, client and server, side.

boaks, 2024-07-24:
To fix the API is the on side.
The other side for me will still be to refer to the usage of the `coap_block_context`.

rlubos, 2024-07-24:
I've push a commit adding the more flag handling `coap_get_block2_option()`.

> The other side for me will still be to refer to the usage of the coap_block_context.

I not sure I follow. Where would you like to reference the coap_block_context use?

boaks, 2024-07-24:
> I'm not sure I follow. Where would you like to reference the coap_block_context use?

Using the `coap_block_context` variants, `coap_block_transfer_init`, `coap_update_from_block` and `coap_next_block` obsoletes for many case the use of that "gets". Not sure where `coap_block_context` didn't work.  

Therefore I would add a comment like:

"Please consider to use the coap_block_context API with coap_block_transfer_init, coap_update_from_block, and   coap_next_block."

boaks, 2024-07-24:
> I'm not really convinced if that's needed, to me those are two different APIs (I. e. parser vs transfer progress tracking).

For now I don't see any good reason to use the "coap_get_block1_option" or "coap_get_block2_option".
Maybe those, who introduced it, show why the variant with `coap_block_context` isn't working for them?
Therefore I would rather go to remove them in the future at all. 
Such an "get" API may mislead more developers to implement something, which is already implemented there.
(Introducing risks of non-compliant implementations, when too less time gets spend in the core-mailing-list and corrclar.)

SeppoTakalo, 2024-07-25:
> For now I don't see any good reason to use the "coap_get_block1_option" or "coap_get_block2_option".
> Maybe those, who introduced it, show why the variant with `coap_block_context` isn't working for them?

For the LwM2M client, which is a CoAP server, at least sending a Block-Wise content this is used
https://github.com/zephyrproject-rtos/zephyr/blob/d1781da2ba420085bbf511235012ce8b9b00dc59/subsys/net/lib/lwm2m/lwm2m_message_handling.c#L2656

`coap_block_context` the endpoint might lose a message and request same block again, therefore we need to recalculate the block context for every request.

boaks, 2024-07-25:
I've only implemented a download client, so I guess, there are a couple of details, I'm not aware of.
My understanding of the `coap_block_context` functions for a server would be:l

`coap_block_transfer_init` 
`coap_update_from_block` (with the received request) => `coap_block_context->current` points to the request part. Fill the response and `coap_append_block2_option`. 

not sure, what I oversee, that this should fail on rereceiving an request.

To be frank, my impression when I implemented the download is more, that the documentation my consider also some sequences of the usage of this function including some corner-cases as rereceiving an request.

PR #76052 (2024-08-03): net: coap: Fix underlying type for block number
The block number in block1/2 options can be encoded on up to 20 bits according to RFC 7959, therefore the underlying type used in helper functions to retrieve the block number should be large enough to hold the result. Therefore, replace the container for block number with uint32_t instead of uint8_t.

Fixes #76000

TODO:
- [x] Add a note in the migration guide, when available.
PR #76690 (2024-08-06): [Backport v3.7-branch] net: coap: Fix underlying type for block number 
Backport of #76052

Fixes #76000
closed 2024-08-03 by dleach02