zephyrproject-rtos/zephyr #76037: pm: enabling `PM_DEVICE_RUNTIME` leads to deprecated symbols

JordanYates, 2024-07-18:
**Describe the bug**

Enabling `PM_DEVICE_RUNTIME` in an application leads to deprecated symbols being enabled by default.

**To Reproduce**

```
west build -b nrf52840dk/nrf52840 samples/hello_world/ -- -DCONFIG_PM_DEVICE=y -DCONFIG_PM_DEVICE_RUNTIME=y
...
warning: Deprecated symbol PM_DEVICE_RUNTIME_EXCLUSIVE is enabled.
...
```

**Expected behavior**

Enabling the standard device PM option should not result in deprecated symbols being enabled.

**Impact**

The warning could give the incorrect impression that `PM_DEVICE_RUNTIME` is deprecated, especially given the similarities in the naming (`PM_DEVICE_RUNTIME` vs `PM_DEVICE_RUNTIME_EXCLUSIVE`).

**Environment (please complete the following information):**

 - Zephyr v3.7.0-rc3

ceolin, 2024-07-18:
Thanks a lot for catching it !

nashif, 2024-07-19:
marking as a blocker due to enabling deprecated features by default.

PR #76040 (2024-07-22): pm: Do not select deprecated symbol
PM_DEVICE_RUNTIME_EXCLUSIVE was deprecated and its behavior is achived with PM_DEVICE_SYSTEM_MANAGED=n.

Fixes #76037
commit f9de58ef (2024-07-22): pm: Do not select deprecated symbol

PM_DEVICE_RUNTIME_EXCLUSIVE was deprecated and its behavior
is achived with PM_DEVICE_SYSTEM_MANAGED=n.

Fixes #76037
closed 2024-07-22 by nashif