zephyrproject-rtos/zephyr #76072: Compiling usbd.h with C++ compiler fails

MSeesko, 2024-07-18:
Hi,

Compiling a Zephyr project with a C++ compiler (CONFIG_CPP=y) leads to a build error, when `zephyr/include/zephyr/usb/usbd.h` gets included.

```
/opt/zephyrproject/zephyr/include/zephyr/usb/usbd.h:507:9: error: designator order for field 'usbd_desc_node::<anonymous>' does not match declaration order in 'usbd_desc_node'
  507 |         }
```

```
/opt/zephyrproject/zephyr/include/zephyr/usb/usbd.h:530:9: error: designator order for field 'usbd_desc_node::ptr' does not match declaration order in 'usbd_desc_node'
  530 |         }
```

Putting the members of the struct in the correct order in those macros fixes the problem.

jadejr, 2024-07-18:
I had the same problem with zbus when `CONFIG_RUNTIME_OBSERVERS=y` (or whatever it's called) is set. Are there any means to tell the C compiler to act like C++ in this regard?

marc-hb, 2024-07-19:
Take a look at https://github.com/zephyrproject-rtos/zephyr/commit/d3a1ce879186eebdb604bc48c23987dce9b1d2d8 and #70403

aescolar, 2024-07-19:
@MSeesko  can you please provide information about which Zephyr version/sha you are using?

MSeesko, 2024-07-19:
I am using Zephyr 3.7.0-rc3

PR #76123 (2024-07-27): include: usb: fix designator order in usbd.h header
Fix "designator order for field 'usbd_desc_node::ptr' does not match declaration order in 'usbd_desc_node'" error when building applications with CONFIG_CPP enabled.

Fixes: #76072
closed 2024-07-27 by nashif