zephyrproject-rtos/zephyr #76090: TF-M: builds download source from external git repository

JordanYates, 2024-07-19:
**Describe the bug**

Since the update to [TFM v2.1.0](https://github.com/zephyrproject-rtos/trusted-firmware-m/pull/105), any builds of TF-M are cloning down sources from this repo: https://github.com/ARM-software/CMSIS_6

Source commit: https://github.com/zephyrproject-rtos/trusted-firmware-m/commit/3d54aca3851590bf5ecf5f52f361b390c925e257

**To Reproduce**

`west build -b nrf9161dk/nrf9161/ns zephyr/samples/hello_world/ -p`
Observe:
```
...
-- Build files have been written to: /home/jordan/code/zephyr/build/tfm/lib/ext/cmsis-subbuild
[1/9] Creating directories for 'cmsis-populate'
[1/9] Performing download step (git clone) for 'cmsis-populate'
Cloning into 'cmsis-src'...
remote: Enumerating objects: 28525, done.
remote: Counting objects: 100% (3662/3662), done.
remote: Compressing objects: 100% (977/977), done.
remote: Total 28525 (delta 3071), reused 3002 (delta 2624), pack-reused 24863
Receiving objects: 100% (28525/28525), 39.74 MiB | 5.76 MiB/s, done.
Resolving deltas: 100% (21001/21001), done.
HEAD is now at d0c460c1 CMSIS-Core(M): Add Cortex-M52 support : add core_m52.h
[2/9] Performing update step for 'cmsis-populate'
...
```

**Expected behavior**

Builds should not be pulling down sources from the internet.
Builds should complete regardless of internet connectivity state.

**Impact**

"Untrusted" (Not zephyrproject) source being pulled down from the internet on each build.
Unable to build TF-M applications without a build system.
Feels like it violates some sort of Zephyr guideline, or should.

**Environment (please complete the following information):**

```
(.zephyr_venv) jordan@TAURUS:~/code/zephyr/zephyr$ git describe
v3.7.0-rc3-73-gac52bd629d7
```

tomi-font, 2024-07-19:
https://github.com/zephyrproject-rtos/zephyr/pull/76094 is a quick fix.
The proper way would be to fork the CMSIS v6 repository and add it as a new West project to Zephyr.

fabiobaltieri, 2024-07-19:
cc @nashif @aescolar

henrikbrixandersen, 2024-07-23:
Adding the TSC label as @d3zd3z requested this to be mentioned to the TSC.

marc-hb, 2024-07-24:
This was flagged in #70215 and other places before.

tomi-font, 2024-07-25:
> This was flagged in #70215 and other places before.

I don't think it's the same issue there. The issue here was introduced by TF-M 2.1.0, which came way after the creation of the issue you are referencing.

PR #76094 (2024-07-25): modules: tf-m: fix git clone during build
As of v2.1.0 TF-M has updated to CMSIS v6 and switched from hosting the sources to depending on the upstream repository, cloning it at build time.

To prevent a download from happening during the build, CMSIS v6 sources are pushed to Zephyr's fork of TF-M and the `CMSIS_PATH` CMake variable is set to point to them.

Fixes #76090.
PR #76359 (2024-08-06): [Backport v3.7-branch] samples: tfm_integration: psa_crypto: Disable sample (prevent fetching online content)
Backport db5a5fa109854981c137e8a1a6f3d210be6fd515 from #76093.

Fixes #76090
closed 2024-07-25 by aescolar