zephyrproject-rtos/zephyr #76128: Networking: Dynamic MTU

JordanYates, 2024-07-20:
**Is your enhancement proposal related to a problem? Please describe.**

The Zephyr network interface APIs' expose an MTU (maximum transmission unit), but this is a compile-time **hardware** MTU, not a runtime **network** MTU.

https://github.com/zephyrproject-rtos/zephyr/blob/d06e95c49b75dfc911f2ab77c2a934abe624c751/include/zephyr/net/net_if.h#L657-L658

https://github.com/zephyrproject-rtos/zephyr/blob/d06e95c49b75dfc911f2ab77c2a934abe624c751/include/zephyr/net/net_if.h#L1234-L1243

The compile-time value in insufficient to avoid fragmentation because while the hardware may support an MTU of X, the network the hardware is currently connected to may only support a MTU of Y. The MTU of LTE networks in particular is highly variable between different providers, with devices jumping between networks as they travel.

**Describe the solution you'd like**

A common pattern where upon the transition into [NET_IF_OPER_UP](https://docs.zephyrproject.org/latest/connectivity/networking/api/net_if.html#c.net_if_oper_state.NET_IF_OPER_UP), the MTU of the current network is provided to the stack and is queryable by end users in the appropriate networking event callback (e.g. `NET_EVENT_L4_CONNECTED`, `NET_EVENT_WIFI_CONNECT_RESULT`)

rlubos, 2024-07-22:
It looks that the MTU stored in `iface->if_dev->mtu` is already possible to override with `net_if_set_mtu()`, isn't that enough? Or do you propose that we store a separate value for the network MTU at the net_if level?

JordanYates, 2025-01-27:
> It looks that the MTU stored in `iface->if_dev->mtu` is already possible to override with `net_if_set_mtu()`, isn't that enough? Or do you propose that we store a separate value for the network MTU at the net_if level?

No that is fine, I missed its existence since barely any drivers use it, but that's not the fault of the API.

closed 2025-01-27 by JordanYates