zephyrproject-rtos/zephyr #76178: IEEE802154 active scan gets filtered in network stack

gumulka, 2024-07-22:
**Describe the bug**

When issuing an active scan of 802.15.4 on the shell, all response packages are dropped by the network stack.

**To Reproduce**

Have something, that answers to 802.15.4 active network scans, like a zigbee coordinator (here, you could use the out of scope nRF sample nrf/samples/zigbee/network_coordinator)

Compile Zephyr with:

```
CONFIG_SHELL=y
CONFIG_NETWORKING=y
CONFIG_NET_SHELL=y
CONFIG_NET_L2_IEEE802154=y
CONFIG_NET_L2_IEEE802154_SHELL=y
CONFIG_NET_L2_IEEE802154_RFD=y
```

Do an active scan on the commandline (see snippet below)

**Expected behavior**

The scan response packages are received and given to the network stack for further processing

**Impact**

Showstopper. I have to force the network interface up before I can scan.

**Logs and console output**

The log from my shell:
```
uart:~$ net iface show 

Interface ieee0 (0x20000b30) (IEEE 802.15.4) [1]
========================================
Interface is down.
uart:~$ net iface up 1
Cannot take interface 1 up (-115)
uart:~$ ieee802154 scan active all 100
Active Scanning (channel set: 0xffffffff, duration 100 ms)...
Done
[00:02:23.907,348] <wrn> net_ieee802154_mgmt: Could not remove PAN ID filter
[00:02:25.347,076] <err> ieee802154_nrf5: Packet dropped by NET stack
[00:02:25.348,358] <err> ieee802154_nrf5: Packet dropped by NET stack
[00:02:25.360,168] <err> ieee802154_nrf5: Packet dropped by NET stack
[00:02:25.546,966] <wrn> net_ieee802154_mgmt: Could not remove PAN ID filter
uart:~$ 
```

The "Packet dropped by NET stack" messages are produced, because the network interface is not up. I don't want to set the interface up, because I'm currently scanning for the network.

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr current main
 - f2b6490dee30b38dfb0ee31902177491c35ebacb
 - Board: NRF52840DK

**Additional context**

The error starts here:
https://github.com/zephyrproject-rtos/zephyr/blob/f2b6490dee30b38dfb0ee31902177491c35ebacb/drivers/ieee802154/ieee802154_nrf5.c#L215-L218

Because:
https://github.com/zephyrproject-rtos/zephyr/blob/f2b6490dee30b38dfb0ee31902177491c35ebacb/subsys/net/ip/net_core.c#L499-L501

aescolar, 2024-07-23:
CC @rlubos @jukkar

fgrandel, 2024-07-23:
@gumulka By the current standard (section 6.3.1), only beacons should be received during active or passive scan. Are you saying that currently no beacons can be received?

If we're not talking about beacons then please help me with some additional context re your expectation. What kinds of packets does the Zigbee co-ordinator send back (frame type, payload, purpose)?

fgrandel, 2024-07-23:
@gumulka I further debugged the issue and I cannot reproduce it. We have a unit test on this. Maybe you have the KConfig variable IEEE802154_NET_IF_NO_AUTO_START set to true? In that case you'd have to call net_if_up() manually. Otherwise the interface should be brought up during initialization which in turn would set the NET_IF_UP flag and any beacons received should then be allowed in.

fgrandel, 2024-07-23:
@gumulka Can you please try to scan from the echo sample instead, e.g.

```shell
west build samples/net/sockets/echo_server --pristine --board nrf52840dk/nrf52840 -DEXTRA_CONF_FILE=overlay-802154.conf
```
The shell is exposed through UART in that sample. In my case this provides me with a properly set up interface that is automatically "UP". Can you properly scan with this sample? (This doesn't mean that we don't have a bug, it just means that we can further hone in on the source of the bug.)

gumulka, 2024-07-23:
>  Are you saying that currently no beacons can be received?

yes, as seen in the logs.

> @gumulka I further debugged the issue and I cannot yet reproduce it. We have a unit test on this. The interface should be brought up during initialization which in turn would set the NET_IF_UP flag and any beacons received should then be allowed in.
> 
> But I can see that you cannot bring the interface up manually in your case. I'll dig deeper because that seems to be the actual cause.

I have just seen, that I forgot the `CONFIG_LOG=y` parameter

How to reproduce on my side:

```shell
mkdir zephyr_76178 
cd zephyr_76178
west init -m https://github.com/zephyrproject-rtos/zephyr --mr main
west update
cd zephyr
cat 76178.patch | git am
cd ..
west build -p -b nrf52840dk_nrf52840 zephyr/samples/basic/blinky
west flash
```

Using the nrf52840dk, obviously.

for the patch, use this git diff [76178.txt](https://github.com/user-attachments/files/16353115/76178.txt)

results in:

```
[00:00:00.448,913] <inf> ieee802154_nrf5: nRF5 802154 radio initialized
*** Booting Zephyr OS build v3.7.0-rc3-102-g099209b7d563 ***
uart:~$ ieee802154 scan active all 500
Active Scanning (channel set: 0xffffffff, duration 500 ms)...
Done
[00:00:08.975,860] <err> ieee802154_nrf5: Packet dropped by NET stack
[00:00:13.372,528] <err> ieee802154_nrf5: Packet dropped by NET stack
[00:00:13.373,809] <err> ieee802154_nrf5: Packet dropped by NET stack
[00:00:13.377,990] <err> ieee802154_nrf5: Packet dropped by NET stack
uart:~$ 
```

gumulka, 2024-07-23:
> @gumulka Can you please try to scan from the echo sample instead, e.g.
>
> ```shell
> west build samples/net/sockets/echo_server --pristine --board nrf52840dk/nrf52840 [...]

Yes, I can do with this example. But here the example does the heavy lifting of putting up the network. I don't need a network that is up to scan for beacons.

fgrandel, 2024-07-23:
> I don't need a network that is up to scan for beacons.

`net_if_up()` brings up the link layer (MAC) - not a full network. Scanning is a link layer activity, so I'd say you need L2 up to scan. You don't need anything above L2 of course (IP, UDP, etc.).

When enabling the interface we currently switch on the receiver so you have to configure at least a channel before you can call `net_if_up()` (which should be optional - see below). Conceptually enabling the interface could be anything, though, therefore a disabled interface should not allow you to scan. So what you're seeing is expected behavior IMHO.

I do see, though, where the confusion might come from:
* If you wanted the interface to be enabled w/o configuring a channel and switching on the receiver, then you'd have to unset the `macRxOnWhenIdle` flag (see IEEE 802.15.4-2020) via net_mgmt API before bringing up the interface. Unfortunately `macRxOnWhenIdle` was implemented w/o `net_mgmt` support (see the thread I created today in #ieee802154) and cannot be read by L2 right now (which is a bug IMO) - CC @rlubos @edmont
* There is an unconditional call to `ieee802154_radio_start()` in `ieee802154_scan()` which dates back from 8 years ago. This should only be called if the receiver is not already on.

Both issues are unrelated to your issue, though, and should be easy to work around by setting a channel before trying to bring up the interface.

fgrandel, 2024-07-24:
See https://github.com/zephyrproject-rtos/zephyr/pull/76265.

closed 2024-10-07 by github-actions[bot]