zephyrproject-rtos/zephyr #76185: scripts: 'west blobs fetch' does not verify digests after downloading

dottspina, 2024-07-22:
Binary blobs and their companion `west blobs` command are really handy.

But running `west blobs fetch` will not verify the digests of downloaded files, which does not match the documented [1] behavior:
 
> A checksum for the blob, which `west blobs` checks after downloading. This is required for reproducibility and to allow bisecting issues as blobs change using Git and west

## Problem description

The current behavior is:

1. if the checksum of the previously downloaded file equals to that in the blob metadata (status `BLOB_PRESENT`), do nothing
2. if the checksum of the previously downloaded file differs from that in the blob metadata (status `BLOB_OUTDATED`), assume versions differ and download the *up to date* file
3. if the blob has not yet been downloaded (status `BLOB_NOT_PRESENT`), download it

None of the 2) and 3) code paths will verify that the checksum of the file just downloaded actually matches the digest in the blob's metadata.

Consequences:

- in the event that the metadata of a module is incorrect, then the user will not notice anything, and may rely on an unexpected binary, e.g. a static library for a different architecture
- this makes these SHA digests less reliable for "reproducibility and bisecting issues as blobs change using Git and west"
- this skips a (basic) mitigation against supply chain attacks

## How to reproduce

1. Change the SHA256 digest of some blob, e.g. blob `lib/esp32c3/libbtdm_app.a` in `modules/hal/espressif/zephyr/module.yml`
2. Run `west blobs fetch hal_espressif`: this will download the file, without warning the user that its checksum does not match that in the blob metadata
3. Users will only notice if they find it suspicious that the file is downloaded again and again  

```
$ west blobs fetch hal_espressif
Fetching blob hal_espressif: path/to/zephyr/modules/hal/espressif/zephyr/blobs/lib/esp32c3/libbtdm_app.a

$ west blobs fetch hal_espressif
Fetching blob hal_espressif: path/to/zephyr/modules/hal/espressif/zephyr/blobs/lib/esp32c3/libbtdm_app.a
```

**Expected behavior**

Something like:

```
$ west blobs fetch hal_espressif
Fetching blob hal_espressif: path/to/zephyr/modules/hal/espressif/zephyr/blobs/lib/esp32c3/libbtdm_app.a
ERROR: The checksum of the downloaded file does not match that in the blob metadata:
- if it is not certain that the download was successful,
  try running 'west blobs fetch hal_espressif'
  to re-download the file
- if the error persists, please consider contacting
  the maintainers of the module so that they can check
  the corresponding blob metadata

Module: hal_espressif
Blob:   lib/esp32c3/libbtdm_app.a
URL:    https://github.com/espressif/esp32c3-bt-lib/raw/b438f60a295183e7c67eb42ae05f4580f4b1ced0/esp32c3/libbtdm_app.a
Info:   Binary libraries supporting the ESP32 series RF subsystems
```

Thanks.

[1] Fetching blobs: https://docs.zephyrproject.org/3.6.0/contribute/bin_blobs.html#fetching-blobs

aescolar, 2024-07-23:
CC @mbolivar-ampere @carlescufi @swinslow
Assigning @carlescufi as the most active recently in this extension cmd.

aescolar, 2024-07-23:
assigning @dottspina as he submitted a fix

PR #76186 (2024-07-30): west: blobs: verify fetched blobs after downloading
**Problem description**

Running `west blobs fetch` does not verify the digest of downloaded files.

Fixes #76185

**Proposed change**

Call `zephyr_module.get_blob_status()` (again) after downloading, and warn the user if the blob status is `BLOB_OUTDATED` (digests don't match).

Note: The proposed error message is intended to not be scary, and assumes incorrect metadata (URL or digest) rather than a *compromised* download:

```
$ west blobs fetch hal_espressif
Fetching blob hal_espressif: path/to/zephyr/modules/hal/espressif/zephyr/blobs/li [... cut]
closed 2024-07-30 by fabiobaltieri