zephyrproject-rtos/zephyr #76204: VL53L0x - Check RangeStatus for measurement result

marcoster, 2024-07-23:
**Is your feature request related to a problem? Please describe.**
In a previous project we needed to check the *RangeStatus* of the sensor to determine if the result is valid (especially status values of 2 and 4 occured in some cases).
The current driver implementation doesn't evaluate the *RangeStatus* field of the *RangingMeasurment* nor is it exposed through the Sensor-Interface to be evaluated in the app using the sensor.

I'd be happy if anyone points me to the correct way to do this!

**Describe the solution you'd like**
What is the way to go here?
1) implement a sensor-channel for the VL53L0x to derive the RangeStatus in the application using it
2) Check the RangeStatus in either vl53l0x_sample_fetch or vl53l0x_channel_get and return a matching result-code
3) An additional direct API to vl53l0x to retrieve the value (uint8_t vl53l0x_get_range_status(struct device *dev))

**Describe alternatives you've considered**
Some hacky way in our app to retrieve the last RangeStatus from the *struct device* that I'm using to fetch the measurement That's at least what I implemented for now in my application:
```
#include "vl53l0x_api.h"

// re-definition from vl53l0x.c, since that's not available outside of there
// keep in sync if zephyr changes something here...
struct vl53l0x_data {
    bool                             started;
    VL53L0X_Dev_t                    vl53l0x;
    VL53L0X_RangingMeasurementData_t RangingMeasurementData;
};

int distance_measure(double *distance, uint8_t *range_status)
{
    int                 ret;
    struct sensor_value value;

    ret = sensor_sample_fetch(m_distance_sensor);
    if (ret) {
        LOG_WRN("sensor_sample_fetch failed with %d", ret);
        return ret;
    }

    ret = sensor_channel_get(m_distance_sensor, SENSOR_CHAN_DISTANCE, &value);
    if (ret) {
        LOG_WRN("sensor_channel_get failed with %d", ret);
        return ret;
    }

    struct vl53l0x_data *drv_data = m_distance_sensor->data;

    *distance     = sensor_value_to_double(&value);
    *range_status = drv_data->RangingMeasurementData.RangeStatus;
    return 0;
}

```

**Additional context**
-

michalpiekos, 2024-12-02:
Looks like your suggestion to implement custom channel is the best solution here. I can do it if you are still interested.

marcoster, 2024-12-05:
Yeah sure, that would be great to cleanup my workaround in the future!
Looking forward to it!

PR #83401 (2025-01-17): drivers: sensor: vl53l0x: add private channels with measurement metadata
This PR adds new channels to return measurement metadata for the VL53L0X sensor. 
It includes:
- Driver changes as requested by #76204 adding range status, SPAD's used, max range, signal rate and ambient rate.
- An updated sample demonstrating usage of new channels.

Closes #76204 

**Testing:**
- Build and flash the `samples/sensor/vl53l0x` application on stm32_min_dev
- Verify output matches the expected behavior described in the documentation.
closed 2025-01-17 by kartben