zephyrproject-rtos/zephyr #76225: drivers: usb: udc: udc_it82xx2 does not report UDC_EVT_SOF

henrikbrixandersen, 2024-07-23:
**Describe the bug**
The USB device_next driver for the ITE IT82XX2 does not report start of frame (`UDC_EVT_SOF`) events.

**Expected behavior**
All USB device controller drivers should report Start of Frame (SoF) events.

**Impact**
Unable to create a hardware-agnostic application making use of SoF events (described further in https://github.com/zephyrproject-rtos/zephyr/pull/76073).

**Environment (please complete the following information):**
 - OS: Any
 - Toolchain: Any
 - Commit SHA: v3.7.0-rc3

jfischer-no, 2024-07-24:
@Chenhongren, can you please take a look, we would need it for v4.0.0.

Chenhongren, 2024-08-15:
> @Chenhongren, can you please take a look, we would need it for v4.0.0.

We are working on this ([link](https://chromium-review.googlesource.com/c/chromiumos/third_party/zephyr/+/5767074)). However, there are other urgent issues we need to address at the moment. We will come back and create a PR ASAP.

commit ea4f1b75 (2024-10-22): udc: it82xx2: emit SOF event

This commit emits SOF event.

Fixes: #76225
closed 2024-10-22 by aescolar