zephyrproject-rtos/zephyr #76228: drivers: usb: udc: udc_mcux_ip3511 does not report UDC_EVT_SOF

henrikbrixandersen, 2024-07-23:
**Describe the bug**
The USB device_next driver for the NXP IP3511 IP does not report start of frame (`UDC_EVT_SOF`) events.

**Expected behavior**
All USB device controller drivers should report Start of Frame (SoF) events.

**Impact**
Unable to create a hardware-agnostic application making use of SoF events (described further in https://github.com/zephyrproject-rtos/zephyr/pull/76073).

**Environment (please complete the following information):**
 - OS: Any
 - Toolchain: Any
 - Commit SHA: v3.7.0-rc3

jfischer-no, 2024-07-24:
@MarkWangChinese can you please take a look, we would need it for v4.0.0. Would it require a HAL driver update?

MarkWangChinese, 2024-07-30:
> @MarkWangChinese can you please take a look, we would need it for v4.0.0. Would it require a HAL driver update?

Hi @jfischer-no , OK, I will handle it.

PR #76681 (2024-08-15): Feature/enable udc sof
Fixes #76228 
Fixes #76227
closed 2024-08-15 by fabiobaltieri