zephyrproject-rtos/zephyr #76237: STM32H743 data in SRAM3 or DTCM not always fully written

biglben, 2024-07-23:
**Describe the bug**
I played around with the retention system and had some issues (most probably not related to the retention system as it uses just memcpy).
If i write to the retained memory and then reboot, depending on the prefix/crc size and the selected ram the behavior is quite strange. It looks to me like some data is not written to the ram.  
With sram0 everything seems to work fine.  
With sram3 if the prefix and checksum is 4 bytes its also fine. But if each is just 2 bytes the crc is not written.  
With dtcm the last byte of the crc and last byte of the prefix is not written.  
I say not written because when i fill the memory with some data before calling `retention_write`, the data are read after the reboot. Read after `retention_write` but before reboot outputs the correct data. If `retention_write` is called twice it works correctly with dtcm and sram3.  
It looks to me like some cache problem but `SCB_CleanDCache` didn't help and the dtcm does not have a cache (as far as i understand).

**To Reproduce**
Unfortunately i didn't found an sample to reproduce it. I quickly hacked this together to have a simple way to reproduce it: https://github.com/biglben/zephyr/tree/debug-retention-h7  
It just tries to read a uint32_t from the retained memory, clears it and writes a new value. In addition it prints out the raw values directly from the ram.  
It can be built with `west build -p -b nucleo_h743zi/stm32h743xx zephyr/samples/subsys/retention/`  
The dts overlay uses the dtcm. Examples for sram3 and sram0 are commented out.

**Expected behavior**
Writting to retained memory, reboot and read from it returns the correct data

**Impact**
Not big. Planned to use dtcm but changed to sram3 (sram0 used for other purpose).

**Logs and console output**
```
--------------------------------------------------------------------
Hello World!
data invalid
data at mem:
be 00 67 45 23 01 00 00 00 00 00 00 00 00 91 00 
Try to write value 1234567

written
data at mem:
be ef 67 45 23 01 00 00 00 00 00 00 00 00 91 15 
*** Booting Zephyr OS build v3.7.0-rc3-103-g7ee76bc3667c ***

--------------------------------------------------------------------
Hello World!
data invalid
data at mem:
be 00 67 45 23 01 00 00 00 00 00 00 00 00 91 00 
Try to write value 1234567

written
data at mem:
be ef 67 45 23 01 00 00 00 00 00 00 00 00 91 15 

```

**Environment (please complete the following information):**

- OS: Linux
- Toolchain: Zephyr SDK (docker.io/zephyrprojectrtos/zephyr-build:v0.26.13)
- Commit: 099209b7d563886b23ddcbb1d59f06fdd9e66ab1

FRASTM, 2024-07-26:
The checksum written by retained_mem_write  is not hold by the sram. After writing the checksum 0x1591, ram value is read by (gdb)
0x30047ff0:	0xbe	0xef	0x67	0x45	0x23	0x01	0x00	0x00
0x30047ff8:	0x00	0x00	0x00	0x00	0x00	0x00	0x91	0x15
but bytes 0x30047ffe & 0x30047fff is 0x0000 after reset

biglben, 2024-07-29:
Thank you very much, that makes sense. I will change and use 4 bytes for prefix and crc.

closed 2024-07-29 by biglben

FRASTM, 2024-07-29:
The other way, if CRC is 16 bit data to be written at offset_crc16, is to read 2 bytes at offset_crc16 - sizeof(crc16) and to write that 2 bytes value at the same place.
It works.