zephyrproject-rtos/zephyr #76250: example usb/cdc_acm coredump on board STMicroelectronics nucleo_h563zi when use "usbd_next_prj.conf"

MY201314MY, 2024-07-24:
**Describe the bug**
when I enable the **CONFIG_USB_DEVICE_STACK=y**, the usb/cdc_acm can work well.
But when I  enable **CONFIG_USB_DEVICE_STACK_NEXT=y**, the system will coredump. 

**To Reproduce**
Steps to reproduce the behavior:
1. check zephyr version
```
commit 099209b7d563886b23ddcbb1d59f06fdd9e66ab1 (HEAD -> mini-h562-usb-1.1.0, origin/main, origin/HEAD, main)
Author: Johan Hedberg <johan.hedberg@silabs.com>
Date:   Mon Jul 22 12:43:46 2024 +0300

    doc: release-notes-3.7: Add more details for Bluetooth HCI drivers
    
    Make sure all added (and removed) HCI drivers are mentioned, and also
    provide a reference to the new HCI driver API.
    
    Signed-off-by: Johan Hedberg <johan.hedberg@silabs.com>
```

2.build
```
west build samples/subsys/usb/cdc_acm -b nucleo_h563zi -p -- -DCONF_FILE:STRING="usbd_next_prj.conf"
```
3.**_west flash_** and inset the usb cable to PC
4. See error
the third parameter is NULL, and caused the system to crash.
[image]

**Expected behavior**

**Impact**

**Logs and console output**

```
*** Booting Zephyr OS build v3.7.0-rc3-102-g099209b7d563 ***
[00:00:00.000,000] <inf> cdc_acm_echo: USB device support enabled
[00:00:00.000,000] <inf> cdc_acm_echo: Wait for DTR
[00:00:00.003,000] <inf> cdc_acm_echo: USBD message: Device suspended
[00:00:04.955,000] <inf> cdc_acm_echo: USBD message: Device resumed
[00:00:04.958,000] <inf> cdc_acm_echo: USBD message: Device suspended
[00:00:05.099,000] <inf> cdc_acm_echo: USBD message: Device resumed
[00:00:05.156,000] <inf> cdc_acm_echo: USBD message: Bus reset
[00:00:05.284,000] <inf> cdc_acm_echo: USBD message: Bus reset
[00:00:05.383,000] <err> os: ***** BUS FAULT *****
[00:00:05.383,000] <err> os:   Precise data bus error
[00:00:05.383,000] <err> os:   BFAR Address: 0x0
[00:00:05.383,000] <err> os: r0/a1:  0x20000544  r1/a2:  0x00000000  r2/a3:  0x0800e4d8
[00:00:05.383,000] <err> os: r3/a4:  0x200000e8 r12/ip:  0x20002484 r14/lr:  0x080033b9
[00:00:05.383,000] <err> os:  xpsr:  0x61000000
[00:00:05.383,000] <err> os: Faulting instruction address (r15/pc): 0x080046b0
[00:00:05.383,000] <err> os: >>> ZEPHYR FATAL ERROR 25: Unknown error on CPU 0
[00:00:05.383,000] <err> os: Current thread: 0x20000aa0 (unknown)

```

FRASTM, 2024-07-24:
Also seen with stm32h573 disco kit, when plugging the USB cable. The function usbd_cdc_acm_ctd  of the ./subsys/usb/device_next/class/usbd_cdc_acm causing a crash at memcpy because buf->data is 0

FRASTM, 2024-07-24:
@loicpoulain  please,  could you help 
  investigation shows that usbd_handle_ctrl_xfer is given a  wrong next_buf by net_buf_frag_del

I also suggest
```patch
diff --git a/subsys/usb/device_next/usbd_ch9.c b/subsys/usb/device_next/usbd_ch9.c
index d3eebcaccc5..c96c27dd3d5 100644
--- a/subsys/usb/device_next/usbd_ch9.c
+++ b/subsys/usb/device_next/usbd_ch9.c
@@ -1101,7 +1101,7 @@ int usbd_handle_ctrl_xfer(struct usbd_context *const uds_ctx,
 
 		/* Remove setup packet buffer from the chain */
 		next_buf = net_buf_frag_del(NULL, buf);
-		if (next_buf == NULL) {
+		if (next_buf == NULL || next_buf->data == NULL) {
 			LOG_ERR("Buffer for data|status is missing");
 			goto ctrl_xfer_stall;
 		}

```

FRASTM, 2024-10-04:
An issue is also seen with the b_g474e_dpow1 (USB FS)
```
*** Booting Zephyr OS build v3.7.0-454-gb7afea3ead7f ***
[00:00:00.000,000] <err> usbd_core: Failed to set default address after bus reset
[00:00:00.000,000] <inf> cdc_acm_echo: USBD message: Bus reset
[00:00:00.000,000] <inf> cdc_acm_echo: USBD message: Stack error
[00:00:00.000,000] <inf> cdc_acm_echo: USB device support enabled
[00:00:00.000,000] <inf> cdc_acm_echo: Wait for DTR
[00:00:00.003,000] <inf> cdc_acm_echo: USBD message: Device suspended
[00:00:00.101,000] <inf> cdc_acm_echo: USBD message: Device resumed
[00:00:00.156,000] <inf> cdc_acm_echo: USBD message: Bus reset
[00:00:00.244,000] <inf> cdc_acm_echo: USBD message: Bus reset
[00:00:00.327,000] <err> udc: Failed to allocate net_buf 0
[00:00:00.327,000] <err> usbd_ch9: Buffer for data|status is missing
[00:00:00.327,000] <err> usbd_ch9: Malformed setup packet
[00:00:00.327,000] <err> udc: Failed to allocate net_buf 7
[00:00:00.327,000] <err> usbd_ch9: Buffer for data|status is missing
[00:00:00.327,000] <err> usbd_ch9: Malformed setup packet
[00:00:00.327,000] <err> udc: Failed to allocate net_buf 7
[00:00:00.328,000] <err> usbd_ch9: Buffer for data|status is missing
[00:00:00.328,000] <err> usbd_ch9: Malformed setup packet
[00:00:00.328,000] <err> udc: Failed to allocate net_buf 255                                                        
[00:00:00.328,000] <err> usbd_ch9: Buffer for data|status is missing                                                
[00:00:00.328,000] <err> usbd_ch9: Malformed setup packet                                                           
[00:00:00.330,000] <err> udc: Failed to allocate net_buf 255                                                        
[00:00:00.330,000] <err> usbd_ch9: Buffer for data|status is missing                                                
[00:00:00.330,000] <err> usbd_ch9: Malformed setup packet                                                           
[00:00:00.332,000] <err> udc: Failed to allocate net_buf 255                                                        
[00:00:00.332,000] <err> usbd_ch9: Buffer for data|status is missing                                                
[00:00:00.332,000] <err> usbd_ch9: Malformed setup packet                                                           
[00:00:00.348,000] <err> udc: Failed to allocate net_buf 255                                                        
[00:00:00.348,000] <err> usbd_ch9: Buffer for data|status is missing                                                
[00:00:00.348,000] <err> usbd_ch9: Malformed setup packet                                                           
[00:00:00.350,000] <err> udc: Failed to allocate net_buf 255                                                        
[00:00:00.350,000] <err> usbd_ch9: Buffer for data|status is missing                                                
[00:00:00.350,000] <err> usbd_ch9: Malformed setup packet                                                           
[00:00:00.351,000] <err> udc: Failed to allocate net_buf 255                                                        
[00:00:00.351,000] <err> usbd_ch9: Buffer for data|status is missing                                                
[00:00:00.351,000] <err> usbd_ch9: Malformed setup packet                                                           

```

erwango, 2024-11-06:
To be tested with https://github.com/zephyrproject-rtos/zephyr/pull/80820.

fpistm, 2024-12-10:
> To be tested with #80820.

Test done but issue always here.

mmahadevan108, 2024-12-12:
I am seeing the below error message when running the HID sample on RT1060 EVK board. However the test seems to be working as I see a HID-mouse device on my Windows machine.

`west build -p -b mimxrt1060_evk samples/subsys/usb/hid-mouse -T sample.usb_device_next.hid-mouse`
```
*** Booting Zephyr OS build v4.0.0-1847-g8e0f1bfdc146 ***
[00:00:00.313,000] <err> udc: Failed to allocate net_buf 1026
[00:00:00.313,000] <err> usbd_ch9: Buffer for data|status is missing
[00:00:00.313,000] <err> usbd_ch9: Malformed setup packet
[00:00:00.319,000] <err> udc: Failed to allocate net_buf 1026
[00:00:00.319,000] <err> usbd_ch9: Buffer for data|status is missing
[00:00:00.319,000] <err> usbd_ch9: Malformed setup packet
```

closed 2025-03-18 by marwaiehm-st