zephyrproject-rtos/zephyr #76252: Reading ADC on STM32H750B-DK.

adtvtk, 2024-07-24:
Hello, I am using STM32H750B-DK board to read ADC. The readings seems to be stuck at an offset of about 1.5V. If I ground the ADC input, the voltage is 1.4V. It does respond to voltages above 1.5V. I checked on all 6 channels on the board as well as one other DK board with same result. 
I tried reading ADC differential mode selection register (ADC_DIFSEL), it is reading 0x00. 
This is the first project I am doing in Zephyr and not able to figure out why the ADC is stuck at an offset. Can you please help?

The code I used is the sample code from https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/drivers/adc/adc_dt

The overlay I used for the board is

/ {
        zephyr,user {
                /* adjust channel number according to pinmux in board.dts */
                io-channels = <&adc3 1>;
        };
};

&adc3 {
 	pinctrl-0 = <&adc3_inp1_pc3_c>;
    pinctrl-names = "default";
	st,adc-clock-source = <SYNC>;
	status = "okay";
	st,adc-prescaler = <1>;
	#address-cells = <1>;
	#size-cells = <0>;
	channel@1 {
		reg = <1>;
		#io-channel-cells = <1>;
		zephyr,gain = "ADC_GAIN_1";
		zephyr,reference = "ADC_REF_INTERNAL";
		zephyr,acquisition-time = <ADC_ACQ_TIME_DEFAULT>;
		zephyr,resolution = <16>;                         

	};
};

djiatsaf-st, 2024-07-29:
hello @adtvtk , 

I did some tests on the same card and indeed, there is this offset problem. 
I looked in the errata sheet to see if there were any hw issues, and I found that the ADC3 can be limited by the LSE activity for some pins. Based on this, I would like to know if you tested on ADC1 or 2? 

Regarding the differential mode, it is not currently supported by the ADC driver, so it is normal that you find this 0x00 value in ADC_DIFSEL register. 

You can check this by setting up this configuration. 

First, you have to activate the differential mode by adding the value: "zephyr,differential" in the overlay file 

You should also know that the differential mode only works on negative channels as specified on this link. 

 https://docs.zephyrproject.org/latest/build/dts/api/bindings/adc/st,stm32-adc.html#:~:text=is%20required.-,zephyr%2Cdifferential,-boolean   

here is an example of the overlay file on ADC1 for testing if you wish :
 
```
/ { 

        zephyr,user { 

                /* adjust channel number according to pinmux in board.dts */ 

                io-channels = <&adc1 1>; 

        }; 

}; 

&adc1 { 

            pinctrl-0 = <&adc1_inn1_pa0_c>; 
            
            pinctrl-names = "default"; 
            
            st,adc-clock-source = <SYNC>; 
            
            status = "okay"; 

            st,adc-prescaler = <1>; 
            
            #address-cells = <1>; 
            
            #size-cells = <0>; 

            channel@1 { 
                            
                            reg = <1>; 
                            
                            #io-channel-cells = <1>; 
                            
                            zephyr,gain = "ADC_GAIN_1"; 
                            
                            zephyr,reference = "ADC_REF_INTERNAL"; 
                            
                            zephyr,acquisition-time = <ADC_ACQ_TIME_DEFAULT>; 
                            
                            zephyr,resolution = <16>; 
                            
                            //zephyr,input-positive  = <1>; 
                            
                            zephyr,differential ; 
                            
                            zephyr,input-negative = <1>; 

            }; 
}; 

```
Also enable logging in console with CONFIG_LOG=y

adtvtk, 2024-07-30:
Thank you for your response. 
I did check on ADC 1. It gave similar results with offset. I will check with the differential input as you have suggested above. 

I tried implementing ADC in CubeIDE. ADC needs to be calibrated. After running calibration , the results were much better.
Right now I am trying to figure out how to implement calibration in Zephyr.