zephyrproject-rtos/zephyr #76261: External source integration for soc/wch/ch32v00x/ch32v003fun.h

fabiobaltieri, 2024-07-24:
## Origin

This is from the https://github.com/cnlohr/ch32v003fun project, original file is https://github.com/cnlohr/ch32v003fun/blob/master/ch32v003fun/ch32v003fun.h.

## Purpose

This is a register map for the WCH CH32V family of microcontrollers.

## Mode of integration

Main tree, `soc/wch/ch32v00x/ch32v003fun.h`.

## Maintainership

@kholia @nzmichaelh @cnlohr would you be happy to have an entry in the MAINTAINERS file for this? (list to be confirmed)

## Pull Request

https://github.com/zephyrproject-rtos/zephyr/pull/73761

## Description

It's just a register map for supporting drivers of these devices, the integration is being developed with no HAL so the proposal is to have the file in the main repository, keep things simpler.

The alternative would be to have it in a separate repository I suppose.

## Dependencies

None.

## Revision

n/a

## License

MIT

## Notes

The process calls for code reformatting for these cases (https://docs.zephyrproject.org/latest/contribute/external.html#integration-in-the-main-tree, point 1). The file is currently off, mainly for comments and line length. Given that this is a soc specific and there's no actual code in the file, I suggest having an exception and integrate the file as-is. This way any future update would generate an understandable diff.

Alternatives would be to either blanket clang-format the file, or integration in an external repository.

nzmichaelh, 2024-07-24:
I'm happy to be a maintainer.

cnlohr, 2024-07-31:
Please be aware I'm happy to upstream any files or direct changes that would make this easier, and to minimize the forkyness of your branch.

fabiobaltieri, 2024-07-31:
Thanks folks, ok based on the conversation in https://github.com/zephyrproject-rtos/zephyr/pull/73761, and in light of the file license and intention to at support for more parts it seems like forking https://github.com/cnlohr/ch32v003fun into a `hal_wch` repo may be the the way to go. @kholia if you agree with that and can make the PR back in that form (using the whole https://github.com/cnlohr/ch32v003fun with just the extra `module.yml` file) I can modify this proposal to request the fork from the TSC and then we can take it from there.

kholia, 2024-07-31:
@fabiobaltieri Roger - I will get this done later today.

kholia, 2024-07-31:
@fabiobaltieri PR is up now @ https://github.com/zephyrproject-rtos/zephyr/pull/76540. Thanks!

nandojve, 2024-08-01:
The RFC is to add 1 item but the whole hal and manufacturer details should be introduced in a way that allow the ARM and RISCV MCUs together, otherwise it does not scale up.

Questions:

1- How I should add pinctrl auto-generated files in the hal?
2- How should we add the BLE blobs for ch32v208? There will be a group that will start to work with this in Ago/24.
3- How ARM chips will be used?
4- What happen if manufacturer wants to be part of Zephyr in the future and assume the their own repos?

This is moving in a very odd way. All parts that are supported by community are simple hosted in the ZephyrProject. See, for instance GigaDevices and Atmel (MCHP).

fabiobaltieri, 2024-08-01:
Blobs would go in the module zephyr.yml file like for any other module, don't see any particular problem with that.

Pinctrl, any issue with those going in the main Zephyr repository? There's other SoC that have them there, seems easier to manage as well.

for ARM and if the vendor wants to take control, if and when that will happen we can always reconsider the architecture of the whole repo, push whatever sha will be current at the time in a branch so that GitHub would keep it for backward compatibility and start with a new architecture in the main branch. I think it's been done before, not a problem.

nzmichaelh, 2024-08-08:
I've started a PR to add minichlink to the Zephyr toolchain: https://github.com/zephyrproject-rtos/sdk-ng/pull/773

@cnlohr FYI

fabiobaltieri, 2024-08-14:
No concern from the TSC but they decided to vote on this next week to let some more people check this out. Sorry this is taking forever, hopefully we are on the home stretch.

fabiobaltieri, 2024-08-21:
Hey, this was approved by the TSC, can we get the repo created? @nashif @stephanosio

stephanosio, 2024-08-23:
https://github.com/zephyrproject-rtos/hal_wch has been created.

fabiobaltieri, 2024-08-23:
Thanks folks.

closed 2024-08-23 by fabiobaltieri