zephyrproject-rtos/zephyr #76276: test: boot: test_mcuboot: build failure with tinycrypt due to multiply-defined symbols

cfriedt, 2024-07-24:
**Describe the bug**

Slight modification of `tests/boot/test_mcuboot` to use tinycrypt and ecdsa p256 showed that duplicate symbols are defined, which causes the build to fail.

Preventing the duplicate symbols from being defined (i.e. not duplicating sources) solves the issue.

 - What target platform are you using? `nucleo_g071rb` but likely all platforms would be affected.
 - What have you tried to diagnose or workaround this issue? solved the issue locally (and will shortly submit a fix to mcuboot)
 - Is this a regression? Unsure.

**To Reproduce**
Steps to reproduce the behavior:
1. apply the patch listed in **Additional context**
2. run
```shell
twister -i -p nucleo_g071rb --build-only -s tests/boot/test_mcuboot/bootloader.mcuboot.tinycrypt
```
3. See error

**Expected behavior**
The sample should build with minor corrections to support tinycrypt

**Impact**
Requires users to carry workarounds.

**Logs and console output**
```shell
/opt/zephyr/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/libzephyr.a(utils.c.obj): in function `_copy':
/Users/cfriedt/zephyrproject/modules/crypto/tinycrypt/lib/source/utils.c:42: multiple definition of `_copy'; app/libapp.a(utils.c.obj):/Users/cfriedt/zephyrproject/bootloader/mcuboot/ext/tinycrypt/lib/source/utils.c:42: first defined here
/opt/zephyr/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/libzephyr.a(utils.c.obj): in function `_set':
/Users/cfriedt/zephyrproject/modules/crypto/tinycrypt/lib/source/utils.c:52: multiple definition of `_set'; app/libapp.a(utils.c.obj):/Users/cfriedt/zephyrproject/bootloader/mcuboot/ext/tinycrypt/lib/source/utils.c:52: first defined here
/opt/zephyr/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/libzephyr.a(utils.c.obj): in function `_double_byte':
/Users/cfriedt/zephyrproject/modules/crypto/tinycrypt/lib/source/utils.c:61: multiple definition of `_double_byte'; app/libapp.a(utils.c.obj):/Users/cfriedt/zephyrproject/bootloader/mcuboot/ext/tinycrypt/lib/source/utils.c:61: first defined here
/opt/zephyr/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/libzephyr.a(utils.c.obj): in function `_compare':
/Users/cfriedt/zephyrproject/modules/crypto/tinycrypt/lib/source/utils.c:70: multiple definition of `_compare'; app/libapp.a(utils.c.obj):/Users/cfriedt/zephyrproject/bootloader/mcuboot/ext/tinycrypt/lib/source/utils.c:70: first defined here
```

**Environment (please complete the following information):**

 - OS: All
 - Toolchain (e.g Zephyr SDK, ...):  Zephyr SDK v0.16.8
 - Commit SHA: d6afb8fb521844ff8a24d70f1432a428b975a1bd, 2cddfcef8932883be64f2a667d942df2e99bb7b9 (v3.6-branch),  mcuboot commit fb2cf0ec3da3687b93f28e556ab682bdd4b85223

**Additional context**
> [!NOTE]  
> It was necessary to adjust the `sysbuild.conf` and `sysbuild/mcuboot.conf` because extra args beginning with `-DSB_` or `-Dmcuboot_` do not override what is contained within the respective configuration files (which may itself be a kind of bug with sysbuild).

```diff
diff --git a/boards/st/nucleo_g071rb/nucleo_g071rb.dts b/boards/st/nucleo_g071rb/nucleo_g071rb.dts
index 9237cd779b0..912dd25583f 100644
--- a/boards/st/nucleo_g071rb/nucleo_g071rb.dts
+++ b/boards/st/nucleo_g071rb/nucleo_g071rb.dts
@@ -21,8 +21,14 @@
 		zephyr,shell-uart = &usart2;
 		zephyr,sram = &sram0;
 		zephyr,flash = &flash0;
+		zephyr,code-partition = &slot0_partition;
+		zephyr,uart-mcumgr = &usart2;
 	};
 
+	aliases {
+		mcuboot-button0 = &user_button;
+		mcuboot-led0 = &green_led_1;
+	};
 
 	leds: leds {
 		compatible = "gpio-leds";
@@ -166,11 +172,31 @@
 		compatible = "fixed-partitions";
 		#address-cells = <1>;
 		#size-cells = <1>;
+		status = "okay";
+
+        boot_partition: partition@0 {
+            label = "mcuboot";
+            reg = <0x0 DT_SIZE_K(32)>;
+            read-only;
+        };
+
+		/* The MCUBoot swap-move algorithm uses the last 2 sectors
+		 * of the primary slot0 for swap status and move.
+		 */
+		slot0_partition: partition@8000 {
+			label = "image-0";
+			reg = <0x00008000 (DT_SIZE_K(44) + DT_SIZE_K(2) + DT_SIZE_K(2) + DT_SIZE_K(2))>;
+		};
+
+		slot1_partition: partition@14800 {
+			label = "image-1";
+			reg = <0x00014800 DT_SIZE_K(44)>;
+		};
 
 		/* Set 2KB of storage at the end of 128KB flash */
 		storage_partition: partition@1f800 {
-			label = "storage";
-			reg = <0x0001f800 DT_SIZE_K(2)>;
+				label = "storage";
+				reg = <0x0001f800 DT_SIZE_K(2)>;
 		};
 	};
 };
diff --git a/tests/boot/test_mcuboot/sysbuild.conf b/tests/boot/test_mcuboot/sysbuild.conf
index 47f00ff3cff..7b3f23a7263 100644
--- a/tests/boot/test_mcuboot/sysbuild.conf
+++ b/tests/boot/test_mcuboot/sysbuild.conf
@@ -1 +1,3 @@
 SB_CONFIG_BOOTLOADER_MCUBOOT=y
+
+SB_CONFIG_BOOT_SIGNATURE_TYPE_ECDSA_P256=y
diff --git a/tests/boot/test_mcuboot/sysbuild/mcuboot.conf b/tests/boot/test_mcuboot/sysbuild/mcuboot.conf
index 69938a0f972..ff71908e9a7 100644
--- a/tests/boot/test_mcuboot/sysbuild/mcuboot.conf
+++ b/tests/boot/test_mcuboot/sysbuild/mcuboot.conf
@@ -1 +1,8 @@
 CONFIG_MCUBOOT_LOG_LEVEL_INF=y
+
+CONFIG_MBEDTLS=n
+CONFIG_BOOT_SIGNATURE_TYPE_RSA=n
+
+CONFIG_TINYCRYPT=y
+CONFIG_BOOT_SIGNATURE_TYPE_ECDSA_P256=y
+CONFIG_BOOT_ECDSA_TINYCRYPT=y
diff --git a/tests/boot/test_mcuboot/testcase.yaml b/tests/boot/test_mcuboot/testcase.yaml
index abb1d3a44b1..ccd958bbef0 100644
--- a/tests/boot/test_mcuboot/testcase.yaml
+++ b/tests/boot/test_mcuboot/testcase.yaml
@@ -35,6 +35,7 @@ tests:
       - mimxrt685_evk/mimxrt685s/cm33
       - nrf52840dk/nrf52840
       - rd_rw612_bga
+      - nucleo_g071rb
     integration_platforms:
       - frdm_k64f
       - nrf52840dk/nrf52840
```

nordicjm, 2024-07-25:
The issue is you are selecting tinycrypt when you should not be, there is a special different tinycrypt symbol that includes what is needed for ECDSA-P256, the normal symbol should not be used. And sysbuild takes care of selecting the symbol anyhow, it should not be in a mcuboot conf file, just the sysbuild conf file

cfriedt, 2024-07-25:
> The issue is you are selecting tinycrypt when you should not be, there is a special different tinycrypt symbol that includes what is needed for ECDSA-P256, the normal symbol should not be used. And [...]

Hmmm... This is fairly confusing. Is it documented anywhere?

nordicjm, 2024-07-25:
> Hmmm... This is fairly confusing. Is it documented anywhere?

Not sure, @d3zd3z ?

cfriedt, 2024-07-26:
closing - it might be worth it to doc the tinycrypt use case

closed 2024-07-26 by cfriedt