zephyrproject-rtos/zephyr #76280: http_server dynamic resources callback/handler API usage

djr-spectrummfg, 2024-07-24:
The new http_server code's dynamic resource handlers are not adequately explained in the documentation, and the API itself is difficult to use practically. In particular, the sample code which merely echos the received data as it arrives is insufficient as it does not reflect real usage of receiving a request, processing it (possibly requiring the whole request) and generating a response (possibly containing data from external sources).

Ideally I would like an API where the handler is called once for each request (possibly truncated), and returns a complete response in one go. Maybe look at the existing API for CoAP servers

djr-spectrummfg, 2024-07-24:
It should also be possible for the dynamic handler to generate arbitrary HTTP headers and status code in the response. Right now it doesn't seem to be possible.

rlubos, 2024-07-25:
It would be good to summarize specific ideas for the dynamic resource API improvements here, so it's clear what to implement/change.

What I found a bit cumbersome with the current design, is that even for simple GET requests, the dynamic callback has to be called at least 2 times, because the only information whether the response payload is the return value of 0. One idea to approach this would be to make the `enum http_data_status status` bidirectional. Currently it's only used to inform the callback whether the request payload is complete, but it could also be used to notify the server whether the reply payload is complete or not, so it doesn't need to call the callback again. That way simple requests could be served in one go.

> It should also be possible for the dynamic handler to generate arbitrary HTTP headers and status code in the response. Right now it doesn't seem to be possible.

Perhaps we need to come up with some sort of reply context structure, which could be filled out by the dynamic handler, so the server could use that information when generating response.

rlubos, 2024-07-31:
>  it would be great if the application could also access the received headers (which unless I've missed something is not possible at the moment)

We'd need to figure out how to best tackle this, we'd probably need some intermediate buffer to store the headers while processing the request, although that'd increase the memory requirement for individual clients. Maybe there's a smarter way to handle this though I don't see right now (ideas welcome).

mrodgers-witekio, 2024-07-31:
> We'd need to figure out how to best tackle this, we'd probably need some intermediate buffer to store the headers while processing the request, although that'd increase the memory requirement for [...]

Perhaps some mechanism to register application callbacks to be called inside the internal ones here:
```
client->parser_settings.on_header_field = on_header_field;
client->parser_settings.on_header_value = on_header_value;
client->parser_settings.on_headers_complete = on_headers_complete;
```
might work, for HTTP1 at least (I'm not familiar enough with HTTP2 to figure out how it could work in this case).

Then the application could either do some action based on the headers "on the fly", or store them in its own buffer for later use.

mrodgers-witekio, 2024-08-13:
@rlubos any thoughts on this: https://github.com/zephyrproject-rtos/zephyr/pull/77003 as an approach to the application accessing request headers?

I also have some ideas on how the "other direction" ie. sending application-defined headers and status codes could work, but not sure I'll have time to put anything together soon.

rlubos, 2024-08-13:
@mrodgers-witekio Thanks for looking into this. I've left a few comments in the PR, although to be honest I think we need to resort to some sort of dynamic allocation for the headers... There's just too many variables when it comes to headers to predict/preallocate every buffer, and for HTTP/2 we'd need some flexibility.

FYI, I'll be off for the rest of August, but I'm quite sure @jukkar will be happy to discuss any HTTP-specific issues.

closed 2024-09-23 by carlescufi