zephyrproject-rtos/zephyr #76299: driver: ssp: fix MLCS bits programming

dnikodem, 2024-07-25:
**Describe the bug**

The MLCS field in the SSP driver is programmed with incorrect bits length. The MLCS field length is a 3-bit value, whereas the macro used to program it attempts to modify 4 bits.

It is necessary to modify the macro used to program the MLCS field and the method used to program the link clock source.

**To Reproduce**

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

marc-hb, 2024-07-25:
Thanks for filing this. I think you said something important in #76258

> In my opinion it is not critical cause probably no one will change the MLCS register value in runtime to select different clock.

Your fix should still be merged of course.

dnikodem, 2024-07-26:
Fix: https://github.com/zephyrproject-rtos/zephyr/pull/76310 - in review

dnikodem, 2024-07-26:
Waiting for CI results : https://github.com/thesofproject/sof/pull/9332

dnikodem, 2024-08-29:
Fix is merged: https://github.com/zephyrproject-rtos/zephyr/pull/76310

closed 2024-08-29 by marc-hb