zephyrproject-rtos/zephyr #76307: Add 11k/v/r roaming support

nxf58150, 2024-07-26:
## Introduction
STA needs to roaming to better AP when RSSI of currently associated AP decreases. Add 11k, 11v and 11r roaming support.

### Proposed change
Once signal change is reported, roaming will be triggered in wpa_supplicant based on configured roaming method. Net mgmt or glue layer will send different commands to wpa_supplicant through ctrl interface and wpa_supplicant will handle reassociation process.

## Detailed RFC
Roaming starts after EVENT_SIGNAL_CHANGE is received in wpa_supplicant. 
11k roaming:
Send commands to wpa_supplicant to send Neighbor report request. Save results of Neighbor report in wifi mgmt and trigger limited scan. Then wpa_supplicant will handle reassociation process after receive scan results. 
[image]
[image]

11v/r roaming:
Send commands to wpa_supplicant to trigger 11v or 11r roaming. The wpa_supplicant will handle the rest of the work, including send corresponding frames and trigger scan.
[image]

If more than one roaming method is supported, roaming method will be selected based on following priority: 11r > 11k > 11v.

krish2718, 2024-07-27:
Thanks @nxf58150 , I have slightly formatted and reworded a few sentences in the description.

Few things we had discussed in the Wi-Fi telco:

1. Review storing neighbor reports in `wifi_mgmt` vs. storing in WPA supplicant itself, this can be a proper upstream feature. We don't want to introduce `state/context` to stateless `wifi_mgmt` layer.
2. `NET_EVENT_WIFI_NEIGHBOR_REP_COMPLETE` can be implemented using the monitor control interface, for this we need to implement `RRM-NEIGHBOR-REP-COMPLETE` event in supplicant, this way there would only be a single mechanism i.e., monitor control sockets.
3. Need to come up wit Kconfig options and Wi-Fi shell commands for these features.
    - Enable necessary options in WPA supplicant.

nxf58150, 2024-08-15:
Hi @krish2718,

Recently, we found one issue of 11r roaming that 11r roaming can't be completed within 60s zperf UDP throughput test. The root cause is that the 11r roaming is triggered in net_mgmt task, which is with very low priority. When running UDP throughput test, the net_mgmt task didn't have much chance to run.  Same issue could be found when doing 11k and 11v roaming tests. Therefore, we plan to change our 11k/v/r roaming implementation.

We plan to create new task(name it "supp_mgmt" temporally) and handle roaming. The priority will be same as zperf task. New files supp_mgmt.c and supp_mgmt.h will be added to glue layer. 

Could you please add your comments on this? 

Thanks you!

krish2718, 2024-08-15:
If it's just a matter of thread `priority`, when supplicant is enabled we can just bump the priority of network management thread, will that work?

> We plan to create new task(name it "supp_mgmt" temporally) and handle roaming. The priority will be same as zperf task. New files supp_mgmt.c and supp_mgmt.h will be added to glue layer.

If at all we need a separate thread, can't we just add it to `supp_event.c` itself? and let `supplicant_send_wifi_mgmt_event` use port to that thread?

nxf58150, 2024-08-16:
Hi @krish2718,

If increasing priority of net_mgmt task is ok for you, then I will change the priority to the same value of supplicant task. This might be the simplest way to resolve the issue and at the same time, keep 11k/v/r roaming plan unchanged. 
Thank you for your advice. I will try it on my side.

krish2718, 2025-01-14:
@nxf58150 can you please review and update if this can be closed? If not please add the pending items.

nxf58150, 2025-01-15:
> [@nxf58150](https://github.com/nxf58150) can you please review and update if this can be closed? If not please add the pending items.

Hi @krish2718 11kvr roaming support is already merged. This issue can be closed. Thanks!

closed 2025-01-15 by nxf58150

krish2718, 2025-03-24:
@nxf58150 For the 802.11 r/v based roaming, how does the firmware decide when to send `WIFI_EVENT_RSSI_LOW`? Is it hard-coded in the firmware? WPA supplicant has an `signal_monitor` https://github.com/zephyrproject-rtos/hostap/blob/main/src/drivers/driver.h#L4160 that provides the params to the firmware, may be we should implement that?

nxf58150, 2025-03-25:
Hi @krish2718 , we have a command nxp_wifi roaming to set the rssi low threshold to firmware.

krish2718, 2025-03-25:
> Hi [@krish2718](https://github.com/krish2718) , we have a command nxp_wifi roaming to set the rssi low threshold to firmware.

Ah, so, a proprietary command? We should then add a wifi_mgmt command and then implement `signal_monitor` OP to make it standardize.