zephyrproject-rtos/zephyr #76311: STM32WBA52 (Revision ID: Rev B) can't run BLE demos successfully. bt_settings: settings_subsys_init failed (err -33)[0m Bluetooth init failed (err -33)

LucienKUANG-vip, 2024-07-26:
STM32WBA52 (Revision ID: Rev B) can't run BLE demos successfully. bt_settings: settings_subsys_init failed (err -33)[0m Bluetooth init failed (err -33)

**Describe the bug**

My customers use STM32WBA52 chip cut2.0 version chip (Revision ID: Rev B) to develop their products. They want to use zephyr system and need to support BLE features.

When I was testing the basic Bluetooth function, I found that the BLE features of this version（cut 2.0） of the chip could not be successfully run in the current project, no matter which version of LL stack library file I used.

The same example (zephyrproject\zephyr\samples\bluetooth) can run normally when I use the Cut1.0 chip.

I also raise it on bugzilla.

**[ticket 187081](https://intbugzilla.st.com/show_bug.cgi?id=187081 "Assigned - Does the STM32WBA52 support Zephyr BLE applications?"): [Ticket 187081]Does the STM32WBA52 support Zephyr BLE applications?**

**To Reproduce**

1. cd /c/Users/usr name/zephyrproject/zephyr
2. west update
3. west blobs fetch stm32
   Fetching blob stm32: C:\Users\lucien\zephyrproject\modules\hal\stm32\zephyr\blobs\stm32wba\lib\LinkLayer_BLE_Full_lib.a
   Fetching blob stm32: C:\Users\lucien\zephyrproject\modules\hal\stm32\zephyr\blobs\stm32wba\lib\stm32wba_ble_stack_llo.a
4. cd  /c/Users/usr name/zephyrproject/zephyr/samples/bluetooth/peripheral
5. west build -b nucleo_wba52cg ./

[231/231] Linking C executable zephyr\zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      394112 B         1 MB     37.59%
             RAM:       61329 B       128 KB     46.79%
           SRAM6:          0 GB        16 KB      0.00%
        IDT_LIST:          0 GB        32 KB      0.00%
Generating files from C:/Users/lucien/zephyrproject/zephyr/samples/bluetooth/peripheral/build/zephyr/zephyr.elf for board: nucleo_wba52cg

6. west flash
7. Open the serial port assistant to view the log

**Logs and console output**

*** Booting Zephyr OS build v3.7.0-rc3-102-g099209b7d563 ***

[00:00:00.000,000] [1;31m `<err>` bt_settings: settings_subsys_init failed (err -33)[0m

Bluetooth init failed (err -33)

FRASTM, 2024-07-26:
@asm5878

piotrParf, 2024-07-26:
probably issue is connected. https://github.com/zephyrproject-rtos/zephyr/issues/75318

LucienKUANG-vip, 2024-07-29:
> probably issue is connected. #75318

> probably issue is connected. #75318

They are different.

asm5878, 2024-08-02:
Could you please let us know the details of your board, please?
I am not aware of an RF board (MB1863) with STM32WBA52 cut 2.0

LucienKUANG-vip, 2024-08-06:
> Could you please let us know the details of your board, please? I am not aware of an RF board (MB1863) with STM32WBA52 cut 2.0

Hello 
This issue has been solved and the board  works normally now. The problem was caused by soldering a STM32WBA52CEU6(512K Flash)instead of a STM32WBA52CGU6 when replacing the cut2.0 chip on the RF board (MB1863).
Thank you for your information.

closed 2024-08-06 by LucienKUANG-vip